subsurface/subsurface #3492: Can only download last few dives on Oceanic ProPlus 4

modhner, 2022-08-20:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I have a ProPlus 4 with 19 dives on it.  Depending on the OS I am running on, it only downloads the last 4, 7, or 8 dives.  It seems like something times out and the download aborts.  After that happens, when I try again, it fails to connect to the device at all.  If I wait long enough and restart Subsurface before retrying then I can get it to do the partial download again.
(Fortunately, using DiverLog+ I was able to get all the dives.)
### Operating system:

Android 12
IPad 15.6
Windows 10

Android phone, iPad, PC

Yes, but the number of dives downloaded before it quits is different.
### Subsurface version:

Windows: 5.0.9
IPad: 3.4.6(5.0.9.7)
Android: 3.4.6(5.0.9)
Ubuntu 20.04.4: subsurface 5.0.8-9-gc44b5017f4dc

Yes.

Official release (except on ubuntu)

### Steps to reproduce:

1) Oceanic ProPlus 4 with 19 dives
2) Turn on the PDC
3) Select "Import from Dive Computer"
4) Enter the device info and "Choose Bluetooth download mode", select the device and save
5) Select "Force download all dives"
6) Click "Download"
### Current behavior:

A) It says "Error opening the device ...".  About half a minute later, the dive computer goes into "BT PAIR" mode.
B) Sometimes it does connect but only four to nine dives are downloaded (depending on the OS).
C) After downloading some of the dives, it sometimes says "Dive data import error".
### Expected behavior:

All 19 dives should be downloaded.

### Additional information:

The attached log file is from when the "Dive data import error" happened.
The log included below was from when it failed to open the device:
Subsurface: v5.0.8-9-gc44b5017f4dc, built with libdivecomputer v0.8.0-devel-Subsurface-NG (77498afe5e2bc1c9c60309202b26f359581271a7)
[33.155604] INFO: Open: transport=32
[33.155608] INFO: Configure: baudrate=38400, databits=8, parity=0, stopbits=0, flowcontrol=0
[33.155609] INFO: Timeout: value=1000
[33.155609] INFO: DTR: value=1
[33.155610] INFO: RTS: value=0
[33.155610] INFO: Sleep: value=100
[33.255729] INFO: RTS: value=1
[33.255732] INFO: Sleep: value=100
[33.355806] INFO: Purge: direction=3
[33.356904] INFO: Write: size=5, data=CD40000184
[34.374508] INFO: Read: size=0, data=
[34.374510] ERROR: Failed to receive the answer. [in src/oceanic_atom2.c:681 (oceanic_atom2_packet)]
[34.374512] INFO: Sleep: value=100
[34.475617] INFO: Purge: direction=1
[34.475620] INFO: Sleep: value=1
[34.484483] INFO: Write: size=5, data=CD40000184
[35.486249] INFO: Read: size=0, data=
[35.486251] ERROR: Failed to receive the answer. [in src/oceanic_atom2.c:681 (oceanic_atom2_packet)]
[35.486252] INFO: Sleep: value=100
[35.590689] INFO: Purge: direction=1
[35.590692] INFO: Sleep: value=2
[35.620275] INFO: Write: size=5, data=CD40000184
[36.624411] INFO: Read: size=0, data=
[36.624413] ERROR: Failed to receive the answer. [in src/oceanic_atom2.c:681 (oceanic_atom2_packet)]

### Mentions:

[subsurface.log](https://github.com/subsurface/subsurface/files/9388039/subsurface.log)

jefdriesen, 2022-08-30:
Can you try to download a full memory dump of the dive computer? Simply enable both the "Save libdivecomputer dumpfile" and "Save libdivecomputer logfile" checkboxes in the download dialg. No dives will be imported, that's normal. Attach both files to the ticket. Note that downloading a full memory dump is very slow over bluetooth, so if you also have the cable, that is certainly the preferred option.

modhner, 2022-09-18:
I tried downloading the memory dump, but it has the same problem.  (I don't have a cable.) The first few tries I got the "error opening device" error.  Then when it did connect, it looked like it was downloading for about a minute and then showed "dive data import error".  It didn't save a .bin file.  After it fails, the PDC periodically displays "bt PAIR".  Is the bluetooth driver on the PC periodically trying to make a connection, and does that interrupt the data download?

jefdriesen, 2022-09-19:
That sounds as if the bluetooth connection was disconnected somehow, and the dive computer is waiting to start the pairing procedure again.

Do you have a log of the failed memory dump download? Maybe it already contains some useful info?

modhner, 2022-09-19:
[subsurface.log](https://github.com/subsurface/subsurface/files/9598259/subsurface.log)

jefdriesen, 2022-11-18:
Can you try with the this build: https://libdivecomputer.org/builds/subsurface/libdivecomputer-0.proplus4.64.dll

Download the file, remove the .proplus.64 suffix from the file name, copy the dll into the subsurface installation directory (backup the existing one, so you can restore it afterwards) and then try to download your dives. Attach the log again.

modhner, 2023-01-01:
I am still getting the "error opening device" error.  It is happening immediately after I try to start the download.
[subsurface.log](https://github.com/subsurface/subsurface/files/10329139/subsurface.log)

modhner, 2023-01-01:
After retrying many times, restarting the sw, disabling and reenabling bluetooth, etc., I did finally get it to download 15 of the 19 dives.  Unfortunately, I did not have the log enabled that time.  But every other time I tried I just got error opening device.

modhner, 2023-01-01:
Even when subsurface is not running, if my PC is turned on and bluetooth is enabled, then the dive computer periodically goes into "bt pair" mode.  (It shows "bt pair" for 30 seconds, then the normal display for about 60 seconds, then "bt pair" appears again, etc.)  So the BT driver on the PC must be polling periodically to see if the device is connected.  I think that is interfering with Subsurface connecting to the device.

jefdriesen, 2023-01-06:
There is indeed an issue with the Pro Plus 4 where it disconnects after about 30 seconds. That's what is happening in your first two log files. That explains why you can sometimes download a few dives, but not all.  I already have a solution for that problem (still under testing).

But what you are showing in that last log (and also the short log in your initial report) is something completely different. In those logs, you receive no response from the dive computer at all. That sounds more like an issue with the bluetooth connection.

If you know how to run a command-line application, then you can try this build:

https://libdivecomputer.org/builds/experimental/windows/dctool-ble-proplus4-autodetect50.exe

You need to make sure your device computer has bluetooth enabled and is paired in Windows, and then execute the command-line application with these options:

`dctool-ble-proplus4-autodetect50.exe -vv -l atom2.log -f atom2 dump -o atom2.bin -t ble`

If it fails to pick your dive computer automatically, perform a manual scan with this command:

`dctool-ble-proplus4-autodetect50.exe -vv scan -t ble`

and then append the MAC address of the dive computer as the last argument to the previous command. (This is still work in progress.)

This build contains the fix for the first issue, and also uses a different BLE implementation than the one used by subsurface. That could make a difference for the latter problem.

modhner, 2023-01-07:
Hi Jeff
I tried the dctool build (with the DC mac address) and one time it worked apparently without any errors, but other times it failed.  This is the end of the output when it worked:

> Event: progress 99.61% (65280/65536)
> INFO: Write: size=7, data=CD401403B40FF0
> INFO: Read: size=20, data=CDE014105A000100000024000000000000000000
> INFO: Read: size=20, data=CDE1141000000000000000000000000000000000
> INFO: Read: size=20, data=CDE2141000000000000000BE4200000992000000
> INFO: Read: size=20, data=CDE3141000000058020000000000003200300000
> INFO: Read: size=20, data=CDE4141000000000000000900200002070807580
> INFO: Read: size=20, data=CDE514107500000000A014000006010601000000
> INFO: Read: size=20, data=CDE614100044901D070916CF03CF0300005D0000
> INFO: Read: size=20, data=CDE7141000572304074200002109000040140000
> INFO: Read: size=7, data=CDC8140349F00A
> Event: progress 99.63% (65296/65536)
> Event: progress 99.66% (65312/65536)
> Event: progress 99.68% (65328/65536)
> Event: progress 99.71% (65344/65536)
> Event: progress 99.73% (65360/65536)
> Event: progress 99.76% (65376/65536)
> Event: progress 99.78% (65392/65536)
> Event: progress 99.80% (65408/65536)
> INFO: Write: size=7, data=CD401503B40FF8
> INFO: Read: size=20, data=CDE015105A015103020400000000040000000000
> INFO: Read: size=20, data=CDE1151000880000008C155C01FFFFFFFFFFFFFF
> INFO: Read: size=20, data=CDE21510FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
> INFO: Read: size=20, data=CDE31510FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
> INFO: Read: size=20, data=CDE41510FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
> INFO: Read: size=20, data=CDE51510FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
> INFO: Read: size=20, data=CDE61510FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
> INFO: Read: size=20, data=CDE71510FF509207012300FFFFFFFFFFFFFFFFFF
> INFO: Read: size=7, data=CDC81503FF9064
> Event: progress 99.83% (65424/65536)
> Event: progress 99.85% (65440/65536)
> Event: progress 99.88% (65456/65536)
> Event: progress 99.90% (65472/65536)
> Event: progress 99.93% (65488/65536)
> Event: progress 99.95% (65504/65536)
> Event: progress 99.98% (65520/65536)
> Event: progress 100.00% (65536/65536)
> Event: model=18006 (0x00004656), firmware=12609 (0x00003141), serial=6952 (0x00001b28)
> INFO: Write: size=8, data=CD4016046A05A500
> INFO: Read: size=5, data=CDC01601A5
> 

The second time I got this output:

> C:\Users\modhn\Downloads>dctool-ble-proplus4-autodetect50.exe -vv -l atom2.log -f atom2 dump -o atom2.bin -t ble D3F0000011E4 > dctool.log
> DATETIME 2023-01-07T16:41:18Z (1673109678)
> VERSION 0.8.0-devel (82ffcc047cc40944862527324779cea92159a31a)
> Opening the I/O stream (ble, D3F0000011E4).
> INFO: Open: address=D3F0000011E4
> INFO: Service: handle=0001, uuid=00001801-0000-1000-8000-00805f9b34fb
> INFO: Service: handle=0014, uuid=00001800-0000-1000-8000-00805f9b34fb
> INFO:   Characteristic: handle=0015, uuid=00002a00-0000-1000-8000-00805f9b34fb, flags=R
> INFO:   Characteristic: handle=0017, uuid=00002a01-0000-1000-8000-00805f9b34fb, flags=R
> INFO: Service: handle=004d, uuid=0000180a-0000-1000-8000-00805f9b34fb
> INFO:   Characteristic: handle=004e, uuid=00002a29-0000-1000-8000-00805f9b34fb, flags=R
> INFO:   Characteristic: handle=0050, uuid=00002a24-0000-1000-8000-00805f9b34fb, flags=R
> INFO:   Characteristic: handle=0052, uuid=00002a23-0000-1000-8000-00805f9b34fb, flags=R
> INFO: Service: handle=0061, uuid=0000180f-0000-1000-8000-00805f9b34fb
> INFO:   Characteristic: handle=0062, uuid=00002a19-0000-1000-8000-00805f9b34fb, flags=R
> INFO: Service: handle=1000, uuid=cb3c4555-d670-4670-bc20-b61dbc851e9a
> INFO:   Characteristic: handle=1100, uuid=5ee30931-ccaa-4870-ab32-6b87ef1f78ed, flags=R
> INFO:   Characteristic: handle=1110, uuid=17738114-18db-467c-b15f-a5bdb84025be, flags=R
> INFO:   Characteristic: handle=1200, uuid=6606ab42-89d5-4a00-a8ce-4eb5e1414ee0, flags=RW(WWR)
> DEBUG: RX: 6606ab42-89d5-4a00-a8ce-4eb5e1414ee0
> INFO:   Characteristic: handle=1300, uuid=a60b8e5c-b267-44d7-9764-837caf96489e, flags=RWN
> INFO:           Descriptor: handle=1302, uuid=00002902-0000-1000-8000-00805f9b34fb, type=2
> DEBUG: TX: a60b8e5c-b267-44d7-9764-837caf96489e
[... 25 lines cut ...]
> INFO:   Characteristic: handle=004e, uuid=00002a29-0000-1000-8000-00805f9b34fb, flags=R
> INFO:   Characteristic: handle=0050, uuid=00002a24-0000-1000-8000-00805f9b34fb, flags=R
> INFO:   Characteristic: handle=0052, uuid=00002a23-0000-1000-8000-00805f9b34fb, flags=R
> INFO: Service: handle=0061, uuid=0000180f-0000-1000-8000-00805f9b34fb
> INFO:   Characteristic: handle=0062, uuid=00002a19-0000-1000-8000-00805f9b34fb, flags=R
> INFO: Service: handle=1000, uuid=cb3c4555-d670-4670-bc20-b61dbc851e9a
> INFO:   Characteristic: handle=1100, uuid=5ee30931-ccaa-4870-ab32-6b87ef1f78ed, flags=R
> INFO:   Characteristic: handle=1110, uuid=17738114-18db-467c-b15f-a5bdb84025be, flags=R
> INFO:   Characteristic: handle=1200, uuid=6606ab42-89d5-4a00-a8ce-4eb5e1414ee0, flags=RW(WWR)
> DEBUG: RX: 6606ab42-89d5-4a00-a8ce-4eb5e1414ee0
> INFO:   Characteristic: handle=1300, uuid=a60b8e5c-b267-44d7-9764-837caf96489e, flags=RWN
> INFO:           Descriptor: handle=1302, uuid=00002902-0000-1000-8000-00805f9b34fb, type=2
> DEBUG: TX: a60b8e5c-b267-44d7-9764-837caf96489e
> INFO:   Characteristic: handle=1400, uuid=a60b8e5c-b267-44d7-9d65-857bad95479f, flags=R
> INFO: Service: handle=ff00, uuid=9e5d1e47-5c13-43a0-8635-82ad38a1386f
> INFO:   Characteristic: handle=ff01, uuid=e3dd50bf-f7a7-4e99-838e-570a086c666b, flags=WNI
[... cut ...]
> INFO:   Characteristic: handle=ff04, uuid=92e86c7a-d961-4091-b74f-2409e72efe36, flags=W
> INFO:   Characteristic: handle=ff06, uuid=347f7608-2e2d-47eb-913b-75d4edc4de3b, flags=R
> ERROR: The semaphore timeout period has expired (-2147024775) [in src/ble-win32.c:504 (win32_ble_notify)]
> ERROR: Failed to enable notifications for the Tx characteristic. [in src/ble.c:742 (dc_ble_open)]
> ERROR: Failed to open the ble device. [in examples/common.c:604 (dctool_ble_open)]
> ERROR: Error opening the I/O stream. [in examples/dctool_dump.c:57 (dump)]
> ERROR: Input/output error
>

modhner, 2023-01-07:
The next run completed successfully again.

modhner, 2023-01-07:
I was able to download all the dives from a different DC of the same kind using your subsurface build.