subsurface/subsurface #3493: Dive site import should ignore UUID values.

tggm, 2022-08-22:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:
I made small script to import dive sites from the SSI webpage for my country. While importing I noted that each `<site>` entry on the XML must have an UUID value. There appears to be some requirements for the value of this field and I eventually managed to import my generated file by providing an incremental UUID value starting at `uuid="10000000"`.

Initially I provided the same UUID value for all entries but that resulted in subsurface merging all entries into one.

I would like to suggest that this value be optional for importing dive sites. This would facilitate users generating and importing their own divesites XML files. Subsurface should generate the site's UUID internally.

### Operating system:
Windows 10/64bit

### Subsurface version:
Version 5.0.9

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

bstoeger, 2022-08-28:
Could you please post more details of what you are trying to achieve? Notably, an example XML file which shows the problem and a precise description of how to reproduce the error.

tggm, 2022-08-28:
First of all this is a feature request not an error.

What I'm trying to achieve is: Import dive sites via a manually generated XML file **without having to provide UUID values.**. 

The ultimate purpose for this is: Import dive sites for a specific region/country to have them ready whenever I need to log a dive on that region. 

Example:
```
<divesites program="subsurface" version="3">
  <site name="Cavernas" gps="38.4729 -9.1973" description="lorem ipsum." uuid="10000000">
    <geo cat="2" origin="2" value="Portugal" />
  </site>
  <site name="Paredes do Cabo" gps="38.4121 -9.2160" description="lorem ipsum" uuid="10000001">
    <geo cat="2" origin="2" value="Portugal" />
  </site>
```

As you can see each <site> element requires an UUID attribute. In this case I'm faking those UUIDs with an incremental values. I could also can put random UUIDs but the problem is the same: They can collide with existing ones. 

This is not an optimal solution. The UUID is clearly an internal value used by Subsurface and shouldn't be on the imported XML files. Subsurface *should* generate the necessary UUIDs during the import process.

Below is an example of the XML I generated based on data obtained on the SSI websites. The file contains the dive sites for Portugal. (extension changed to TXT)

[Portugal-Divesites.txt](https://github.com/subsurface/subsurface/files/9440152/Portugal-Divesites.txt)

bstoeger, 2022-08-29:
Incrementing the UUID is of course a very bad idea, since you will generate duplicates. To generate UUIDs you should follow the code in `core/divesite.c`. The whole thing does seem a bit fishy, but hey.

Removing the empty UUID check is simple:
```
From acb9d11d16a8e649edc02224243ce6bac5474a5d Mon Sep 17 00:00:00 2001
From: Berthold Stoeger <bstoeger@mail.tuwien.ac.at>
Date: Mon, 29 Aug 2022 07:25:21 +0200
Subject: [PATCH] parser: allow import of dive sites without UUID

---
 core/parse.c | 12 ++++++------
 1 file changed, 6 insertions(+), 6 deletions(-)

diff --git a/core/parse.c b/core/parse.c
index efe5c5bb1..e8161b395 100644
--- a/core/parse.c
+++ b/core/parse.c
@@ -247,13 +247,13 @@ void dive_site_end(struct parser_state *state)
 {
 	if (!state->cur_dive_site)
 		return;
-	if (state->cur_dive_site->uuid) {
-		struct dive_site *ds = alloc_or_get_dive_site(state->cur_dive_site->uuid, state->sites);
-		merge_dive_site(ds, state->cur_dive_site);
 
-		if (verbose > 3)
-			printf("completed dive site uuid %x8 name {%s}\n", ds->uuid, ds->name);
-	}
+	struct dive_site *ds = alloc_or_get_dive_site(state->cur_dive_site->uuid, state->sites);
+	merge_dive_site(ds, state->cur_dive_site);
+
+	if (verbose > 3)
+		printf("completed dive site uuid %x8 name {%s}\n", ds->uuid, ds->name);
+
 	free_dive_site(state->cur_dive_site);
 	state->cur_dive_site = NULL;
 }
-- 
2.34.1
```

I'll leave it to @dirkhh to decide if this is the right thing to do.

dirkhh, 2022-08-29:
This `uuid` check makes no sense. Please submit a PR to kill that.

PR #3509 (2022-08-31): parser: allow import of dive sites without UUID
Fixes #3493.

### Describe the pull request:

- [ ] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Changes made:

Don't ignore non-UUID dive sites on import.

### Related issues:

Fixes #3493.

### Release note:

Done.
commit 6d77d814 (2022-08-31): parser: allow import of dive sites without UUID

Fixes #3493.
closed 2022-08-31 by dirkhh