subsurface/subsurface #3494: Reading device info from Suunto VYper Novo fails

tomas-kucera, 2022-08-23:
### Describe the issue:

- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I am trying to read device info from Suunto Vyper Novo.
Downloading dives works fine.

### Operating system:

macOS Monterey 12.5 - English version

Suunto Vyper Novo

### Subsurface version:

Subsurface 5.0.9

no idea

Official release

### Steps to reproduce:

1) connect Suunto Vyper Novo via USB
2) open Subsurface
3) go to File > Change settings on dive computer
4) select FTDI or the USB mount
5) select Suunto Vyper Family computer
6) click Connect
7) click Retrieve available details 

### Current behavior:

Currently it show a little progress and hangs. App is responding but it never finishes reading.

### Expected behavior:

Read and displayed compoter's settings.

### Additional information:

libdivecomputer logfile:
Subsurface: v5.0.9, built with libdivecomputer v0.8.0-devel-Subsurface-NG (107f5b14e31f276a9b5accde2c5032ea49602403)
[139.905296] INFO: Open: transport=1
[139.905300] INFO: Configure: baudrate=2400, databits=8, parity=1, stopbits=0, flowcontrol=0
[139.905692] INFO: Timeout: value=1000
[139.905696] INFO: DTR: value=1
[139.905838] INFO: Sleep: value=100
[140.007133] INFO: Purge: direction=3
[141.851132] INFO: Sleep: value=500
[142.354214] INFO: RTS: value=1
[142.355477] INFO: Write: size=5, data=0500240120
[142.355484] INFO: Sleep: value=200
[142.560796] INFO: Purge: direction=1
[142.561395] INFO: RTS: value=0
[143.567826] INFO: Read: size=0, data=
[143.567832] ERROR: Failed to receive the answer. [in src/suunto_vyper.c:212 (suunto_vyper_transfer)]
[143.567881] INFO: Sleep: value=500
[144.071664] INFO: RTS: value=1
[144.072860] INFO: Write: size=5, data=05001E0219
[144.072866] INFO: Sleep: value=200
[144.278065] INFO: Purge: direction=1
[144.278520] INFO: RTS: value=0
[145.284470] INFO: Read: size=0, data=
[145.284474] ERROR: Failed to receive the answer. [in src/suunto_vyper.c:212 (suunto_vyper_transfer)]

### Mentions:

glance-, 2022-08-23:
This settings dialog only support the old Vyper family of computers, and not the Vyper Novo, which is a different family, and more related to the Vyper2 and D9 than the old Vyper.

Shure, the app shouldn't hang. This code got room for improvements.

If you're interested in adding support for the D9-family, there's some hints at what's what in the memory at https://sarnau.info/pc-protocol-of-suunto-d9/

I used his reverse engineering of the Vyper family for writing the configuration support for those, that's available at https://sarnau.info/pc-protocol-of-suunto-spyder/.

tomas-kucera, 2022-08-23:
OK, thanks for the explanation. I might have a look and potentially help with some analysis of the protocol, but I am no C coder, so that is something out of my expertise. :/

tomas-kucera, 2022-08-23:
Currently I am struggling with transport protocol. Need to specify some but cannot really get it working with dctool. :( I even looked into the code but did not get anything that would lead me to success. Some more hints / help / docs would be handy. :)

tomas-kucera, 2022-08-24:
> This settings dialog only support the old Vyper family of computers, and not the Vyper Novo, which is a different family, and more related to the Vyper2 and D9 than the old Vyper.
>
> Shure, the app [...]

I have looked and tried to use the details in the links but with no real success. I used the driver to dump memory and looked for the content in the addresses but they do not seem to match.

My VPRN is V1.0.7 and IMD V1.6.

Anyone, who can advice on the memory layout of the Suunto Vyper Novo?

Thanks in advance!