subsurface/subsurface #3501: No details in information box about surface event ("yellow triangle sign") and event triangle not aligned properly

sdroege, 2022-08-29:
### Describe the issue:
- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I have a dive log from a Suunto Zoop Novo with the following content here

```xml
  <event time='0:00 min' type='25' flags='1' name='gaschange' cylinder='0' />
  <event time='0:00 min' type='9' name='surface' />
  <event time='41:05 min' type='9' name='surface' />
  <event time='42:33 min' type='9' name='surface' />
  [...]
  <sample time='41:00 min' depth='1.48 m' />
  <sample time='41:20 min' depth='1.72 m' />
  <sample time='41:40 min' depth='2.49 m' />
  <sample time='42:00 min' depth='2.34 m' />
  <sample time='42:20 min' depth='1.83 m' />
```

The second surface event is shown in the dive profile with a yellow triangle sign. According to the manual, hovering over the sign should give some details about the event in the information box but that doesn't seem to work in this case, see screenshot below:

[image]

The cursor was centered on the yellow triangle at this point. Note also that the triangle is centered at 41:40 instead of 41:05. If going to 41:05 (cursor quite a bit left of the yellow triangle), still nothing is shown:

[image]

### Operating system:

Debian sid using flatpak 1.12.7

### Subsurface version:

v5.0.9 installed via flatpak, but previous versions had the same behaviour

### Steps to reproduce:

Create a dive log as above with multiple surface events. I tried manually adding it to another dive log like this and it had the same effect.

### Current behavior:

The event yellow triangle sign is visible as seen above but at the wrong position and hovering over it does not show any additional information in the information box (at the triangle position as well as the actual time of the event).

### Expected behavior:

The information box would show that this is a surface event.

### Additional information:

### Mentions:

jefdriesen, 2022-09-05:
I'm not sure what kind of info you expect to see here. The info box clearly shows "surface" in your first screenshot (last line).

atdotde, 2022-09-06:
I tried it. I don't think it looks good:
[image]

sdroege, 2022-09-06:
I like it, it makes it clear that this is related to the event instead of some other information

atdotde, 2022-09-06:
Any other opinions, @dirkhh @bstoeger ?

dirkhh, 2022-09-06:
I don't feel strongly about it... but if we do use this, it should be consistent in size with the font, but consistent in color / look with the actual icon used...
Honestly in 'white on black' it's unlikely to be more obvious, but in yellow or red it might work better

atdotde, 2022-09-12:
[image]
Here is a version with the yellow icon (the white on black thing was a unicode warning sign, so actually text). I had to make the inbox take rich text (poor person's html). This adds some extra vertical space. What do you think? If you like it, you find it in #3520.

dirkhh, 2022-09-12:
I will of course nit pick (you expect that from me, right?) - we need a tiny bit of white space before the text. But other than that, I think it's very nice!

atdotde, 2022-09-12:
You mean like this? 
[image]

And yes, you know how much I admire your attention to detail. But I could not help to think of [this](https://dilbert.com/strip/1997-07-12)...

dirkhh, 2022-09-12:
Hmm... not sure I'm ok with that association...

Yes, this looks much better.

Also, I'm on a dive trip. I'm actually even looking at your PR. Give me some credit. And if you are honest, you'll agree that this looks much better...

atdotde, 2022-09-12:
OK ok, it does look better...

closed 2022-09-12 by atdotde