subsurface/subsurface #3516: Feature Request: Use local xml file on Android (same as PC)

dstu69, 2022-09-05:
### Describe the issue:
- [x ] New feature request

### Issue long description:
Unlike on a PC (and from what I read, on iOS, too), where the divelog can be saved to a known file (location), on the android app, the XML file can't be located on the android app.
The request is to allow the app user to open/save a local XML file

### Operating system:

Android 12

Samsung S10e

### Subsurface version:

3.4.6 (5.0.9)

N/A

Official Release

### Steps to reproduce:

N/A

### Current behavior:

Can't locate the xml file on the exposable storage

### Expected behavior:

Ability to select which xml file to use, similarly to the way it's done in Windows

### Additional information:

### Mentions:

dirkhh, 2022-09-05:
The mobile app doesn't use an XML file. It uses a git repository that is synced to cloud storage.

The latest beta versions have added a feature that allows you to export an XML file that represents the dives in the app via email. There is no way at this point for the app to import an XML file.

closed 2022-09-05 by dirkhh

dstu69, 2022-09-05:
> The mobile app doesn't use an XML file. It uses a git repository that is synced to cloud storage.
> 
> The latest beta versions have added a feature that allows you to export an XML file that represents the dives in the app via email. There is no way at this point for the app to import an XML file.

Hi Dirk,
Thanks for your prompt response to my request.
2 things:
1. I tried to run the latest apk (5.0.9-48) on my Samsung phone, but it failed to run. FYI.
2. Using a rooted android device, I managed to view the logged dives, but as you said, it's not stored in XML format and although I was able to understand how to convert some of the data to xml, there are some fields (mostly the <event>'s fields and <samples>' air, ndl, temperature and cns that I don't know what it is ) that I didn't succeed to properly translate. See attached source dive and xml in progress
Can you refer me to the necessary code to identify how to convert a computer dive entry into a <sample> line?
Regards,
David
[Divecomputer.txt](https://github.com/subsurface/subsurface/files/9490525/Divecomputer.txt)
[Dive-1.txt](https://github.com/subsurface/subsurface/files/9490526/Dive-1.txt)

glance-, 2022-09-05:
Copy / clone / push the whole git repo to a computer running subsurface desktop, open the repo in the form of /path/to/git/repo/[branch] and then import the xml file into subsurface and save it back to the git repo, and copy / push the git repo back to the phone.

I sometimes use a rooted device and the pre-built git binaries from Termux to run a git daemon on the rooted device serving the subsurface-mobile git repo to my local wifi network to/from which I can push/pull dive data.

If you're of the hacking type you can modify the source and build a subsurface-mobile which doesn't use the git-repo as storage but rather a xml file in ex. /sdcard/. But if you do, and you break things, you get to keep all the pieces.

dirkhh, 2022-09-06:
> 1. I tried to run the latest apk (5.0.9-48) on my Samsung phone, but it failed to run. FYI.

You are focusing on the wrong problem

Instead of weird hacks on rooted devices... let's figure out why that latest app doesn't work.
How are you trying to install this?
What's the error message?

dstu69, 2022-09-06:
> > 1. I tried to run the latest apk (5.0.9-48) on my Samsung phone, but it failed to run. FYI.
> 
> You are focusing on the wrong problem
> 
> Instead of weird hacks on rooted devices... let's figure out why that latest app doesn't work. How are you trying to install this? What's the error message?

I downloaded the apk (5.0.9-48-gd306...) from the test versions folder and when I try to install (permission of unknown source enabled), it says "There was a problem while parsing the package".

I also tried on my tablet (also Samsung) and got the exact same error

dirkhh, 2022-09-06:
Yeah, the CI/CD created APKs aren't signed, so they are just for playing in a simulator.
Please join the 'beta' program on Google Play (just at the bottom of the regular Subsurface-mobile page in the app) and you'll be offered a newer version that has been signed with the production keys and that will install on top of an existing version.
Sometimes it takes a few hours for Google to offer the beta version once you join, so please be patient.
In that version which currently should be 3.4.7(5.0.9.35) you'll be able to go to Dive management -> Export and there select Export Subsurface XML and once you tap Next it will open the Android share screen that allows you to send this as an email.

dstu69, 2022-09-06:
> Yeah, the CI/CD created APKs aren't signed, so they are just for playing in a simulator. Please join the 'beta' program on Google Play (just at the bottom of the regular Subsurface-mobile page in [...]

Thanks a lot Dirk,
I followed your instructions and managed to copy the xml (ssrf) to the PC and merged the dives into my XML file.
It would be great to be able to import this XML back into the mobile app one day to allow both devices to have all the information (dives, sites etc)
Again, thanks for your help and amazing app!!

dirkhh, 2022-09-06:
That's what the Subsurface cloud is for. It manages all that for you. Managing edits on different devices with XML is a major pain and intentionally not supported