subsurface/subsurface #3518: Add Apple Watch Ultra support for the new diving watch

mswdev, 2022-09-07:
### Describe the issue: I would like to see support added for the new Apple Watch Ultra watch for the new diving features.
https://www.apple.com/apple-watch-ultra/

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

N/A

### Operating system:

N/A

### Subsurface version:

N/A

### Steps to reproduce:

N/A

### Current behavior:

N/A

### Expected behavior:

N/A

### Additional information:

N/A

### Mentions:

N/A

dirkhh, 2022-09-08:
As if there was a snowballs chance in hell that Apple will provide information on how to download from their watch.

closed 2022-09-08 by dirkhh

mswdev, 2022-09-08:
> As if there was a snowballs chance in hell that Apple will provide information on how to download from their watch.

Don't be so quick to make assumptions and give it some more time for information to be released. Apple stated that third-party dive apps can be made to support the dive functions and that Oceanic+ would just be the first app. 

This would mean that it is likely Subsurface could integrate functionality into the watch for the iOS app and potentially more...

From Apples site:
"Apple Watch Ultra may be used for recreational scuba diving **(with a compatible third-party app from the App Store)**."

dirkhh, 2022-09-08:
I realize that my answer was a bit harsh - I've had a few past experiences with Apple's rules for how open source software can interact with their hardware. Generally the way this works is that for a small fee (think $$$) you can buy access to the documents, and for a smaller fee per user (think $) you can then use that IP. Obviously that works if you sell an expensive app. It doesn't work for open source software.
I might be wrong. I would like to be wrong. If I am, we will of course support the Apple Watch once the documentation becomes available and one of the developers finds the time to implement support.

I wouldn't be holding my breath, if I were you... (but you shouldn't do that as a diver, anyway)

mswdev, 2022-09-08:
@dirkhh I appreciate the quick response and I definitely understand the position now.

It will be interesting to see what further information is released coming soon. It's clear that the Apple Watch Ultra is targeted towards the occasional recreation diver who only dives a handful of times and year and likely rents all of their gear.

However, it could make for a good backup dive computer especially for those who already daily use an Apple Watch like myself. I definitely understand that they often lock down many opportunities and I'm hopeful that there is opportunity here for third-parties to join in such as Subsurface, but only time will tell.

I did purchase the Apple Watch Ultra so when I get it by the end of this month I can provide more information as to the potential feasibility of this.

On a side note: Other than the Google Group, does Subsurface have a Discord or Slack for the community to improve communication within the Subsurface community? If not would you be opposed to one being created either as an official channel by Subsurface or ran by the community?

dirkhh, 2022-09-08:
The main developers of Subsurface have been around open source for some 30+ years... in other words, we are pretty old. Slack and Discord and other things the twenty-somethings are in love of tend to mostly annoy me.
We don't need more ways for people to talk to us... we need more people willing to invest time into the development of the software :)
That said, you are welcome to create whatever channels you want - as long as you don't expect the developers to be in those channels...

mswdev, 2022-09-23:
@dirkhh Just want to keep this issue ticket open as more information comes available.

Apple has added more information regarding the Apple Watch Ultra and accessing its dive capabilities through the API.

The Apple WatchOS9 documentation page now states at the bottom of the page that "Details on how to access submerged depth and pressure APIs in the watchOS SDK will be available later this year."

It's still fairly vague on the details and up in the air but is looking more promising that this could be an opportunity without having to pay or not having the ability to add support at all.

It seems like Apple is gatekeeping the documentation until the Oceanic+ app is released to give them a head start over anyone else wanting to make a scuba/dive app for the watch. According to Oceanic+ their app for the Ultra Watch will not come out until "later this fall".

Apple is integrating the Apple Watch Ultra with their own "Depth" app, but this app does not provide full dive computer capabilities and it's clear they're trying on Oceanic+ and others to make their own full dive computer apps for the Watch.

[image]

References:
https://developer.apple.com/watchos/
https://support.apple.com/en-us/HT213334

Nesslax, 2024-01-12:
you can reopen this ticket there are some news here : https://developer.apple.com/documentation/coremotion/accessing_submersion_data/

mswdev, 2024-01-12:
> you can reopen this ticket there are some news here : https://developer.apple.com/documentation/coremotion/accessing_submersion_data/
> 
> 

Incredible! I knew it would be coming. Would love to see support added for Subsurface. Would be a good backup to my primary dive computer.

djunkins, 2024-01-12:
It looks like the entitlement widely available now only supports submersion data down to 6 meters (20 feet). There is still the same form there has been for almost a year to express interest in the full entitlement to access data down to 40 meters (130 feet). I filled that form out last year for my AWU2UDDF app that can be used to export dive data from the Apple Watch to be imported into Subsurface (and other dive log software), but never heard back from Apple. My app gets access to the data post dive through the HealthKit interface, but does not have realtime access during a dive.

-Doug