subsurface/subsurface #3519: Support for Garmin Descent MK2i directly through the SubSurface app

mswdev, 2022-09-08:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:
I noticed that the Subsurface app does not include direct support through the app for the Garmin Descent MK2i or any other Garmin dive watches such as the G2.

I tried researching Subsurface FAQ and other issue requests to see what the reason on why this was and only saw comments that it was not supported and there was not interest in doing so.

I have not been able to find an answer on why this is the case whether it is a personal choice or if there is an actual limitation in being able to connect to the Garmin Descent via the Subsurface app. Could we please get a more technical answer on why support will not or cannot be added directly into the Subsurface app?

### Operating system:

N/A

### Subsurface version:

N/A

### Steps to reproduce:

N/A

### Current behavior:

N/A

### Expected behavior:

N/A

### Additional information:

N/A

### Mentions:

N/A

dirkhh, 2022-09-08:
Garmin explicitly requested that we do not connect to the watch via Bluetooth as their protocol talking to their backend service could get confused by that and they told me that they were worried that there was a risk of data loss for the user - which is something that I of course don't want to do / be accused of causing.
So that's why we don't support the Garmin dive computer via BLE on iOS and Android.
for the Mk1 there is indeed a way to access files on a plugged in device from an app and we could conceivably add this; the APIs for that have changed dramatically in the last couple of Android releases and I am under the impression that meeting the Android requirements for an app to be allowed to do this would be very hard for us. So we decided not to implement that. But that could of course be done.
The Mk2 uses MTP to talk to the dive computer - I haven't found a library that works on Android that supports that protocol, so there wer are out of luck, I think.
I hope that answers your question.

The simple solution is to connect a computer and your phone via the Subsurface cloud and simply download on the computer. I have a Mk2i with me on every dive (love the GPS coordinates) and keeping things in sync between computer and phone seems reasonably easy to me.

mswdev, 2022-09-08:
@dirkhh I appreciate the response, that provides a lot more clarification!

Interesting stance by Garmin on how their backend service could have issues from that lol. I hope they improve whatever protocols their using in the future so an easier and more seamless integration could be made.

Attaching the device to the PC every time I want to sync data will be a slight annoyance, but I'm definitely still willing to do it so I can use Subsurface!

closed 2022-10-03 by dirkhh