subsurface/subsurface #3532: v5.0.10 terminates after selecting a few dives with "manual" profiles

riaanmarx, 2022-10-06:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I imported some dives using time/depth data exported from another application.
After opening Subsurface, I am able to click on, and view the dive info of these dives, but after clicking on a few of them (even ones that was displayed before), the application would just terminate with no error message.

In Windows event viewer, the following is logged in an application error log event:
Faulting application name: subsurface.exe, version: 0.0.0.0, time stamp: 0x00000000
Faulting module name: subsurface.exe, version: 0.0.0.0, time stamp: 0x00000000
Exception code: 0xc0000005
Fault offset: 0x00000000000ecdf3
Faulting process id: 0x52b0
Faulting application start time: 0x01d8d984ac8feddd
Faulting application path: C:\Program Files\Subsurface\subsurface.exe
Faulting module path: C:\Program Files\Subsurface\subsurface.exe
Report Id: 3eb0d21c-03ff-42cb-868e-a249ca4ef5b0
Faulting package full name: 
Faulting package-relative application ID: 

### Operating system:
Microsoft Windows 11 Enterprise [Version 10.0.22000.978] / Build 22000
HP ZBook 15 G6 (Bang & Olufsen) laptop

### Subsurface version:

v5.0.10

I did not notice it happening on v5.0.8

Official release for Win x64

### Steps to reproduce:

1) Open Subsurface
2) Allow Subsurface to load the default file (my dives)
3) repeatedly click on dives that were imported from raw time/depth data

### Current behavior:

The application terminates (closes) with no error.

### Expected behavior:

The application should display the dive information of the selected dive.

### Additional information:

### Mentions:

[errorlogfile.zip](https://github.com/subsurface/subsurface/files/9726983/errorlogfile.zip)

riaanmarx, 2022-10-06:
> This message is from an EXTERNAL SENDER - be CAUTIOUS, particularly with links and attachments.
>
> I think this is an access error. Hmm. I tried to reproduce it here locally on Linux and can't.
> Need to try on Windows as that often behaves differently when it comes to memory errors.
> Thanks for the report!
>
>
>
> This message is for the designated recipient only and may contain privileged, proprietary, or otherwise confidential information. If you have received it in error, please notify the sender immediately and delete the original. Any other use of the e-mail by you is prohibited. Where allowed by local law, electronic communications with Accenture and its affiliates, including e-mail and instant messaging (including content), may be scanned by our systems for the purposes of information security and assessment of internal compliance with Accenture policy. Your privacy is important to us. Accenture uses your personal data only in compliance with data protection laws. For further information on how Accenture processes your personal data, please see our privacy statement at https://www.accenture.com/us-en/privacy-policy.
>
> www.accenture.com

I agree Dirk.
From my development experience that  0xc0000005 error code means “access violation”, which normally points to buffer overruns or simply addressing memory that was not allocated for the process.

Just a correction on the error report: I have uninstalled v5.0.10, and reinstalled v5.0.8 to verify and although v5.0.8 takes much longer to do so, it actually did terminate in the same way.

Kind regards,

riaanmarx, 2022-10-08:
Is there a way to do an exception dump maybe, to help identify the track trace or something like that? (a debug build maybe or something with debug symbols or logging)?
My development experience is mostly (about 20 years) with C#, and although I have some exposure to C++, I am not comfortable in it.

bstoeger, 2022-10-20:
Could you provide a (minimal) log that shows the problem for you? I don't see anything with valgrind.

bstoeger, 2022-10-20:
Never mind, I see that you attached an example.

bstoeger, 2022-10-20:
I get this, but only once:
```
==64096==    at 0x405973: DivePlannerPointsModel::loadFromDive(dive*) (diveplannermodel.cpp:152)
==64096==    by 0x361233: ProfileWidget::editDive() (profilewidget.cpp:261)
==64096==    by 0x36145B: ProfileWidget::plotCurrentDive() (profilewidget.cpp:196)
==64096==    by 0x34C96E: MainWindow::selectionChanged() (mainwindow.cpp:333)
==64096==    by 0x99C2792: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3)
==64096==    by 0x2E9143: DiveListView::diveSelectionChanged(QVector<QModelIndex> const&) (divelistview.cpp:262)
==64096==    by 0x99C2792: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3)
==64096==    by 0x3F0B65: MultiFilterSortModel::selectionChanged(QVector<QModelIndex> const&) (moc_filtermodels.cpp:162)
==64096==    by 0x429695: MultiFilterSortModel::selectionChangedSlot(QVector<QModelIndex> const&) (filtermodels.cpp:45)
==64096==    by 0x99C2792: ??? (in /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3)
==64096==    by 0x3F0B15: DiveTripModelBase::selectionChanged(QVector<QModelIndex> const&) (moc_divetripmodel.cpp:158)
==64096==    by 0x414549: DiveTripModelTree::divesSelected(QVector<dive*> const&) (divetripmodel.cpp:1395)
==64096==  Address 0x1f89a184 is 28 bytes before a block of size 103,568 in arena "client"
```

How do you even import these dives?

riaanmarx, 2022-10-20:
I think these ones may have been "injected" into the SSRF file using a C# application I wrote.
The data came from 3 separate logging applications....some of which were still DOS C apps....don't laugh....
I had to decode the binary data into text, but thought it would be better to inject the data directly into the SSRF file as XML in order to also include the warnings and alarms.

The issue may very well stem from my XML not being exactly as expected. I matched existing (uploaded) SSRF profile data as close as possible and had not noticed any issues in the version of SubSurface I used at the time, so I thought I would be OK...assumptions...
Us developers always try to push the envelope of the software we use I guess.

riaanmarx, 2022-10-20:
One item I noticed on another issue I logged was that if I changed the computer model from "manually added dive" to "Manually added dive", the profile would display normal as expected, whereas previously it displayed as if the dive was being planned. The planning vs display mode detection of the profile component may have changed at some point and not it thinks I am still planning, but loading the profile as a "plan" may be causing the symptom we experience here.

I changed the XML in the SSRF file to all point to the correct computer or "Manually added dives" which also seems to solve this issue for me so I suspect the two are linked.

riaanmarx, 2022-10-20:
you can refer to "Bookmarks in CSV-imported profiles not possible/displayed #3533" issue for screenshots of the profiles mentioned above.

bstoeger, 2022-10-20:
Yes, we removed the "edit mode" and manually added dives are now always shown in edit mode.

I will look into this, when I find time.