subsurface/subsurface #3533: Bookmarks in CSV-imported profiles not possible/displayed

riaanmarx, 2022-10-07:
### Describe the issue:

- [?] Bug
- [x] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

Please enable the adding and display of bookmarks on CSV-imported dives similar to how it is added/displayed in dives uploaded from dive computers.

### Operating system:

Windows 11 Enterprise

HP ZBook 15 laptop

### Subsurface version:

5.0.10

Official release

### Steps to reproduce:

Import a dive profile from CSV with depth/time points.
Select the imported dive
Right-click on the profile image

### Current behavior:

Nothing happens
If a bookmark is added manually in the ssrf XML file, it is ignored (not displayed)

### Expected behavior:

Context menu popup to enable adding bookmark.
Bookmarks added to the profile to be displayed in the profile as it is done on profiles imported from dive computers.

### Additional information:

### Mentions:

riaanmarx, 2022-10-07:
I have many profiles from previous computers/software packages exported to CSV and other text formats. These contain warnings from the DCs, and I would like to see those on the profile. I thought bookmarks would be the perfect way, but they are not displayed on these imported dives.

dirkhh, 2022-10-07:
This seems to describe two issues / requests
- ability to import bookmarks / events via CSV (so how would that even work? a special column that says "bookmark here"?) @mturkia 
- ability to add bookmarks in the profile (that should work and if it doesn't feels like a bug)

Can you @riaanmarx share one of those imported dives with warnings from the DC that aren't displayed, just so I have a better idea of what's going wrong?

mturkia, 2022-10-08:
I would assume that it would be a special bookmark column that we would need to implement. I have not yet seen any csv export that would include bookmarks, so it would be good to have such sample to know for sure how it is done.

riaanmarx, 2022-10-08:
Good day everyone.
[errorlogfile.zip](https://github.com/subsurface/subsurface/files/9738839/errorlogfile.zip)
Please find attached an SSRF file with 2 profiles - both containing events/alarms.

riaanmarx, 2022-10-08:
Having the ability to import the events / alarms with the CSV would be a bonus.
Currently I just added them as bookmarks/events in the ssrf XML file using a C# application I wrote.

riaanmarx, 2022-10-08:
I took some screenshots for you from v5.0.5 (which displayed the warnings correctly)
[image]

riaanmarx, 2022-10-08:
and here the same dive with v5.0.10
[image]

riaanmarx, 2022-10-08:
I also just noticed that it does not draw the temperature graph either for the manual dives in v5.0.10, but it does for the dives directly uploaded from the dive computer.

riaanmarx, 2022-10-10:
If I manipulate the SSRF file to alter the <divecomputer> tag to same one used on a later dive, the profile is displayed as in v5.0.5
[image]

riaanmarx, 2022-10-11:
It almost seems as if the profile component was thinking the application is in DivePlanning mode and was showing the dive edit profile instead of the display dive profile?

mturkia, 2022-10-11:
Yes, the profile with a lot of circles on the depth line looks like planning mode.

Do you have a csv dive from another computer with events that you would like to have imported?

riaanmarx, 2022-10-11:
Unfortunately not. The original source of the data was 1) a binary encoded file I decode manually and 2) a SQLite database from another log application. I wrote some code in C# to extract the data from these sources to CSV and to XML directly.

Since logging the issue I have updated the ssrf file's XML to use the actual dive computer models in the model attribute of the divecomputer nodes and that seems to have solved the problem now. All the profiles now show as expected with the temperature and events, and I can add manual bookmarks as in v5.0.5. I suspect v5.0.10 contains an update than checks for something in the XML to decide weather the dive is to displayed as planning vs display mode, and my ssrf file simply had to be modified to fix that issue.

Is there an XML schema document for the ssrf file I could use to ensure I have not caused any other unforeseen symptoms by editing the ssrf XML directly?

bstoeger, 2022-10-20:
Yes, the thing is that your dives are by definition not manually added dives, but dive-computer downloaded dives.

We probably should allow bookmarks, events, etc. for manual dives, so I think this report is valid. Nevertheless, I suggest you adapt your script.

riaanmarx, 2022-10-21:
Agreed - that makes sense. The profile import was a once-off process when I migrated to SubSurface, but I have amended the profiles accordingly.