subsurface/subsurface #3538: Update not working / Windows software control panel shows strange entry

sjvudp, 2022-10-18:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

After updating Subsurface 5.01 to 5.0.10 I noticed that in the Windows Control Panel the Software is listed with install date in 2020 (2020-11-11) and as version 4.6.4.
So I thought the update isn't working well and I uninstalled the version, and then reinstalled it.
However I still see that install date as 2020-11-11 and version as 4.6.4.

### Operating system:

Windows 10 21H2

### Subsurface version:

5.0.10

Official Release

### Steps to reproduce:

### Current behavior:

[image]

### Expected behavior:

The update (installation) should set the entries in Windows Control Panel correctly.

### Additional information:

### Mentions:

dirkhh, 2022-10-18:
I can't reproduce that here.
[image]

As always with Windows, I'd try rebooting and checking again...

PeterZaal, 2022-10-19:
Can't reproduce either:

[image]

sjvudp, 2022-10-29:
Well, I did not only reboot, but also installed the latest Windows updates. Subsurface is still displayed as 4.6.4.
Unfortunately I have no experience regarding where that information comes form.

[image]

Digging in the registry editor I found:
`Computer\HKEY_USERS\S-1-5-21-3652416834-1906715085-2507927773-1000\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall\Subsurface`
`DisplayVersion` is set to `4.6.4` there.

mikeller, 2026-10-17:
Hi @sjvudp,

Thanks for flagging the strange entry in Windows' software/Control Panel listing. Could you confirm whether you're still seeing this on a current install? This looks like it may be a one-off installer/registry artefact from a particular update sequence rather than a general issue, so knowing if it persists on a fresh install would help us tell the difference.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.

sjvudp, 2026-11-15:
I'm really sorry for the late response:
I had updated to 6.0.5576, but the entry still wasn't updated. Then I updated to 6.0.5738, but still the entry looks unchanged:

[image]

My guess is that there should be some entry under `Computer\HKEY_LOCAL_MACHINE\SOFTWARE`, but I couldn't find a related one.
However I found this one:
```
[HKEY_CURRENT_USER\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall\Subsurface]
"DisplayName"="Subsurface"
"DisplayIcon"="C:\\Program Files (x86)\\Subsurface\\subsurface.ico"
"UninstallString"="C:\\Program Files (x86)\\Subsurface\\Uninstall.exe"
"DisplayVersion"="4.6.4
```
If I look at that `Uninstall.exe`, the date is current, however.

mikeller, 2026-11-15:
AI-drafted; reviewed by Michael.

Hi @sjvudp, thanks for checking again, and no need to apologise. The unchanged `4.6.4` registry value confirms that Windows is still reading an old installation entry. It does not, by itself, tell us whether the Subsurface program you launch was updated.

Could you check **Help → About** in the Subsurface window you normally use and tell us the version it shows? If it shows `6.0.5738`, please also right-click its running process in Windows Task Manager, choose **Open file location**, and tell us the folder path. That will help us distinguish a stale Windows listing from an installation that is still launching the old program.

I checked the installer: it writes the Windows listing under the current Windows user's registry entry. The installation path you found is under `Program Files (x86)`, while the current installer's default is the 64-bit Program Files folder. A second installation or an installer run under another administrator account is therefore possible, but we need the running version and path before concluding that. Please don't delete the registry entry or uninstall anything yet; we'll investigate the installer behaviour once we've identified which installation is running.

sjvudp, 2026-11-15:
> Could you check Help → About in the Subsurface window you normally use and tell us the version it shows?

I had checked it: It's the current version. Also it had been installed in the "normal" path (I don't run Windows at the moment, but Linux). As can be seen in feature request #5133 the installer tried to update the installation in `Program Files (x86)` directory; so I guess that's where the binaries are.
My guess is that some old installer left information in `HKEY_CURRENT_USER`, while a newer installer *might* have left information in `HKEY_LOCAL_MACHINE`, and Windows prefers information from `HKEY_CURRENT_USER`.
the only thing is that I could not find installer information in `HKEY_LOCAL_MACHINE`.

mikeller, 2026-11-15:
AI-drafted; reviewed by Michael.

Hi @sjvudp, thanks for the detailed follow-up, and for pointing to #5133.

I compared the installer script across versions. The old 4.6.4 installer used the same registry key (`HKCU\Software\Microsoft\Windows\CurrentVersion\Uninstall\Subsurface`) and the same remembered install-folder key (`HKCU\Software\Subsurface`) that the current one does. It differs in the default folder: 4.6.4 was a 32-bit installer, so it chose `Program Files (x86)`. Current installers use the 64-bit `Program Files`, but they first look up the folder remembered in `HKCU\Software\Subsurface`, and that value takes priority. This is why updates have kept going into `Program Files (x86)`. It also fits what you saw in #5133, and it is a flaw in the installer that we'll look into.

What I can't explain from the scripts is why the Windows list still shows `4.6.4`. Every install writes that same entry, and the uninstaller removes it. So a reinstall should have replaced the old value. An old key name or an old folder doesn't account for it. I also checked that `HKEY_CURRENT_USER\Software` isn't split into separate 32-bit and 64-bit views, so that isn't the cause either. Your finding that nothing was written under `HKEY_LOCAL_MACHINE` is consistent with the installer, which only uses `HKEY_CURRENT_USER`.

One possible cause is that the installer ran elevated under a different Windows account from the one you normally use, so it updated that account's registry rather than yours. That is only a guess, and I don't know whether it applies to you. If it's easy to check, please let me know whether you sign in as a standard user and enter separate administrator credentials when the installer asks. Don't go out of your way if you no longer have access to the Windows machine.

The Subsurface program itself is the current version, so your logbook and installed software aren't affected by this. Please ignore the `4.6.4` label and avoid editing the registry by hand for now.

sjvudp, 2026-11-16:
> One possible cause is that the installer ran elevated under a different Windows account from the one you normally use, so it updated that account's registry rather than yours. 

Well I had a similar idea presented that the registry entry found is "shadowing" another one, so I made two tests (I did configure two users, an "admin" and a "normal" one):

1. Log in as admin and list the programs. There I saw: [image]
2. Switching back to the normal user, then delete the registry key `[HKEY_CURRENT_USER\SOFTWARE\Microsoft\Windows\CurrentVersion\Uninstall\Subsurface]`. After that no "Subsurface" was shown as installed an more

What this may mean: It seems the installer handles the install as "user local", but still uses the "shared" program files directory.
(AFAIK "per-user" installs should be installed somewhere within `%USERPROFILE%` instead).
Agreed, I know more programs that didn't get it right initially, so a program icon had been installed on admin's desktop only, and the normal user could not find it (maybe it should have been installed inside `C:\Users\Default\Desktop` instead), but most of those apps had fixed their installers meanwhile (I install all my programs that way for at least 10 years).

The current installer indicates that it wants "admin rights" ([image])
(maybe there were versions in the past that did not request that, so the registry ended up in the normal user, but I don't really know).

Still, my other programs show up for the normal user as well as the admin user (just see the screenshots already provided), so I guess Subsurface "does something differently".

Maybe peek at the installers of some opensource like 7-zip, Audacity, GIMP, HandBrake, Hugin, Firefox, VLC media player, WinMerge, Zotero (just to name a few where installation works as expected).

mikeller, 2026-11-16:
AI-drafted; reviewed by Michael.

Hi @sjvudp, thank you for running those two tests. They help a lot, and your observation that Subsurface "does something differently" looks right.

Here is how I read it, so please correct me if it doesn't match what you saw. The installer is machine-wide for the files, the shortcuts and the install folder. But it writes the Apps & Features entry, and the remembered install folder, under `HKEY_CURRENT_USER` (see the [installer script](https://github.com/subsurface/subsurface/blob/master/packaging/windows/subsurface.nsi.in)). When you elevate with separate administrator credentials, that is the **administrator's** registry, not your normal user's. Each update would then refresh the admin account's entry, while your normal account kept the old 4.6.4 one. That fits your results, but it is an inference from the script and your tests, and I haven't reproduced it on your machine.

I've opened a draft pull request for this: https://github.com/subsurface/subsurface/pull/5138. It writes the entry and install location machine-wide, and removes old per-user entries: the elevating account's, and those of other users who are logged on while the installer or uninstaller runs. It is an **unmerged draft, not a release**, and so far it has only been compiled, never run on Windows, so it needs testing there before anyone should rely on it. There is no test installer to download yet. If CI produces one, I'll point you to it, but only try it if you still have a Windows machine handy.

Two limits of that draft: it does not move an existing `Program Files (x86)` installation, so that location would still be reused. And stale entries of users who are logged off are left alone.

Your logbook and the installed program are not affected, as this only concerns how Windows lists the installation.

Thanks also for the pointers to other installers; they are a useful reference.