subsurface/subsurface #3539: wrong date/missing dives in statistics plots

KimDelmar, 2022-10-19:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Two problems in the Dive Statistics window, with Base Variable specifically set to Date/no binning:
a) when selecting a subset of dives to plot, the last two dives of the series are not plotted (if >=5 dives are selected.
    If <=4 dives are selected none of the dives are plotted.   Everything works OK if any kind of date binning is applied.
b) Date axis has the month off-by one (earlier), eg for dives on 18-Oct axis label is "18 9"

### Operating system:
MacOS Monterey 12.6, MBP M1

### Subsurface version:
5.0.10 Qt5 official

### Steps to reproduce:
1) Select 10 dives from a dive series
2) View -> Dive Statistics, Choose base variable: Date, and Binning: none. Choose "Data:" Max Depth
3) set Restriction: Restrict to Selection:  only 8 dives are plotted; last two are missing
4) close statistics window completely and select 4 dives, and repeat steps 1-3. Nothing is plotted.
5) Observe X axis - displayed month is one less than actual dive date. eg 18-Oct displayed as "18 9"

### Current behavior:
a) dives are missing from plots when viewing selected/subsets of dives,
b) date axis "month" field is incorrect

### Expected behavior:
All selected dives should be plotted. Date axis should match actual dive date.

### Additional information:
n/a

### Mentions:
n/a

bstoeger, 2022-10-19:
Thank you for the report. This are two different bugs, so in general two reports would make things easier, but no problem.

I think I understand the first issue: The end of the axis is set at the start of the last day, right? So the dives from the last day are not displayed?

The second thing is hopefully fixed by #3540.

For the record: While looking at this, I found a third bug: When having dives selected and switching to "no binning" mode, the selected dives are shown in blue (instead of yellow for selected dives). Only when hovering over them do they get the correct color.

bstoeger, 2022-10-19:
First problem now hopefully also fixed (or at leas "patched") in #3540.

bstoeger, 2022-10-19:
Third bug now also hopefully fixed.

KimDelmar, 2022-10-20:
Thanks for the fixes (and the bonus fix!)  Confirmed that with the latest build the date axis reads correctly now, and the dives in the last "bin" are now displayed.

However I still see the one remaining  issue where if 4 or fewer dives are selected, then when restricting to the selected dives, nothing is displayed.

bstoeger, 2022-10-20:
I see - I will check that out later. Seems like one needs at least three days for the axis to display correctly. Abusing the histogram axis was obviously a bad idea.

closed 2022-10-21 by dirkhh

KimDelmar, 2022-10-21:
Thanks Berthold - confirming that everything reported in this ticket is now works for me.