subsurface/subsurface #3548: Cosmiq+ import shows wrong metrics

mabakerm, 2022-11-02:
### Describe the issue:

- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Since a couple of weeks, I get wrong imports when downloading my Cosmiq+ data into subsurface.
Depths of 642 meters are showing. Tempratures of 6500C

### Operating system:

Deeplblue Cosmiq+
Windows 11 and iphone 8

### Subsurface version:

Offical version 5.0.10

### Steps to reproduce:

Import of any new dives. Tried different times with different dives

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

jefdriesen, 2023-01-09:
I'll need a download log to investigate. Simply download dives from your dive computer with both the "Save libdivecomputer logfile" and "Force download of all dives" checkboxes enabled. After the download you may cancel without importing any dives to avoid re-importing previously downloaded dives. Attach the resulting log file to the bug report.

blue-notes-robot, 2025-07-07:
Hey, got the same issue. My last 27 dives suddenly show 643m and temperature of 6553 Celsius. These dives are showing correctly on the dive computer itself and older dives did sync corectly!
Attached is my libdivecomputer logfile.

[subsurface.log](https://github.com/user-attachments/files/21106848/subsurface.log)

blue-notes-robot, 2025-07-08:
I can! My last dive was max depth 13.2m and the temperature was 20.3°C. Subsurface shows 642m and 6553.5°C

blue-notes-robot, 2025-07-17:
Hey, just tried the fix, here are the observations:
1. Previously corrupted dives now have a dive profile in V-shape (probably just interpolation to max depth) to the correct depth. That aligns with the basic info you mentioned earlier
2. Previously corrupted dives now have the temperature data missing entirely (probably was also all garbage)

I wonder why the device seems to send garbage. Memory corruption?
Anyways, it seems that is the best we can do with the available data, so thanks for that!