subsurface/subsurface #3549: Use password text input on iOS login screen

Hillshum, 2022-11-05:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

On iOS, the opening screen asks for cloud credentials using what appear to be standard text fields. As a result, the password is not obscured (which I suspect might be an intended feature), but the system also does not offer to autofill.

### Operating system:

iOS 16.0.3, iPhone 14 Pro

### Subsurface version:
3.4.7(5.0.9.33), installed from App Store

### Steps to reproduce:
1. Open app for first time

### Current behavior:
Password is displayed to user and no offer to autofill

### Expected behavior:
Password hidden (possibly with an option to reveal it) and system offers to autofill password.

### Additional information:

### Mentions:

notionparallax, 2026-11-12:
This was addressed by #5043 by @mikeller (merged 2026-10-19). The cloud-login password field now uses TextInput.Password, so it is masked and reported to iOS as a secure text entry, which enables Password AutoFill.
I don't think the issue was linked to the PR, so the issue has stayed open. Seems closable?

dirkhh, 2026-11-12:
I am sad that this is the decision that was made. Password input on iOS has been one of the many batshit stupid mis-features. Among the many reasons why I despise iOS and only use it under duress.
Anyway - when I initially implemented this that was indeed an intentional design choice.
But clearly the time where that matters has passed.

closed 2026-11-12 by dirkhh

mikeller, 2026-11-12:
> I am sad that this is the decision that was made. Password input on iOS has been one of the many batshit stupid mis-features. Among the many reasons why I despise iOS and only use it under duress. [...]

@dirkhh: Frankly, this wasn't a conscious decision to remove something that was done intentionally. On the back of #4895 and a number of similar changes, my assumption was that this was another case of a field where mobile data entry wasn't optimised for the current version of Qt. At the time I could not find any indication of possible reasons to not do what seemed to be a legitimate fix, and a response to a user request for a better UX.
Happy to revert this, but also keen to know what the reasons are to not use  the default mechanism, not least of all to be able to add it as a comment in the code to stop others from 'fixing' it in the future.

dirkhh, 2026-11-12:
I find password entry on mobile keyboards without visual feedback the dumbest things ever. On my old iPad it usually takes me three or four tries to correctly enter a password. I'm sure it's a me problem 
Hence my comment that this is not something where my opinion matters. 

The current breakneck speed of AI driven development has nothing to do with the project I used to maintain. When I wake up I have several hundred emails, mostly ai tools talking to each other. I'm not interested in participating in that. That's not why I do open source. 

I'm not going to stand in the way of this. I'll go back to focusing on keeping the backend to and running and let you continue with the rest