subsurface/subsurface #3555: Cannot access logbook on mounted ntfs volume

stumpjr, 2022-11-14:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I have a dual boot System, For data exchange i use a ntfs partition that is accessible from Windows and Linux.
This desription is for the Linux side
The partion is mounted with uid and gid.
I can create, write and delete files via cli on this partition as normal user (not root).
It is also possible to create files and folders via nautilus file manager but wirh subsurface the directory where the partition is mounted is not shown.

### Operating system:

Ubuntu 22.04 
Language: German

Acer spin5 notebook

On Windows i can access this partition (represented as drive D:) and it works as expected

### Subsurface version:

### Steps to reproduce:

Windows 5.0.10
Linux 5.0.10
### Current behavior:

Cannot acces logbook on mounted NTFS Volume

### Expected behavior:

Access to Logbook files on mounted NTFS Volume

### Additional information:

### Mentions:

atdotde, 2022-11-14:
I cannot see why this should not work. Can you be a bit more specific in at which step it fails? Is there an error message?

bstoeger, 2022-11-15:
> It is also possible to create files and folders via nautilus file manager but wirh subsurface the directory where the partition is mounted is not shown.

Nautilus is a Gnome app. Do you see files with another Qt app? Where is the partition mounted?

stumpjr, 2022-11-16:
The partition is mounted on /daten as shon in the hardcopy i sent.
I installed the QT-Creator to have another QT-App, started QT-Creator and went to "File->Open" dialog, here i can Navigate to "/daten" and see the files as well as the subdirectory "SubSurfaceLogbuch"
See appended Hardcopy
[image]

atdotde, 2022-11-16:
It would be quite unusual to mount it into the root directory (i.e. /daten) and there is no indication the the Nautilus screenshot that it actually is. A more usual place would for example be /mnt/daten

Could you please open a terminal an show the output of the "df" command?

stumpjr, 2022-11-16:
I tried different mount points all with the same result.
Here is the result of the df command.
[image]

atdotde, 2022-11-16:
And is there an error message if you do 

subsurface /daten/SubSurfaceLogbuch/LogbuchJuergen.ssrf

?

stumpjr, 2022-11-16:
This are the messages i got. I brought the Terminal to the front for complete view

[image]

stumpjr, 2022-11-16:
I knew you mention that :-)
I did the cat you wanted, after that i made a working copy also in /daten/SubSurfaceLogbuch, just to show you that read write access is given. I opened the working copy via gvim
[image]

atdotde, 2022-11-16:
And then subsurface opens the local copy I presume...

stumpjr, 2022-11-16:
No, the working copy is also on /daten/SubSurfaceLogbuch. And the behavior on subsurface is the same. I made this copy just to make sure i do not damage anything when opening the file via a normal editor .
The "cat" has been done on the original file like you wanted to see it.

bstoeger, 2022-11-16:
How did you install Subsurface?

This could be some virtualization problem. For example, when Ubuntu switched Chromium to Snap, I couldn't access some of my files anymore, because they were in a directory which was deemed "unsafe". That was very annoying and thankfully they fixed it after a year.

stumpjr, 2022-11-16:
It is a snap install
which subsurface gives 
/snap/bin/subsurface

bstoeger, 2022-11-16:
Ok, then some misconfiguration(?) of the Snap package would be my guess. Sadly, I don't know the slightest thing about virtualization (except that it tends to break things).

@dirkhh?

dirkhh, 2022-11-16:
Yeah, the Snap doesn't have access to that volume. I'm not sure how to add that to the Snap configuration (I don't use them myself). So either ask in a Snap related forum - or install our native version of the app for Ubuntu...
Sorry about that.

stumpjr, 2022-11-16:
I tried to compile subsurface on my own but failed miserably with this error.
Project ERROR: Unknown module(s) in QT: location positioning
So i downloaded the Appimage from the website 
1. try copied the AppImage to /daten and started it from there -> works! AND /daten is shown in the file->openLogbook 
2. try copied the AppImage to my homedir -> works also AND /daten is shown in the file->openLogbook 
So i think this has something to do with the snap

bstoeger, 2022-11-16:
There is also a regular native Ubuntu package:

https://launchpad.net/~subsurface/+archive/ubuntu/subsurface

stumpjr, 2022-11-16:
Yes this works !
Thank you so much for the support to all of you.
And also many thanks to the developers of the program. I know that you Dirk are one of them.
A lot of my diving friends use subsurface it is extraordinary good

closed 2022-11-16 by stumpjr

atdotde, 2022-11-16:
Just for completeness: If you google "snap ntfs" you will find that this is a known problem. You can either try to mount the partition into your home directory or do something along the lines of 

sudo snap connect subsurface:removable-media

but none of this seems to solve the problems universally.