subsurface/subsurface #3560: Shift dive time does not work

hhmeyer, 2022-11-21:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Shift dive time does not work. The input of the time offset does not take keyboard input.

### Operating system:

macOS 13.0.1 Ventura

iMac

### Subsurface version:

5.0.10

Can't say.

official relase

### Steps to reproduce:

1) Select a dive with the right mouse button and choose "Shift dive times" from the context menue
2) A window (that annoyingly can't be moved) is shown stating Current time: 0:0, Shifted time: 0:0
[image]
3) Try to enter a time difference

The input field does not take any input from the keyboard. The up/down arrows only influence the minutes. Changing the time to one minute earlier using the down arrow sets the shifted time to Thu, Jan 1, 1970 00:59.
[image]

### Current behavior:

Feature is unusable.

### Expected behavior:

Time of selected dive is shifted by a given value.

### Additional information:

### Mentions:

jwobser, 2022-12-29:
I was able to reproduce the Jan 1 1970 UI error on v5.0.10 / MacOS 13.0.1. 
However, upon clicking 'OK' the dive times are shifted appropriately in the dive list, so the error only seems to be in displaying the time in the dialog.

[I suspect the issue lies with updating the UI from the time string](https://github.com/subsurface/subsurface/blob/31277daa71b2ed7312672bbfa554a234c9ee5bdc/desktop-widgets/simplewidgets.cpp#L105) but I'm not familiar with Qt to know why it is not working correctly on Mac.

I'm also unable to compile from source on an M1 mac mini; Qt is trying to link using x86_64 framework libraries.

jwobser, 2023-01-01:
Looks like Windows is also impacted. #3359 

@bstoeger  @dirkhh  I believe `first_selected_dive()` is returning a null pointer, as this `if` block to initialize the dialog strings never executes. This leaves `when` initialized to `0` and when `ShiftTimesDialog::changeTime()` executes it is only using the `amount` offset, hence returning 1970 JAN 1

```C++
if (d) {
		when = d->when;
		ui.currentTime->setText(get_dive_date_string(when));
		ui.shiftedTime->setText(get_dive_date_string(when));
	}
```

bstoeger, 2023-01-18:
Sorry, I can't reproduce this and do not understand what is going on. The context menu to shift dive times should only appear when dives are selected. Thus, `first_selected_dive` should not return NULL.

Please try the path in #3573. The dialog should not start at all if no dive is selected.

> A window (that annoyingly can't be moved) 

Perhaps a Qt/MacOS issue with modal dialogs?. On Linux, I can move the dialog.

bstoeger, 2023-01-18:
> I assume the externally linked [struct dive_table dive_table](https://github.com/subsurface/subsurface/blob/acd385048ae61248c7f961318ed99b4aa627928c/core/divelist.h#L22) imported from core/divelist.h used by get_dive(int) is defined in [core/parse.c](https://github.com/subsurface/subsurface/blob/acd385048ae61248c7f961318ed99b4aa627928c/core/parse.c#L21), but when I did a grep I also found one in subsurface-downloader-main.cpp

Forget about the version in subsurface-downloader-main.cpp. That's a distinct standalone executable.

bstoeger, 2023-02-11:
@hhmeyer: Could you please check whether current master fixes your problem?

hhmeyer, 2023-02-12:
@bstoeger: Sorry, I'm just a user and not a developer. I would need a binary for testing.

mikeller, 2026-10-17:
This looks to be fixed: PR #3573 "desktop: pass dive list to ShiftTimesDialog" replaced the fragile dive lookup inside the Shift Dive Times dialog that was causing the 1970 date to show, passing the selected dive list in directly instead.

Closing this as fixed, though with a bit less certainty than usual since we don't have an explicit confirmation that this resolved it for the original reporter. Please reopen (or file a fresh issue) if the Shift Dive Times dialog still shows a 1969/1970 date on a current build.

closed 2026-10-17 by mikeller