subsurface/subsurface #3561: CCR dive recorded with Suunto eon core doesn't record set point switch

bchaja, 2022-11-21:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I'm diving CCR and using my old stunt eon core as backup with set set point, unfortunately even that the set point has been switched from low to high this isn't reflected in the software, and looks like subsurface pin it to 0.7 over whole dive that wasn't the case.

### Operating system:
macOS Ventura

### Subsurface version:
5.0.10

### Steps to reproduce:
CCR dive recorded with set set point on suunto eon core doesn't record set point switch

### Current behavior:
no set point switch recorded, so initial set point of 0.7 recorded thru whole dive that as an effect gives completely incorrect info on the graph like eg, subsurface calculated ceiling etc

### Expected behavior:
set point switch should be recorded and deco graph adjusted accordingly

### Additional information:
please find exported dive from subsurface, the set point has been switched at 14m what isn't reflected 

### Mentions:

[dive1.ssrf.zip](https://github.com/subsurface/subsurface/files/10060383/dive1.ssrf.zip)

bchaja, 2022-11-21:
going into information tab and flicking from ccr to oc and then back switch set point for whole dive from 0.7 to 1.3

jefdriesen, 2023-01-09:
I'll need a download log to investigate. Simply download dives from your dive computer with both the "Save libdivecomputer logfile" and "Force download of all dives" checkboxes enabled. After the download you may cancel without importing any dives to avoid re-importing previously downloaded dives. Attach the resulting log file to the bug report.