subsurface/subsurface #3563: Subsurface Mobile Android doesn't recognize Oceanic VT3

ForceRedo, 2022-11-27:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I am trying to connect my old Oceanic VT3 with USB to my Android phone. It works with DiveMate, so the connection itself is working.

On the dive computer screen it says "Info: connecting..." forever. And the little blue bar is moving across the screen. It can't ba cancelled either.

### Operating system:

I'm running A motorola G7 with Android 10
### Subsurface version:

I'm using the current android subsurface mobile version 3.4.7 (5.0.9.35)

### Steps to reproduce:

1) start subsurface
2) go to the computer/sync screen
3) plug in Oceanic VT3 and select it from the list
4) hit "download"

### Current behavior:

it seems to hang on "info: Connecting..."

### Expected behavior:

downloading my dives

### Additional information:

### Mentions:

mikeller, 2026-10-17:
Hi @ForceRedo,

Thanks for reporting that the mobile app doesn't recognise your Oceanic VT3. To dig into this we'd need a log capture of the failed connection/detection attempt, enabling verbose logging in the app's settings before trying to connect and sharing the output would help a lot.

If we don't get a log within the next 4-6 weeks we'll close this as unconfirmed, but please comment or reopen whenever suits.

ForceRedo, 2026-10-19:
Hi @mikeller ,

thanks a lot for picking this up.
As this is a little older, I'm afraid I have changed mobiles in the meantime. I can get my Oceanic to sync with my laptop, so I do get the data.

Thanks again

closed 2026-10-19 by ForceRedo