subsurface/subsurface #3564: Import Dive Data from Apple Watch Ultra

djunkins, 2022-11-29:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [X] Discussion request

### Issue long description:

I am interested in implementing the import of dive data from the Apple Watch Ultra into Subsurface. According to my research, dive profile data from both the Depth (bottom timer) app and the Oceanic+ (dive computer) app are stored in the Apple HealthKit framework with the depth stored every 2-3 seconds. Swift functions are available to read the data from the iPhone the watch is paired with (if I understand correctly.) 

The discussion I think that it is worth having is whether this should be part of the Subsurface IOS app (challenging because it currently does not include access to the Swift HealthKit libraries) or whether I should implement this as a help app that is standalone but writes the data to an XML file (or some other format) that can be imported into Subsurface. I think the former is cleaner from a usability aspect, but I'm not sure how high of a hurdle it is to implement in the existing mobile app framework.

Further discussion will be needed once the profile data can be imported to see how things like the dive site, gas settings, GFs, device data, temperature, etc. can be imported, but I am just trying to import the dive profile first.

Any thoughts?

### Operating system:

IOS 16.X and WatchOS 9.X

### Subsurface version:

### Steps to reproduce:

### Current behavior:

Apple Watch Ultra dive profile data is able to be imported.

### Expected behavior:

Dives recorded on the Apple Watch Ultra (through either app currently available) can be imported into Subsurface.

### Additional information:

### Mentions:

djunkins, 2022-11-30:
Well, I have some sample data on my iPhone now from using the Depth app on my watch, tying it to a spool and dropping it in the Lake Washington Ship Canal for a few minutes. I'm going to start by using the HealthKit library to try to read the depth data and create a UDDF file to be imported into Subsurface as a standalone helper app. Once I have that working we can figure out if there is a way to merge it with the Subsurface mobile app.

dirkhh, 2022-11-30:
UDDF... why? Why aren't you just writing Subsurface XML? That seems far easier...

djunkins, 2022-11-30:
I was thinking that UDDF is a more open standard and once I figure out how to pull the data out of HealthKit, that initial standalone helper app could be used to pull the data into other logbooks as well, even if we are able to port the code into the subsurface-mobile app for direct import into the subsurface XML format.

dirkhh, 2022-11-30:
I find UDDF painful - but of course less painful then some of the other formats.
The rationale makes sense as you are indeed thinking of writing a standalone app - so this is the right thing to do.
The other thing that needs figuring out, of course, is how to then share a file between two apps. This is something that has been incredibly painful in the past (which is why the mobile apps so far don't have an import function).

But let's burn that bridge when we get to it. I'm sure an _actual_ iOS developer will find this incredibly easy... it's luddites like me who can't figure it out...

atdotde, 2022-11-30:
I am interested in this as well. As far as I understand it so far, in order to directly read the depth data, you need an "entitlement" from Apple which so far only Oceanic got for their official diving app. And I would not be surprised if it stays that way on the one hand to protect the advantage of Oceanic (for which they might have paid a lot to Apple) and on the other hand that not everybody and their dog write a dive computer app that bends lots of divers for which the blame then might come back to Apple.

Just reading past depth data from HealthKit might have lower hurdles. But still keep in mind that Apple might restrict to which developers they might give this privilege which also has a lot of privacy implications. And that they might withdraw it at any later point without proper justification with or without further code review. So it might be a good idea at least for a while to keep this a separate app from Subsurface mobile. 

Is my impression correct that you would need to write a Swift app (or at least Objective C) in order to access HealthKit data and that you would not get far with our C++ plus Qt approach?

atdotde, 2022-11-30:
Two things I looked at that might be useful: https://developer.apple.com/videos/play/wwdc2020/10664/ (Getting started with HealthKit) and https://github.com/richardmg/qtdd13_qmlapp (a Qt app that accesses Images on iOS apparently).

djunkins, 2022-11-30:
Thanks for the pointers.

djunkins, 2023-01-02:
Just to close things out here, it turns out the Apple Watch is writing depth and water temperature information to HealthKit independent of whatever dive computer app is running. I wrote a helper application that pulls dive profile data from Apple HealthKit on the iPhone that is paired with the Apple Watch Ultra and exports it as a UDDF file. The application is currently available for testing in TestFlight. 

Details for how to get access to it can be found at https://awu2uddf.scuba-sites.com/GettingStarted.html. Hopefully it will be available in the App Store before too much longer.

closed 2023-01-02 by djunkins