subsurface/subsurface #3577: Build Fails, macOS

arrmo, 2023-01-28:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Trying to build on macOS - I can build `-mobile`, for example, but `-build-deps` fails, with this error,
```
gcc -DHAVE_CONFIG_H -I. -I../../examples -I..  -I../src   -mmacosx-version-min=12.3 -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk -DUSE_DARWIN -Wall -Wmissing-prototypes -MT trexist.o -MD -MP -MF .deps/trexist.Tpo -c -o trexist.o ../../examples/trexist.c
../../examples/files.c:24:10: fatal error: 'ptp.h' file not found
#include "ptp.h"
         ^~~~~~~
mv -f .deps/tracks.Tpo .deps/tracks.Po
1 error generated.
make[2]: *** [files.o] Error 1
make[2]: *** Waiting for unfinished jobs....
mv -f .deps/folders.Tpo .deps/folders.Po
mv -f .deps/trexist.Tpo .deps/trexist.Po
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

### Operating system:

macOS 13.2

Mac mini M1

### Subsurface version:

Building, from master branch

master 0ccb762a0 build-system: disable the daily Snap build for now

### Steps to reproduce:

As above, trying to build on macOS, with the command,
```
bash subsurface/scripts/build.sh -build-deps -both
```

### Current behavior:

Error message, as above.

### Expected behavior:

Successful build.

### Additional information:

### Mentions:

dirkhh, 2023-01-28:
If I'm not mistaken, that's an error attempting to build `libmtp` - which you only need if you have a Garmin Descent mkII dive computer.  And it really doesn't need to build the examples at all. So that is something where maybe we can improve the build script a little to make sure we don't try that.

Out of curiosity, which version of Qt are you building with? Obviously x86 binaries with Qt5 work just fine, but I haven't played with this in a while (too busy with real life), so something may have atrophied here...

arrmo, 2023-01-28:
First off - thanks so much for the quick reply! Really appreciated. And I admit - also glad to see I'm not 100% crazy ... LOL. Only saying that because I was doing some hunting, was thinking it was `libmtp` related as well.

OK, so trying to build now then with `libmtp` installed, from brew => nope, same failure. Very odd, as `libmtp` seems to include `ptp.h` (at least from Sourceforge)?

Qt - good question 😜. I tried installing the noted version from Qt itself, but that didn't seem to work. so then I installed Qt5 from brew (`brew install qt5`) ... and set `LDFLAGS`, `CPPFLAGS` ... then seems to compile (after also adding `cmake`, via brew). That said, the resulting desktop executable crashes ... but perhaps I need to address the `ptp.h` issue first?

BTW, I am capturing this minor items (like brew installs), will get you a PR later if you want, NP at all.

Thanks again.

arrmo, 2023-01-29:
BTW, thinking the crash is due to this?
```
Termination Reason:    Namespace CODESIGNING, Code 2 Invalid Page
```

Thanks!

atdotde, 2023-02-01:
I ran into the pth.h issue as well. I remember having it seen in the past and at the time I could somehow work around. This is so stupid as it only happens when you build the examples for libmtp. It would probably be best if we could simply not try to build those but my understanding of autotools does not suffice to let it ignore that directory. Anybody knows how to do that?

arrmo, 2023-02-08:
OK, some digging, and `libmtp` (with `ptp.h`) is being downloaded, using
https://github.com/subsurface/subsurface/blob/master/scripts/get-dep-lib.sh

But the includes from that download are not being added (to the include path?)?

Thanks!

arrmo, 2023-02-08:
OK, some more digging. If I build `libmtp` from the downloaded directory ... it works fine, including examples. Only when subsurface tries to build it, then it fails.

Do you see the same?

Thanks!

dirkhh, 2023-02-08:
Nice work - obviously this is one of those "but it used to work here" things... and in general I prefer keeping source and build directories separate - but as far as tracking down the issue, this is excellent.

Would you like to create a pull request (and maybe for that remove `mkdir` and the commented out `cd` plus the `echo` line following it? If you don't want to deal with that, then I can handle that myself as well.

arrmo, 2023-02-08:
> but it used to work here

LOL - yep, fully understand that 😆 

> Would you like to create a pull request (and maybe for that remove `mkdir` and the commented out `cd` plus the `echo` line following it? If you don't want to deal with that, then I can handle that myself as well.

You bet, will do it - NP at all. Happy to help if I can. On a plane right now, but will do it later today.

BTW, on to my next issue now ... LOL. libxml2 is failing to build (packaging, ios) ... very odd, as `configure` is fine when run manually, but not from the script. Seems to get stuck checking `clang`, while the manual command moves on to `gcc`. Odd.

Thanks!

arrmo, 2023-02-09:
Create the PR - by all means comment and edit. I think I did it like you asked. Also, the `./configure`, vs `../configure`, not being one level down.

Seeing some odd issues building with Qt ... some seems to need Qt Creator, other homebrew qt (qt5). Do you see this also? And - is Python needed for the iOS build? Seems like it is, but I don't see it noted (but I may have missed it!).

Thanks!

arrmo, 2023-02-09:
OK, I see the packaging issue 😆. In `packaging/ios/build.sh`, it calls out
```
IOS_QT=~/Qt
```

OK, so hard coded to Qt Creator. But inside that, it's hard coded also, causing this ...
```
+ ~/Qt/5.15.2/ios/bin/qmake ~/src/subsurface/Subsurface-mobile.pro -spec macx-ios-clang CONFIG+=iphoneos CONFIG+=Device CONFIG+=release
+ make
/bin/sh: ~/Qt/5.15.2/ios/mkspecs/features/uikit/devices.py: /usr/bin/python: bad interpreter: No such file or directory
```

So Qt is forcing to go to `/usr/bin/python` ... which doesn't exist, and I can't even create a symlink there (not permitted by the OS). 

Thoughts?

Thanks!

dirkhh, 2023-02-09:
> Seeing some odd issues building with Qt ... some seems to need Qt Creator, other homebrew qt (qt5). Do you see this also? And - is Python needed for the iOS build? Seems like it is, but I don't see it noted (but I may have missed it!).

I never build using QtCreator, but I also NEVER use Homebrew Qt as that used to have a TON of issues in the past.
I install Qt directly from the official Qt binaries (which by default of course also gets you QtCreator - I just don't use that for the build).

I don't recall if Python is needed, to be honest. You should be able to tell from the GitHub Action setup... and that doesn't seem to imply a need for Python... so... 🤷🏼‍♂️

dirkhh, 2023-02-09:
> OK, I see the packaging issue 😆. In `packaging/ios/build.sh`, it calls out
> 
> ```
> IOS_QT=~/Qt
> ```
> 
> OK, so hard coded to Qt Creator. But inside that, it's hard coded also, causing this ...

Ah - that's what you mean by "Qt Creator"... the official Qt binaries. Yes, that's what I do in fact always use.

> ```
> + ~/Qt/5.15.2/ios/bin/qmake ~/src/subsurface/Subsurface-mobile.pro -spec macx-ios-clang CONFIG+=iphoneos CONFIG+=Device CONFIG+=release
> + make
> /bin/sh: ~/Qt/5.15.2/ios/mkspecs/features/uikit/devices.py: /usr/bin/python: bad interpreter: No such file or directory
> ```

So apparently that's something that Qt wants. Where is python on your system? And which OS are you on? The system I build iOS binaries with is on 12.6.3 and that has /usr/bin/python3... I wonder why this works for me...

> So Qt is forcing to go to `/usr/bin/python` ... which doesn't exist, and I can't even create a symlink there (not permitted by the OS).

I'd edit the mkspecs file... of course... eventually I need to bite the bullet and port the iOS build to Qt6 as Qt5 is no longer supported and as you seem to prove will soon (or maybe already has) stopped working.

dirkhh, 2023-02-09:
That's because the official Qt5 build doesn't support arm64. So you need to build for x86-64 with that. Only the official Qt6 supports "fat" binaries with both architectures (or arm64 binaries).

arrmo, 2023-02-09:
Thanks! Glad you have it all straight - I sure don't ... LOL. So I installed Qt6 in parallel, tried a build with `./subsurface/scripts/build.sh -build-with-qt6 -build-deps -fat-build`, but now,
```
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [libdivecomputer.la] Error 1
```
I admit, pulling my hair out a bit - and I can't afford that 😜. Trying to find the right "recipe" ... then will try to capture this in the docs.

FYI, if I build these two items (`build-deps`, `both`) with Homebrew qt5, no issue. Just me?

Thanks!

arrmo, 2023-02-10:
OK, trying to reset all, go back to the aqt installed Qt (v5.15.2), seeing this - will keep digging, but in case you have seen it,
```
CMake Deprecation Warning at CMakeLists.txt:11 (cmake_policy):
  The OLD behavior for policy CMP0071 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- GCC optimization flags: -O2
CMake Warning (dev) at /opt/homebrew/Cellar/cmake/3.25.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (git2) does
  not match the name of the calling package (LIBGIT2).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindLIBGIT2.cmake:43 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cmake/Modules/HandleFindGit2.cmake:15 (find_package)
  CMakeLists.txt:146 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- building with Qt 5.15.2
-- building without built-in user manual support
-- building without printing support
-- system name Darwin
CMake Error at CMakeLists.txt:422 (add_subdirectory):
  The source directory

    /Users/rmorris/src/subsurface/mobile-widgets

  does not contain a CMakeLists.txt file.

-- Configuring incomplete, errors occurred!
```
Thanks!

arrmo, 2023-02-10:
OK, figured that one out - I'm a git idiot ... 😜. Back to the issue with libdivecomputer. Digging!

arrmo, 2023-02-11:
FYI, a few nuggets 😆.
- I tried adding `-release` to the build command ... but it still seems to want to build libdivecomputer with debug symbols?
- I can manually build libdivecompter, no issue at all (i.e. make from inside that directory). It only has issues from build.sh.
- I'm running on a Mac mini M1, so arm64 ... perhaps that's why no debug symbols exist for x86_64?

Thoughts?

Thanks!!

arrmo, 2023-02-12:
OK, I think I see the issue(s) 😆. A few things,
- I'm building on an M1 (arm64), not an x86-64 (Intel) ... this does seem to cause at least some of the issues. I can't change the CPU of course (LOL!), so need to find a way to make it work on an M1
- as `build.sh` notes, `this implies a Qt6 build (as m1 isn't supported in Qt5)` ... so I have to use Qt6

Any and all pointers much appreciated! The first bullet above seems to be causing me grief with the `-build-deps`, and of course that's key to get the rest working. Odd that it worked with Homebrew before, but can't seem to get back to that recipe again, for some reason.

Thanks!

arrmo, 2023-02-12:
FYI, closer 😄. I will submit a PR to capture this later, once all working ... but as part of the `aqt` install, modules need to be added. It helps quite a bit. Now having issues with `googlemaps` - seems this is needed, due to Qtlocation issues in Qt6. But from `googlemaps`, 
```
Project ERROR: Unknown module(s) in QT: location
```

Hmm - still trying to figure this one out. this is what `googlemaps` is supposed to provide (I think).

Thanks!

arrmo, 2023-02-14:
OK, winding my way through ... LOL! From `build.sh`,
```
        # build the googlemaps map plugin

        cd "$SRC"
        ./${SRC_DIR}/scripts/get-dep-lib.sh single . googlemaps
        pushd googlemaps
        mkdir -p build
        mkdir -p J10build
        cd build
        if [ "$PLATFORM" = Darwin ]  && [[ $QT_VERSION == 6* ]]; then
                # since we are currently building QtLocation from source, we don't have a way to easily install
                # the private headers... so this is a bit of a hack to get those for googlemaps...
                # regardless of whether we do a fat build or not, let's do the 'native' build here
                echo "$QMAKE INCLUDEPATH=$INSTALL_ROOT/../qtlocation/build/include/QtLocation/6.3.0 QMAKE_APPLE_DEVICE_ARCHS=$(uname -m) ../googlemaps.pro"
                $QMAKE "INCLUDEPATH=$INSTALL_ROOT/../qtlocation/build/include/QtLocation/6.3.0" QMAKE_APPLE_DEVICE_ARCHS="$(uname -m)" ../googlemaps.pro
        else
                $QMAKE "INCLUDEPATH=$INSTALL_ROOT/include" ../googlemaps.pro
        fi
```

It seems like `qtlocation` should be cloned also? Perhaps from `https://code.qt.io/cgit/qt/qtlocation.git/`? Wondering if this is missing in the dependencies list (and script)?

Thanks!

dirkhh, 2023-02-14:
> FYI, a few nuggets 😆.
>
> * I tried adding `-release` to the build command ... but it still seems to want to build libdivecomputer with debug symbols?
> * I can manually build libdivecompter, no issue at [...]

digging out from a few days of overload with other projects over here...
yes, the handling of 'release' is completely inconsistent and broken because it caused all kinds of inconsistencies in the past. Definitely worth revisiting. It's an even worse can of worms on Windows
Of course, being on m1 but building against Qt5 still gets you x86_64 - which makes things even more confusing

> OK, I think I see the issue(s) 😆. A few things,
>
> * I'm building on an M1 (arm64), not an x86-64 (Intel) ... this does seem to cause at least some of the issues. I can't change the CPU of course [...]

And using Qt6 opens its other set of problems as there are some components that we use that were missing or ill-supported last time I played with this (but that was a while ago).
 
> Any and all pointers much appreciated! The first bullet above seems to be causing me grief with the `-build-deps`, and of course that's key to get the rest working. Odd that it worked with Homebrew before, but can't seem to get back to that recipe again, for some reason.

The reason that all my scripts avoid Homebrew is that it's basically impossible to build downloadable binaries (i.e. a DMG) with Homebrew that people on an older system can "just install". That's the design philosophy of Homebrew. That's why there is all that cruft about building all the dependencies from souce

dirkhh, 2023-02-14:
> 
> It seems like `qtlocation` should be cloned also? Perhaps from `https://code.qt.io/cgit/qt/qtlocation.git/`? Wondering if this is missing in the dependencies list (and script)?

QtLocation wasn't part of the Qt6 distribution six months ago. I managed to build a hacked together fork locally, but never got this to a stable state. I think that either has recently changed or is about to change - but again, I haven't had the time to play with Qt6 on m1 in a while (ironically, typing this on an m1 MBP)

arrmo, 2023-02-14:
> digging out from a few days of overload with other projects over here...

No worries! Fully understand, happens here all the time 😆.

> Of course, being on m1 but building against Qt5 still gets you x86_64 - which makes things even more confusing
> 
> >

Agreed. That one had my head spinning, LOL! I can avoid it for now with only building dependencies ... but glad to hear I wasn't just missing the point. I thought it was me - if it needs fixes, we can work through that. Just wasn't sure if I was off base to start.

> And using Qt6 opens its other set of problems as there are some components that we use that were missing or ill-supported last time I played with this (but that was a while ago).
> 
> >

Agree with you. But seems required now, at least to build on an M1, right?

> The reason that all my scripts avoid Homebrew is that it's basically impossible to build downloadable binaries (i.e. a DMG) with Homebrew that people on an older system can "just install". That's [...]

Agree with you. I'd say avoid that - `aqtinstall` works fine, same issues as Qt Creator (so use either one, but not Homebrew).

> QtLocation wasn't part of the Qt6 distribution six months ago. I managed to build a hacked together fork locally, but never got this to a stable state. I think that either has recently changed or is [...]

😜. OK, not just me again - glad to hear that, LOL. I found with the (Qt) modules installed I get a lot further, but seems I need to manually add Qtlocation ... but not sure yet how to build that 😞 

Thanks!

arrmo, 2023-02-15:
Dang it - pulling my hair out with qtlocation 😞. I can get it installed (cloned) as a dependency, but I admit ... not sure how to build it (and connect it as a Qt module?). Wondering if anyone else has experience with this? I'm not real familiar with CMake, that's not helping ... LOL.

Thanks!

arrmo, 2023-02-17:
OK, I have `qtlocation` building and installing now. Fighting with `googlemaps` 😄. Will post the info if / when I can get that building.

dirkhh, 2023-02-17:
I really appreciate having someone else helping with this. Time has been fairly hard to find lately and more people figuring this stuff out (and documenting / scripting it) really makes my life easier.

arrmo, 2023-02-17:
NP at all! Glad to help if I can - and your notes above did help, wasn't sure if it was just me or not 😆.

I do have QtLocation building now, and actually the dependencies ... some slight issues yet, still working on them. Will get back to you!

arrmo, 2023-02-18:
FYI, OK ... so QtLocation builds (and installs), so then really don't need googlemaps (right?). Asking because this breaks when trying to build the iOS app - safe to try to remove that? I'll mess with it, just looking for some guidance 😆. I do have the desktop app building now (Qt6), mobile failing right now.

And FYI, googlemaps has some issues with Qt6, like noted here,
https://github.com/vladest/googlemaps/issues/42#issuecomment-1435399622

Oh, one more question ... there are some items in `CMakeLists.txt` that I wonder about ... like deleting some items from the Qt install. OK to mess with that? Not sure if you auto-generated that file, or if it's OK to manually edit it.

Thanks!

PS: For now at least I'm also removing armv7 from iOS, as that breaks things (seems that stopped ~ iOS 10?).

dirkhh, 2023-02-18:
Googlemaps is the magic that allows us to use the "nice" tiles in the map display. So without googlemaps, we'll be using something like OpenStreetMap (which, admittedly, has come a LONG way since we first looked at it), but of course doesn't have satellite imagery.

The CMakeLists.txt is hand crafted and full of crud - much of which may be left over from earlier attempts to hack around bugs. What often works is to use `git blame` to look back as to when the particular code was introduced and what the commit log tells you about the rationale.

armv7 is indeed no longer supported.

arrmo, 2023-02-19:
> Googlemaps is the magic that allows us to use the "nice" tiles in the map display. So without googlemaps, we'll be using something like OpenStreetMap (which, admittedly, has come a LONG way since we first looked at it), but of course doesn't have satellite imagery.

That makes sense, but ... Googlemaps isn't built into the desktop app - is that expected? Or at least ... I can get `-desktop` to fully build, yet I know that Googlemaps is broken.

> The CMakeLists.txt is hand crafted and full of crud - much of which may be left over from earlier attempts to hack around bugs. What often works is to use `git blame` to look back as to when the particular code was introduced and what the commit log tells you about the rationale.

Thanks! Will dig into this more - just wanted to make sure.

> armv7 is indeed no longer supported.

OK, will remove.

And - I have the desktop app building, also qtlocation (and installed to / as part of Qt). but it complains that MapWidget.qml failed to load ... it seems like it's not picking up the qtlocation build perhaps. Seen this before?

Thanks again.

dirkhh, 2023-02-19:
It will build - but it will run without working maps -- as you point out :)

arrmo, 2023-02-19:
OK, got it! So `-buid-with-map` is needed (for that) ... correct?

Any reason why it's optional for desktop, but mandatory for iOS? Asking because I will work to get it going with Qt6, but (temporarily) would like to put it aside ... get the other issues worked out first / separately => get the rest of iOS working, then take care of googlemaps.

Make sense? Any concerns?

dirkhh, 2023-02-19:
Traditionally no one but me (and occasionally one or two others) built iOS, and for the official apps I of course wanted maps to work.

Feel free to make changes there that make sense to you! What would be nice is to have things like that in their own commit with a good commit message.

arrmo, 2023-02-19:
Makes sense, and agreed! Will also try to pull it together in logical pieces. The "bypass" googlemaps is only temporary, to debug - working through that part as well.

Next up, to figure out where `lrelease` is ... LOL.

Thanks!

arrmo, 2023-03-03:
Apologies for the delayed update - was off diving for a few days 😄. I have found a few things that need to be done, to get parts of this working - still struggling to get it all going, but before getting too far ... how do you want to track the items I'm finding? Open different tickets, to keep them separate, or list them here, or some other way? I'm good with whatever you want, just to fine a way that works best for you.

BTW, are you able to build fully (to iOS), on your M1?

Thanks!

dirkhh, 2023-03-03:
I build IOS on my older x86 iMac - mostly because "it works".

Collecting things here is fine. Ideally this should result in a PR that fixes them all :)

arrmo, 2023-03-05:
OK, some good news, and one struggle 😜. The good news,
1) I am able to get Subsurface to build with Qt6, and it does run on my M1 Mac device. This is without maps.
2) I (think I) have QtLocation building, and installing (to the Qt install)

The struggle - I'm really being blocked by googlemaps on Qt6, and this issue,
https://github.com/vladest/googlemaps/issues/42#issuecomment-1437816052

It may be me, but even just googlemaps standalone - I can't get it to build, it's blocked by this issue. And from what I can tell, this should fail. Or am I missing something?

Just need to get over this (googlemaps) hurdle, then I think I have the macOS part of things working. Next will be iOS, and a few other issues I see there, but they don't make sense to worry about until googlemaps is working - as it's also needed for the iOS build.

Thanks!

arrmo, 2023-03-08:
Woohoo! I have googlemaps building - fully. It's what I thought it was ... missing (new) declarations, when `errorOccurred` was added as a new signal. Not sure how it was building before (for others), but it is here now 😄.

One thing now - where/how to get the xml file for QtLocation? I have that building as well, but it doesn't generate an xml file?

Thanks!

arrmo, 2023-03-12:
OK, fully building on macOS 😄 ... with Qt 6.5.0 (which now includes QtLocation, stumbled across that). It's up and running, but ... no display in the map window (but no error message also). So, how to run in debug mode, on macOS? I can't seem to find that 😞.

iOS build is still failing, it seems to insist on building the simulator as x86_64. Not sure why. Still trying to track that down, but if you have any pointers 😉 

Thanks!

mikeller, 2023-03-19:
> Qt6Gui could not be found because dependency GLESv2 could not be found.

Just taking a potshot here, but could it be that you need to explicitly include libGLESv2, in the same way as what is done for ARM builds on linux:

https://github.com/subsurface/subsurface/blob/5b263a8f4eb54ac1872b4a4c39479ee78b03b9d3/CMakeLists.txt#L316-L320

arrmo, 2023-03-20:
> Just taking a potshot here, but could it be that you need to explicitly include libGLESv2, in the same way as what is done for ARM builds on linux:

You are most likely right! I admit, `extra-cmake-modules` is where it seems to break, and I can replicate that by just trying to build this one item, with the command (for debug),
```
cmake --fresh -DSHARE_INSTALL_DIR=.. -DCMAKE_PREFIX_PATH=/Users/rmorris/Qt/6.5.0/ios/lib/cmake -DQT_MAJOR_VERSION=6 -DQT_HOST_PATH=/Users/rmorris/Qt/6.5.0/macos/lib/cmake -DQT_HOST_PATH_CMAKE_DIR=/Users/rmorris/Qt/6.5.0/macos/lib/cmake ../extra-cmake-modules
```

Also in the output is the stack,
```
Call Stack (most recent call first):
  /Users/rmorris/Qt/6.5.0/ios/lib/cmake/Qt6/QtPublicDependencyHelpers.cmake:111 (find_dependency)
  /Users/rmorris/Qt/6.5.0/ios/lib/cmake/Qt6Quick/Qt6QuickDependencies.cmake:39 (_qt_internal_find_qt_dependencies)
  /Users/rmorris/Qt/6.5.0/ios/lib/cmake/Qt6Quick/Qt6QuickConfig.cmake:40 (include)
  /Users/rmorris/Qt/6.5.0/ios/lib/cmake/Qt6/Qt6Config.cmake:157 (find_package)
  tests/CMakeLists.txt:15 (find_package)
```

I admit, still trying to wade through, see where to add the item you note 😄 

Thanks!

dirkhh, 2023-04-18:
Trying to check back in on this one... should we start capturing all of your progress in a PR?
That will make it easier for others to try and help address the remaining gaps.

arrmo, 2023-05-12:
Sure, NP. I admit, not sure I have a lot of good yet, struggling to get builds working ... but I can, NP here. Do you want it on a different branch, to not mess with the main path?

Thanks!

dirkhh, 2023-05-12:
submit it as a pull request, I'll take a look and maybe continue to work on it.
Obviously this is something we need to get a handle on and I need to get back into doing more with Subsurface after several months of not doing as much as I should...

arrmo, 2023-07-09:
My sincere apologies! Life got in the way, catching back up now 😞. Before I try to submit a PR, should I pull the latest code, clean up my local store, then take a new run at it?

Thanks!

dirkhh, 2023-07-09:
Generally that seems to be a smart approach. I really appreciate help with Qt6. I'm getting quite demoralized by it.

arrmo, 2023-08-04:
OK, let's see if we can get to a common place 😆. All this is on macOS. I first installed qt6, using aqtinstall,
```
pip install --upgrade aqtinstall
aqt install-qt -O ~/Qt mac desktop 6.6 -m all; aqt install-qt -O ~/Qt mac ios 6.6 -m all;
```
Then added this to my path (in `.zshrc`),
```
export PATH="/Users/rmorris/Qt/6.6.0/macos/bin/:$PATH"
```

Now, to build with qt6, googlemaps needs to change, so in `scripts/get-dep-lib.sh`, make the change to,
```
diff --git a/scripts/get-dep-lib.sh b/scripts/get-dep-lib.sh
index 45e19c75b..f367365e7 100755
--- a/scripts/get-dep-lib.sh
+++ b/scripts/get-dep-lib.sh
@@ -169,7 +169,8 @@ for package in "${PACKAGES[@]}" ; do
                        git_checkout_library breeze-icons $CURRENT_BREEZE_ICONS https://github.com/kde/breeze-icons.git
                        ;;
                googlemaps)
-                       git_checkout_library googlemaps master https://github.com/Subsurface/googlemaps.git
+#                      git_checkout_library googlemaps master https://github.com/Subsurface/googlemaps.git
+                       git_checkout_library googlemaps master https://github.com/vladest/googlemaps.git
                        ;;
                hidapi)
                        git_checkout_library hidapi master https://github.com/libusb/hidapi.git

```

And, then I try to build, from `src`,
```
PS4='+(${BASH_SOURCE}:${LINENO}): ${FUNCNAME[0]:+${FUNCNAME[0]}(): }' bash -x ./subsurface/scripts/build.sh -build-with-qt6 -build-deps -desktop -build-with-map
```

It gets quite far along, even googlemaps builds, but I error out at,
```
[ 25%] Automatic MOC for target subsurface_desktop_preferences
/Users/rmorris/src/subsurface/core/configuredivecomputerthreads.cpp:2221:8: error: no matching function for call to 'hw_ostc3_device_fwupdate'
                rc = hw_ostc3_device_fwupdate(m_data->device, qPrintable(m_fileName), m_forceUpdate);
                     ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/rmorris/src/install-root/include/libdivecomputer/hw_ostc3.h:58:1: note: candidate function not viable: requires 2 arguments, but 3 were provided
hw_ostc3_device_fwupdate (dc_device_t *abstract, const char *filename);
^
1 error generated.
```

Have you seen this issue before? Perhaps do as I have above, see if we can get to the same place?

Thanks!

mikeller, 2023-08-04:
>                rc = hw_ostc3_device_fwupdate(m_data->device, qPrintable(m_fileName), m_forceUpdate);

Ok, this is a change that I made in the Subsurface-DS9 'not-fork' of libdivecomputer: https://github.com/subsurface/libdc/pull/47

From the looks of it your build doesn't use the HEAD of the `Subsurface-DS9` branch in the libdivecomputer submodule.

arrmo, 2023-08-04:
Funny, I just checked that 😆. Forced it back to `Subsurface-DS9` ... cleaning, trying the build again. Will let you know!

arrmo, 2023-08-04:
Hmm ... seems like subsurface itself is also broken?
```
[ 23%] Building CXX object stats/CMakeFiles/subsurface_stats.dir/quartilemarker.cpp.o
/Users/rmorris/src/subsurface/core/configuredivecomputerthreads.cpp:2221:8: error: no matching function for call to 'hw_ostc3_device_fwupdate'
                rc = hw_ostc3_device_fwupdate(m_data->device, qPrintable(m_fileName), m_forceUpdate);
                     ^~~~~~~~~~~~~~~~~~~~~~~~
/Users/rmorris/src/install-root/include/libdivecomputer/hw_ostc3.h:58:1: note: candidate function not viable: requires 2 arguments, but 3 were provided
hw_ostc3_device_fwupdate (dc_device_t *abstract, const char *filename);
^
1 error generated.
```

And I checked, 
```
libdivecomputer % git branch -v
* Subsurface-DS9 064e198 [behind 122] garmin: relax string parsing sanity checks
```

Thanks!

mikeller, 2023-08-04:
> [behind 122]

Your branch is behind - `git submodule update` in `subsurface` will get it to where it needs to be. (Note: This does not happen automatically on a `git pull`, which is a bit of a pitfall.)

arrmo, 2023-08-05:
That did help, thanks! I was using `git pull`, not right it seems 😜.

Gets further in the build now, next up to fight with,
```
[ 43%] Linking CXX static library libsubsurface_models_desktop.a
In file included from /Users/rmorris/src/subsurface/core/libdivecomputer.c:41:
In file included from /Users/rmorris/Qt/6.6.0/macos/lib/QtCore.framework/Headers/QtGlobal:1:
In file included from /Users/rmorris/Qt/6.6.0/macos/lib/QtCore.framework/Headers/qglobal.h:37:
/Users/rmorris/Qt/6.6.0/macos/lib/QtCore.framework/Headers/qtypes.h:68:15: error: expected parameter declarator
static_assert(sizeof(ptrdiff_t) == sizeof(size_t), "Weird ptrdiff_t and size_t definitions");
              ^
/Users/rmorris/Qt/6.6.0/macos/lib/QtCore.framework/Headers/qtypes.h:68:15: error: expected ')'
/Users/rmorris/Qt/6.6.0/macos/lib/QtCore.framework/Headers/qtypes.h:68:14: note: to match this '('
static_assert(sizeof(ptrdiff_t) == sizeof(size_t), "Weird ptrdiff_t and size_t definitions");
             ^
2 errors generated.
make[2]: *** [core/CMakeFiles/subsurface_corelib.dir/libdivecomputer.c.o] Error 1
make[1]: *** [core/CMakeFiles/subsurface_corelib.dir/all] Error 2
```

Thanks!

arrmo, 2023-08-05:
OK, I went back to Qt v6.5.2, as I had that working before ... builds, no issues! I have Subsurface (desktop) working with Qt6 (v6.5.2). @dirkhh, try this as well, with the updates noted above?

I want to try v6.6.0 again, clean build ... see if that works.

Then, next up - the mobile build.

Make sense?

arrmo, 2023-08-05:
OK, fails with v6.6.0 ... likely a Qt bug, or new feature / change. Desktop builds clean, with v6.5 (aqtinstall pulls v6.5.2). Let me know if it works for you!

Mobile build fails, but I see some hard-coded Qt5 items. We'll need to fix those. But first, see if the info above gets you to Desktop working.

BTW, I have a script that "cleans" the build. I need to get that posted yet 😆

Egbertdepauw, 2024-08-26:
What's the current status on building with Qt6?

I am using an Macbook Pro M3 Max and can build v6.5.2 and v6.6.3 with some minor changes to get googlemaps working.

I am using Qt6 from the official Qt installers and selected the QtLocation lib in the official installer via the "MaintenanceTool.app" app

I've made changes to get-dep-lib.sh to use the vladest master instead of the included version.

```
diff --git a/scripts/get-dep-lib.sh b/scripts/get-dep-lib.sh
index ec261d46f..36e13560a 100755
--- a/scripts/get-dep-lib.sh
+++ b/scripts/get-dep-lib.sh
@@ -169,8 +169,8 @@ for package in "${PACKAGES[@]}" ; do
 			git_checkout_library breeze-icons $CURRENT_BREEZE_ICONS https://github.com/kde/breeze-icons.git
 			;;
 		googlemaps)
-			git_checkout_library googlemaps master https://github.com/Subsurface/googlemaps.git
-			;;
+			git_checkout_library googlemaps master https://github.com/vladest/googlemaps.git
+      ;;
 		hidapi)
 			git_checkout_library hidapi master https://github.com/libusb/hidapi.git
 			;;
```

changed the map-widget/qml/MapWidget.qml imports.

```
diff --git a/map-widget/qml/MapWidget.qml b/map-widget/qml/MapWidget.qml
index f028bd1c3..a183d9449 100644
--- a/map-widget/qml/MapWidget.qml
+++ b/map-widget/qml/MapWidget.qml
@@ -1,7 +1,7 @@
 // SPDX-License-Identifier: GPL-2.0
-import QtQuick 2.5
-import QtLocation 5.3
-import QtPositioning 5.3
+import QtQuick 
+import QtLocation 
+import QtPositioning 
 import org.subsurfacedivelog.mobile 1.0
```

and did the same in map-widget/qmlmapwidgethelper.cpp

```
diff --git a/map-widget/qmlmapwidgethelper.cpp b/map-widget/qmlmapwidgethelper.cpp
index f957fac49..6f947f6ac 100644
--- a/map-widget/qmlmapwidgethelper.cpp
+++ b/map-widget/qmlmapwidgethelper.cpp
@@ -251,8 +251,8 @@ QString MapWidgetHelper::pluginObject()
 {
 	QString lang = getUiLanguage().replace('_', '-');
 	QString cacheFolder = QString::fromStdString(system_default_directory() + "/googlemaps").replace("\\", "/");
-	return QStringLiteral("import QtQuick 2.0;"
-			      "import QtLocation 5.3;"
+	return QStringLiteral("import QtQuick;"
+			      "import QtLocation;"
 			      "Plugin {"
 			      "    id: mapPlugin;"
 			      "    name: 'googlemaps';"
```
 
I can only build the desktop build at the moment. And googlemaps is working, however panning and zooming in the map are not.

I can build with these settings:

First 
```
bash -x ./subsurface/scripts/build.sh -build-with-qt6 -build-with-map -build-deps
```
Second
```
bash -x ./subsurface/scripts/build.sh -build-with-qt6 -build-with-map -desktop

```

If I try to build it in one step for the first time it fails, but like described above, if I do it in 2 steps it seems to work.

Egbertdepauw, 2024-08-26:
Panning and zooming working. had to add some code to the MapWidget.qml as to expected, because Qt6 handles panning and zooming in QtLocation differently.

```
       property geoCoordinate startCentroid
       startCentroid: newCenter

       PinchHandler {
            id: pinch
            target: null
            onActiveChanged: if (active) {
                map.startCentroid = map.toCoordinate(pinch.centroid.position, false)
            }
            onScaleChanged: (delta) => {
                map.zoomLevel += Math.log2(delta)
                map.alignCoordinateToPoint(map.startCentroid, pinch.centroid.position)
            }
            onRotationChanged: (delta) => {
                map.bearing -= delta
                map.alignCoordinateToPoint(map.startCentroid, pinch.centroid.position)
            }
            grabPermissions: PointerHandler.TakeOverForbidden
        }
        WheelHandler {
            id: wheel
            // workaround for QTBUG-87646 / QTBUG-112394 / QTBUG-112432:
            // Magic Mouse pretends to be a trackpad but doesn't work with PinchHandler
            // and we don't yet distinguish mice and trackpads on Wayland either
            acceptedDevices: Qt.platform.pluginName === "cocoa" || Qt.platform.pluginName === "wayland"
                             ? PointerDevice.Mouse | PointerDevice.TouchPad
                             : PointerDevice.Mouse
            rotationScale: 1/120
            property: "zoomLevel"
        }
        DragHandler {
            id: drag
            target: null
            onTranslationChanged: (delta) => map.pan(-delta.x, -delta.y)
        }
```

dirkhh, 2024-08-27:
I have a branch where I am desperately working on getting the mobile app to work with Qt 6.7.
The machine I'm doing that one is at home (and I am diving), so I don't know when I'll have the time to clean the fixes up and create a PR. I do know that there are some more changes needed, so I'm not surprised that this would be failing at this point.

I'd love to get a pull request with the updates that you have made so far...

Right now it's a requirement that any changes are done in a way that they work for Qt5 and Qt6 as the main binaries are all still compiled with Qt5.

Egbertdepauw, 2024-08-27:
I managed to get it building with Qt6.7.2 However I didn't test mobile support and if it builds and works with Qt5

Egbertdepauw, 2024-08-27:
I cannot build with Qt 5.15.2 but on my system also the master branch doesn't build with Qt 5.15.2

dirkhh, 2024-08-27:
Given that most of our test builds are done with Qt 5.15, I'm curious what fails there for you...

[1 more reply]

closed 2026-10-17 by mikeller