subsurface/subsurface #3589: Import Issue importing i330R .zxu with German language

togglefog, 2023-02-07:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

If i´m trying to import a .zxu file from an i330R Divecomputer nothing will apere in subsurface. Only an empty divelog is added (no depth information, time or anything else is added even if in the selection screen before the import these information have been shown).
This only happens if Subsurface is set to German language. After a language change to english everything works fine.

### Operating system:

Windows 10, German language

Same behavior on Ubuntu 22.04

### Subsurface version:

5.0.10

Official release

### Steps to reproduce:

- Set Subsurface to German (language setting)
- import a .zxu file
- accept import settings (none will make a difference)
- check the new added dive for mistakes (the imported dive is empty :( )

- set Subsurface to English (language setting)
- import the same .zxu file
- accept import settings (they are preselected correct now)
- the dive is added correct to subsurface

### Current behavior:

### Expected behavior:

### Additional information:

See these posts in the Subsurface Googlegroup for more INformation:
https://groups.google.com/g/subsurface-divelog/c/TOrg_Bw1Qng/m/M4RBN-uJAwAJ
https://groups.google.com/g/subsurface-divelog/c/TOrg_Bw1Qng/m/gQpf9WOKAwAJ
https://groups.google.com/g/subsurface-divelog/c/TOrg_Bw1Qng/m/P1hfb3-OAwAJ

### Mentions:

bstoeger, 2023-02-11:
I think I can reproduce this with the DL7.zxu file. I would guess that it's a decimal-separator thing. Will investigate.

bstoeger, 2023-02-11:
This seems to be some XSLT/locale issue. When using the German locale, I don't get `<sample>` tags, when using the English locale, I do.

Since I don't understand XSLT at all, the only way I could fix this would be to rewrite the DAN-parser from scratch. Therefore reassigning to @mturkia.