subsurface/subsurface #3849: Dive planner doesn't respect 24hr time format preference

Ole-NM, 2023-03-22:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

- Windows 10 (22H2)
- Fedora 37 (6.2.7 kernel)

### Subsurface version:

5.0.10 (official release)

### Steps to reproduce:

1) Open a logbook
2) _File_ > _Preferences_ > _Language_
3) select `hh:mm` from the **_Time format_** dropdown (or type `HH:mm`) and save.
4) _Log_ > _Plan dive_

### Current behavior:

The dive planner's time format is still 12hr (AM/PM). It is impossible to enter a time in 24hr format.

[image]

### Expected behavior:

Similar to the normal dive view, the start time in the dive planner should reflect the 24hr time format preference.

[image]

### Additional information:

When hovering over the time format field in _Preferences_ > _Language_ you can see the following tips:

[image]

However when you try to save or apply "HH:mm" it is changed to "hh:mm" (12hr clock hours), eventhough 24hr clock is still used in normal dive view.
I tried saving "mm:HH" which isn't useful, but it's notable that it wasn't changed to "mm:hh" like in the previous option with "HH:mm".

### Mentions:

mikeller, 2023-03-23:
Thanks for reporting this bug @Ole-NM. I have proposed a fix for the incorrect date formatting in the planner in #3850.

As for the way that the format string is interpreted, and the discrepancy from what the tooltip says, I suspect that the tooltip is simply wrong, and the intent is to use the [format string rules for QDateTime](https://www.copperspice.com/docs/cs_api/class_qdatetime.html#a9efc30932238f1a0dbc9e186dd2a9e0c).
For the de-capitalisation of HH I suspect that this is an attempt to 'sanitise' the input, as `HH` in the format string is producing somewhat weird output:

```
HH | hour with a leading zero (00 to 23, even with AM/PM display)
```

I'll see what can best be done to improve this, and open a separate pull request.

closed 2023-03-23 by Ole-NM

mikeller, 2023-03-23:
> Thank you @mikeller. I built it with your fix and it works as expected.
> 
> It's possible to change the tooltips with something like this
> 
> ```c++
> ui->timeFormatEntry->setToolTip("Time format options:\n"
>                                 "h\thour without leading zero\n"
>                                 "hh\thour with leading zero\n"
>                                 "m\tminute without leading zero\n"
>                                 "mm\tminute with leading zero");
> ```
> 
> in [desktop-widgets/preferences/preferences_language.cpp](https://github.com/subsurface/subsurface/blob/master/desktop-widgets/preferences/preferences_language.cpp?rgh-link-date=2023-03-23T13%3A54%3A04Z#L34-L36) (behind line 36?)
> 
> [image]
> 
> I'm not sure which options should be listed, or if the mentioned [qt-docs](https://doc.qt.io/archives/qt-4.8/qdatetime.html#fromString) should be linked there.
> 
> edit: `AP` (for AM/PM) and `ap` (for am/pm) should be listed, too.

Yeah, I was thinking something similar - and to probably include the link to the full spec for people to find, as it is quite hard to locate since it only seems to exist in the documentation for a deprecated version of Qt but not in the documentation for any current version.
But I think the 'sanitisation' (`HH` => `hh`) should also be removed as this is confusing, and the warning that pops up when characters like `/` are used in the format string should go as well as it is misleading.

PR #3850 (2023-03-29): Desktop: Fix time format in dive planner.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Make the time edit respect the configured time format. Also make the date and time format change when the preferences are changed.

### Changes made:

1. make the time edit respect the configured time format;
2. make the date and time format change when the preferences are changed.

### Related issues:

Fixes #3849.

### Additional [... cut]
commit 468c1d8d (2023-03-29): Desktop: Fix time format in dive planner.

Make the time edit respect the configured time format. Also make the
date and time format change when the preferences are changed.

Fixes #3849.