subsurface/subsurface #3858: Dive Computer Serial Number

DJP-MSDT, 2023-04-03:
### Describe the issue:

- [ ] Bug
- [ My Dive Log contains logged dives over many years during which time several diffent dive computers have been used. I annot manually edit the dive computer information] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
 When connecting a Dive Computer to download dive logs the computer specific information is not always shown in the Extra Info tab. When using other Dive Log software such as Suunto DM5 the dive computer Serial number is always shown and it provides a great reference point to identify or produce reports based on that specific Dive Computer Make Model and Serial Number.
I would ask that you make the Extra Info tab in Subsurface editable. Then if on a selection of dives the Dive Computer specific information has not been registered when connecting the Dive Computer the User can then manually update the fields to show the 4 most important information tabs, those being Dive Computer Owner, Make, Model and Serial Number. Optionally the purchase date may be vauable to users, and also if multiple dive computer profiles have been downloaded from more than one dive computer relating to the same dive then easily identifiable labels that show the dive computer from which the dive log initiated would be a great asset to Subsurface.  As a dive professional I always dive with two dive computers so that I have a backup if one fails during a dive. So being able to show both dive logs from the same dive specific to each dive computer would be a great enhancement.

### Operating system:
 Windows 7 Prof
 Mares Puck Air and Suuunto Zoop Novo.
 Yes

### Subsurface version:
 5.0.10
 Yes
 Official Release
 NA

### Steps to reproduce:
 1) Import from Dive Computer 2) Click on Extra Info 3) View Listed information under Key and Value fields. 4) Be able to add / edit field where information is missing, incorrect or known by the user 

### Current behavior:
 Extra Info fields where shown cannot be edited

### Expected behavior:
 Ability to Add, Edit or Relace Key Label and Value fields

### Additional information:

### Mentions:

rafsalvioni, 2023-04-10:
Let me understand: do you wanna edit extra information because your old dives doenst have DC's serial number? Or is it a bug that sometimes your current DC doenst give serial number?

DJP-MSDT, 2023-04-11:
Foremost it is to correctly register a specific dive computer to each dive. Most of my old dive logs do not show the dive computer serial number. I have used more than one Suunto Mosquito in the past so I would like to enter the serial number from that dive computer. The log already shows it is a Suunto Mosquito but does not show the serial number.

Additionally since becoming a PADI Dive Pro I have used more than one Dive Computer. For instance a Suunto Mosquito on my wrist and  Mares Puck Air or a Suunto Vyper 2. I wish to log the profiles from both dive computers to each dive that they were used on. So for old dives where I used 2 dive computers I would like to edit the Extra Information for each of the 2 dive profiles shown on the same dive.

Thirdly, I have noticed that when trying to retrieve extra information when a dive computer is attched to a USB port and recognized by the software that no extra information is ever transmitted to the dive log. I guess this could be a bug, or it could be a weak/poor battery or poor contacts.

It is not clear why specific dive computer information is not received nor is it clear just what type of extra information that Subsurface can receive.

I also guess that each Dive Computer manufacturer will record different information relevant to that specific make and model but just what extra information should be possible to receive from each manufactuer and model is not documented anywhere (or I have'nt found it)

I hope that this e-mail adds some clarification.

I appreciate that this may be a wide ranging scope with complexities so if the simplist way is to allow the User to edit the Extra Info fields manually that at least will provide relevant and helpful information.

If I can help any further then please contact me and thank you in  advance for your time and effort.

David Parker

rafsalvioni, 2023-04-11:
I had a similiar problem with my DC... I dont known if was a issue from libdivecomputer (Subsurface use him) or Subsurface at own... My old dives didnt have serial number, but, newest have. I solved this problem put the serial manually on logbook, editing Subsurface's XML (.ssrf) file. For now, maybe this solution is good for you.

I did some scripts that I used to solve (for me) some questions about Subsurface. The serial's number issue is one for them. If you want to try, here is: https://rafsalvioni.github.io/subsurface-tools/fix-serial.html

To use it, you could export dives that you wanna solve using option "Subsurface XML". After, open this file in a text editor, use its contents in my script and follow the steps. The script will generate a new code updated. You could download it and open it with Subsurface to test

mikeller, 2026-10-18:
Hi @DJP-MSDT,

Making the Extra Info tab genuinely editable (rather than just a read-only display of whatever the dive computer reported) would need a defined data model for user overrides and careful handling of what happens on a future re-download, a moderate amount of work. Given rafsalvioni's workaround script already provides a path to patch in serial numbers and similar metadata, are you still looking for this to be natively supported in the app, or does that workaround cover your need well enough? Knowing that will help us prioritise this against other requests.