subsurface/subsurface #3862: SAC and Delayed Surface Marker Buoy (DSMB)

Julaye, 2023-04-16:
### Describe the issue:

- [ ] Bug
- [X] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

SAC and Delayed Surface Marker Buoy (DSMB)

The calculation of the SAC can be disrupted by sending a DSMB ...

In this example, we go from < 15 l/min to > 60 l/min long enough to lose 9 bar.

On this dive, from 178 at the entry to 69 bar at the exit, this represents 9/(178-69) = 8% difference between the diver's consumption and the total diver + DSMB consumption.

Would it be possible to detect this particular point and adjust the SAC accordingly?

[image]

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mikeller, 2023-04-16:
It is an interesting point that you are raising. I have personally never used air integration, or for that matter looked at dive logs including air integration data. But based on the fact that air consumption rate is strongly dependent on workload I would suspect that for a lot of divers / dive types / dive sites the consumption rate will not be as constant as the profile that you are showing, or your profiles in general. Having to swim fast to catch up with a buddy, having to swim into a current to get back to the mooring line, or other activities will likely cause spikes similar to what your dive log is showing when launching the DSMB.
For this reason I think it will be very hard to create logic that can reliably detect a DSMB launch and remove it from the SAC rate calculation.
So the only thing that I can see to implement this feature will be to add a way to manually mark parts of the dive as 'ignored for SAC rate calculation'.

steffen-kdab, 2023-04-30:
That's one serious DSMB :) 
9 bar, even from a small 80cf tank is 100L of air, at 10m depth enough to fill a 200L bag. Under normal circumstances, filling a DSMB should not be noticable on the air consumption graph.

mikeller, 2026-10-18:
Hi @Julaye,

Following up on this: automatic detection of DSMB-inflation gas use isn't reliably feasible, any sudden high-consumption spike (swimming against current, assisting a buddy, etc.) looks statistically identical to a DSMB fill in the pressure data, so a heuristic would inevitably misclassify genuine workload variation too.

A manual alternative seems more workable: letting you mark a specific time range of the dive as "excluded from SAC rate calculation", rather than Subsurface trying to guess. Would that address what you're after? If there's interest, that's a bounded, moderate-complexity feature we could scope properly, it just needs a UI design for selecting/marking the excluded range, which nobody's done yet.