subsurface/subsurface #3868: Dive statistics on Mac (M1) terminates subsurface

wrogner, 2023-04-21:
### Describe the issue:

- [ x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When selecting View / Dive statistics (Cmd-T), subsurface closes, a dialog Reopen, Report..., Ignore appears.
Selecting View / Yearly statistics (Cmd-Y) works (opens a window containing all dives ordered by year)

### Operating system:

MacOS 13.3.1 on MB Pro w/ Apple M1 Max

Same problem on MacOS 12.6.5 on MBPro w/ Intel Core i7 Dual Core

### Subsurface version:

5.0.10 (official release)

### Steps to reproduce:

1) Open subsurface (log automatically selected)
2) Open Dive statistics

### Current behavior:

Drive statistics terminates subsurface.
Behaviour is log-agnostic (tried with 3 divelogs and a freshly created on)

### Expected behavior:

A window with dive statistics opens up

### Additional information:

### Mentions:

[testlog.ssrf.zip](https://github.com/subsurface/subsurface/files/11296175/testlog.ssrf.zip)

mikeller, 2023-04-21:
Not sure what the problem is in this case, but I tried to reproduce on an x86 Mac running 12.6.4, and 'Dive statistics' open just fine. Confirmed with my own log and with the test log that you provided:

[image]

Updating MacOs to 12.6.5 now to check if this makes a difference.

One possible reason for this happening could be that the crash is caused by a custom setting in Subsurface, can you please try if you can reproduce it in Subsurface running with a new user with default settings, by starting Subsurface from a Terminal with `/Applications/Subsurface.app/Contents/MacOS/Subsurface --user=test`?

Also, a log of the debug output before the crash will help: Run Subsurface from a Terminal with `/Applications/Subsurface.app/Contents/MacOS/Subsurface -v -v -v`, then after the crash grab the output written to the Terminal after the program start and paste it here.

wrogner, 2023-04-22:
using user test:

Segmentation fault: 11

Problem Report content:
--------

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Subsurface [12010]
Path:                  /Applications/Subsurface.app/Contents/MacOS/Subsurface
Identifier:            org.subsurface-divelog
Version:               5.0.10 (5.0.10)
Code Type:             ARM-64 (Native)
Parent Process:        bash [11981]
Responsible:           Terminal [11979]
User ID:               501

Date/Time:             2023-04-22 10:24:18.6679 +0200
OS Version:            macOS 13.3.1 (22E261)
Report Version:        12
Anonymous UUID:        BED28FDF-E777-6503-ECAE-8E55B865B679

Sleep/Wake UUID:       6746C14F-47E7-46AD-A03C-B428C33D8F38

Time Awake Since Boot: 820000 seconds
Time Since Wake:       1007 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [12010]

VM Region Info: 0 is not in any region.  Bytes before following region: 4374642688
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104bfc000-105270000    [ 6608K] r-x/r-x SM=COW  ...OS/Subsurface

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   QtQuick                       	       0x107cb23a8 QSGSoftwareRenderableNode::update() + 544
1   QtQuick                       	       0x107cb4a6c 0x107bb8000 + 1034860
2   QtQuick                       	       0x107ceeb24 QSGNodeVisitorEx::visitChildren(QSGNode*) + 108
3   QtQuick                       	       0x107cb6008 0x107bb8000 + 1040392
4   QtQuick                       	       0x107ca826c QSGAbstractSoftwareRenderer::nodeAdded(QSGNode*) + 124
5   QtQuick                       	       0x107ca7ee4 QSGAbstractSoftwareRenderer::nodeChanged(QSGNode*, QFlags<QSGNode::DirtyStateBit>) + 164
6   QtQuick                       	       0x107ce2750 QSGNode::appendChildNode(QSGNode*) + 144
7   Subsurface                    	       0x104eee238 ChartPixmapItem::render(StatsTheme const&) + 172
8   Subsurface                    	       0x104f22000 StatsView::updatePaintNode(QSGNode*, QQuickItem::UpdatePaintNodeData*) + 160
9   QtQuick                       	       0x107c9fe50 QQuickWindowPrivate::updateDirtyNode(QQuickItem*) + 2676
10  QtQuick                       	       0x107c9abd8 QQuickWindowPrivate::updateDirtyNodes() + 316
11  QtQuick                       	       0x107c9a88c QQuickWindowPrivate::syncSceneGraph() + 356
12  QtQuick                       	       0x107c4119c QQuickRenderControl::sync() + 68
13  QtQuickWidgets                	       0x10639e4b0 0x106394000 + 42160
14  QtWidgets                     	       0x107667e8c QWidget::event(QEvent*) + 984
15  QtQuickWidgets                	       0x1063a1eac QQuickWidget::event(QEvent*) + 720
16  QtWidgets                     	       0x10761d330 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 272
17  QtWidgets                     	       0x10761eedc QApplication::notify(QObject*, QEvent*) + 3396
18  QtCore                        	       0x10902486c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 208
19  QtWidgets                     	       0x107664ef4 QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) + 1920
20  QtWidgets                     	       0x107665124 QWidget::setGeometry(QRect const&) + 208
21  QtWidgets                     	       0x10764f028 QWidgetItem::setGeometry(QRect const&) + 712
[... 993 lines cut ...]
Model: MacBookPro18,4, BootROM 8422.100.650, proc 10:8:2 processors, 64 GB, SMC 
Graphics: Apple M1 Max, Apple M1 Max, Built-In
Display: DELL U4021QW, 5120 x 2160 (Ultra-wide 5K), Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Mar  8 2023 18:27:05 version 20.96.27.0.8.7.144 FWID 01-ec2ac09e
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
PCI Card: pci8086,15f0, USB eXtensible Host Controller, Thunderbolt@3,0,0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: 4-Port USB 3.0 Hub
USB Device: Logitech BRIO
USB Device: PSSD T7
USB Device: OneTouch HDD
USB Device: 4-Port USB 3.0 Hub
USB Device: USB 10/100/1000 LAN
USB Device: 4-Port USB 2.0 Hub
USB Device: USB2.0 HID
USB Device: 4-Port USB 2.0 Hub
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Device: U4021QW, Dell, 1, 61.2
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

wrogner, 2023-04-22:
using max verbosity:

Console log:

-----

Profile calculation for dive  74 took 2 ms  -- calculated ceiling preference is false
Profile calculation for dive  139 took 8 ms  -- calculated ceiling preference is false
Profile calculation for dive  146 took 8 ms  -- calculated ceiling preference is false
Profile calculation for dive  146 took 11 ms  -- calculated ceiling preference is false
Segmentation fault: 11
$
------

No more information on the console

wrogner, 2023-04-22:
Tried another path:

I have subsurface installed via homebrew. Brew uses https://subsurface-divelog.org/downloads/Subsurface-5.0.10-qt6-universal.dmg as its source.

-> Problem arises with this version

Downloaded subsurface from [subsurface-divelog.org](https://subsurface.github.io/download/)

-> Dive statistics is shown beautifully (no errors)

I tried both versions from the downloads directory:

**Subsurface-5.0.10** -> Intel version -> **works** but not native M1
**Subsurface-5.0.10-qt6-universal** -> M1 support -> **breaks**

=> Seems there is an error in the universal image.

mikeller, 2023-04-22:
@wrogner: Ha, at least that explains the discrepancy in behaviour - apologies, I had not picked up on the fact that you had to be using the `Subsurface-5.0.10-qt6-universal.dmg` binary on M1. Using this binary I do get the same crashes that you are describing on my x86 Mac.
Unfortunately the error log and problem report produced by a crash of the published binary do not contain much helpful information. I'll see if I can get a version with debug information to build on my relatively puny Mac to get a better idea of what is going on.

mikeller, 2023-04-25:
I managed to get the x86 version of Subsurface to build on MacOS with Qt6 (which is what is used for the universal build), and this shows the same segmentation fault when opening the Dive statistics page - and with this version I was able to get the full stack trace:

<details>
<summary>Problem Report</summary>
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Subsurface [78561]
Path:                  /Users/USER/*/Subsurface.app/Contents/MacOS/Subsurface
Identifier:            org.subsurface-divelog
Version:               5.0.10-167-g3199129aa3df (5.0.10-167-g3199129aa3df)
Code Type:             X86-64 (Native)
Parent Process:        zsh [461]
Responsible:           Terminal [394]
User ID:               501

Date/Time:             2023-04-25 13:54:54.8972 +1200
OS Version:            macOS 12.6.5 (21G531)
Report Version:        12
Anonymous UUID:        83F318F7-D3D9-D0AF-830F-769F811CB53C

Time Awake Since Boot: 140000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [78561]

VM Region Info: 0 is not in any region.  Bytes before following region: 4537647104
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10e770000-10ede4000    [ 6608K] r-x/r-x SM=COW  ...OS/Subsurface

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   QtQuick                       	       0x1115b174a QSGSoftwareRenderableNode::update() + 1018 (qsgsoftwarerenderablenode.cpp:151)
1   QtQuick                       	       0x1115b3fa0 bool QSGSoftwareRenderableNodeUpdater::updateRenderableNode<QSGImageNode>(QSGSoftwareRenderableNode::NodeType, QSGImageNode*) + 272 (qsgsoftwarerenderablenodeupdater_p.h:99)
2   QtQuick                       	       0x1115f3b2c QSGNodeVisitorEx::visitChildren(QSGNode*) + 124 (qsgadaptationlayer.cpp:327)
3   QtQuick                       	       0x1115b5960 QSGSoftwareRenderableNodeUpdater::updateNodes(QSGNode*, bool) + 1408 (qsgsoftwarerenderablenodeupdater.cpp:246)
4   QtQuick                       	       0x1115a47ed QSGAbstractSoftwareRenderer::nodeAdded(QSGNode*) + 109 (qsgabstractsoftwarerenderer.cpp:231)
5   QtQuick                       	       0x1115a4286 QSGAbstractSoftwareRenderer::nodeChanged(QSGNode*, QFlags<QSGNode::DirtyStateBit>) + 182 (qsgabstractsoftwarerenderer.cpp:72)
6   QtQuick                       	       0x1115e6817 QSGRootNode::notifyNodeChange(QSGNode*, QFlags<QSGNode::DirtyStateBit>) + 48 (qsgnode.cpp:1240) [inlined]
7   QtQuick                       	       0x1115e6817 QSGNode::markDirty(QFlags<QSGNode::DirtyStateBit>) + 84 (qsgnode.cpp:634) [inlined]
8   QtQuick                       	       0x1115e6817 QSGNode::appendChildNode(QSGNode*) + 135 (qsgnode.cpp:417)
9   Subsurface                    	       0x10ea79ca5 ChartPixmapItem::render(StatsTheme const&) + 149
10  Subsurface                    	       0x10eab2c1f StatsView::updatePaintNode(QSGNode*, QQuickItem::UpdatePaintNodeData*) + 175
11  QtQuick                       	       0x11159b890 QQuickWindowPrivate::updateDirtyNode(QQuickItem*) + 3280 (qquickwindow.cpp:2138)
12  QtQuick                       	       0x1115956f5 QQuickWindowPrivate::updateDirtyNodes() + 325 (qquickwindow.cpp:1884)
13  QtQuick                       	       0x1115953b4 QQuickWindowPrivate::syncSceneGraph() + 404 (qquickwindow.cpp:552)
14  QtQuick                       	       0x11153189c QQuickRenderControl::sync() + 76 (qquickrendercontrol.cpp:366)
15  QtQuickWidgets                	       0x1102ade5a QQuickWidgetPrivate::render(bool) + 58 (qquickwidget.cpp:372)
16  QtWidgets                     	       0x110ee8995 QWidget::event(QEvent*) + 1301 (qwidget.cpp:9076)
17  QtQuickWidgets                	       0x1102b1ee2 QQuickWidget::event(QEvent*) + 962 (qquickwidget.cpp:1719)
18  QtWidgets                     	       0x110e99227 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 247 (qapplication.cpp:3284)
19  QtWidgets                     	       0x110e9a04c QApplication::notify(QObject*, QEvent*) + 508
20  QtCore                        	       0x11342b6ca QCoreApplication::notifyInternal2(QObject*, QEvent*) + 170 (qcoreapplication.cpp:1115)
21  QtWidgets                     	       0x110ee5626 QWidgetPrivate::setGeometry_sys(int, int, int, int, bool) + 2198 (qwidget.cpp:7331)
22  QtWidgets                     	       0x110ee58d3 QWidget::setGeometry(QRect const&) + 227 (qwidget.cpp:7219)
23  QtWidgets                     	       0x110ecd423 QWidget::setGeometry(int, int, int, int) + 19 (qwidget.h:889) [inlined]
24  QtWidgets                     	       0x110ecd423 QWidgetItem::setGeometry(QRect const&) + 707 (qlayoutitem.cpp:478)
25  QtWidgets                     	       0x110ea63d3 QBoxLayout::setGeometry(QRect const&) + 1027
[... 493 lines cut ...]

      },
      "deploymentId" : 240000263
    },
    {
      "rolloutId" : "6112e14f37f5d11121dcd519",
      "factorPackIds" : {
        "SIRI_TEXT_TO_SPEECH" : "634710168e8be655c1316aaa"
[... cut ...]
      "treatmentId" : "c28e4ee6-1b08-4f90-8e05-2809e78310a3",
      "experimentId" : "6317d2003d24842ff850182a",
      "deploymentId" : 400000013
    }
  ]
}
}
```
</details>

From the look of the stack trace the problem happens with QtQuick. This is in the process of being refactored at the moment in #3863 (@bstoeger this might be of interest to you), so I think the best course of action is to wait for the refactoring to be completed before investigating this problem further - potentially it will have resolved itself with the refactoring.

bstoeger, 2023-04-26:
This is a duplicate of #3534, but I don't know how to merge the reports.

I very much doubt that #3863 will change anything in that regard. I believe this is a problem with the way we hook into QML/QtQuick.

I cannot reproduce the crash with Qt6 on Linux/x86, and I don't own a Mac. I do however experience some very strange behavior: When I go to the statistics tab for the first time, the main window closes and reopens. WTH?

bstoeger, 2023-04-26:
The backtrace is interesting though, in the sense that it got farther than I expected:
```
7 Subsurface 0x104eee238 ChartPixmapItem::render(StatsTheme const&) + 172
```
So it got to where it actually renders the statistics. Hm.... Something not initialized correctly?

bstoeger, 2023-04-26:
I'm not sure if this is related, but something is clearly broken on Qt6. Every time the statistics module is shown, the StatsView class is destroyed an recreated.

Destruction:
```
Destroy:

#0  StatsView::~StatsView (this=0x555556661370, __in_chrg=<optimized out>) at /home/bs/src-qt6/subsurface/stats/statsview.cpp:65
#1  0x0000555555726f3d in QQmlPrivate::QQmlElement<StatsView>::~QQmlElement (this=0x555556661370, __in_chrg=<optimized out>) at /usr/include/x86_64-linux-gnu/qt6/QtQml/qqmlprivate.h:99
#2  QQmlPrivate::QQmlElement<StatsView>::~QQmlElement (this=0x555556661370, __in_chrg=<optimized out>) at /usr/include/x86_64-linux-gnu/qt6/QtQml/qqmlprivate.h:99
#3  0x00007ffff798a8dc in ?? () from /lib/x86_64-linux-gnu/libQt6QuickWidgets.so.6
#4  0x00007ffff798d0c8 in QQuickWidget::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt6QuickWidgets.so.6
#5  0x00007ffff738baab in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#6  0x00007ffff5527858 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#7  0x00007ffff73c9181 in ?? () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#8  0x00007ffff73c9160 in ?? () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#9  0x00007ffff73d07d6 in QWidget::setParent(QWidget*, QFlags<Qt::WindowType>) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#10 0x00007ffff755ea92 in ?? () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#11 0x00005555557dbb6f in MainWindow::setQuadrantWidget (this=<optimized out>, pos=0, q=..., splitter=...) at /home/bs/src-qt6/subsurface/desktop-widgets/mainwindow.cpp:1447
#12 MainWindow::setQuadrantWidget (this=<optimized out>, splitter=..., q=..., pos=0) at /home/bs/src-qt6/subsurface/desktop-widgets/mainwindow.cpp:1440
#13 0x00005555557dbc19 in MainWindow::setQuadrantWidgets (this=0x555555ff23a0, splitter=..., left=..., right=...) at /home/bs/src-qt6/subsurface/desktop-widgets/mainwindow.cpp:1459
#14 0x00005555557dea38 in MainWindow::setApplicationState (state=<optimized out>, this=0x555555ff23a0) at /home/bs/src-qt6/subsurface/desktop-widgets/mainwindow.cpp:1491
#15 MainWindow::setApplicationState (this=0x555555ff23a0, state=<optimized out>) at /home/bs/src-qt6/subsurface/desktop-widgets/mainwindow.cpp:1479
#16 0x000055555572ed73 in MainWindow::qt_metacall (this=0x555555ff23a0, _c=QMetaObject::InvokeMetaMethod, _id=42, _a=0x7fffffffcd30)
    at /home/bs/src-qt6/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_mainwindow.cpp:559
#17 0x00007ffff5572408 in ?? () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#18 0x00007ffff663fee6 in QAction::triggered(bool) () from /lib/x86_64-linux-gnu/libQt6Gui.so.6
#19 0x00007ffff66452c5 in QAction::activate(QAction::ActionEvent) () from /lib/x86_64-linux-gnu/libQt6Gui.so.6
#20 0x00007ffff66453f8 in QAction::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt6Gui.so.6
#21 0x00007ffff738baab in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#22 0x00007ffff5527858 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#23 0x00007ffff6652aaf in QShortcutMap::dispatchEvent(QKeyEvent*) () from /lib/x86_64-linux-gnu/libQt6Gui.so.6
#24 0x00007ffff6652ecb in QShortcutMap::tryShortcut(QKeyEvent*) () from /lib/x86_64-linux-gnu/libQt6Gui.so.6
#25 0x00007ffff63e47ae in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) ()
   from /lib/x86_64-linux-gnu/libQt6Gui.so.6
#26 0x00007ffff6395e7a in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) () from /lib/x86_64-linux-gnu/libQt6Gui.so.6
#27 0x00007ffff01eaa0f in ?? () from /usr/lib/x86_64-linux-gnu/qt6/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so
#28 0x00007ffff5572326 in ?? () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#29 0x00007ffff42784b3 in QDBusPendingCallWatcher::finished(QDBusPendingCallWatcher*) () from /lib/x86_64-linux-gnu/libQt6DBus.so.6
#30 0x00007ffff5564d58 in QObject::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#31 0x00007ffff738baab in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#32 0x00007ffff5527858 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#33 0x00007ffff5527a37 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#34 0x00007ffff5717397 in ?? () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#35 0x00007ffff431549d in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#36 0x00007ffff4370178 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#37 0x00007ffff43141b0 in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#38 0x00007ffff5714f20 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#39 0x00007ffff55319aa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#40 0x00007ffff552a71c in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt6Core.so.6
[... cut ...]
#45 0x00007ffff55319aa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#46 0x00007ffff552a71c in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#47 0x000055555571e3de in main (argc=<optimized out>, argv=<optimized out>) at /home/bs/src-qt6/subsurface/subsurface-desktop-main.cpp:111
```
This happens when we add the statistics widget. Apparently the reparenting causes a full reload of the QML page. I have no idea if this is a bug of Qt6, but in any case it does NOT happen with Qt5.

bstoeger, 2023-04-26:
This (cool 1990ies retro) code is new in Qt6's `QQuickWidgetPrivate::handleWindowChange()` function:
```
    // In case of !isPersistentSceneGraph or when we need a new context due to
    // the need to share resources with the new window's context, we must both
    // invalidate the scenegraph and destroy the context. QQuickRenderControl
    // must be recreated because its RHI will contain a dangling pointer to
    // the context.

    delete offscreenWindow;
    offscreenWindow = nullptr;
    delete renderControl;

    renderControl = new QQuickWidgetRenderControl(q);
    initOffscreenWindow();

    QObject::connect(renderControl, SIGNAL(renderRequested()), q, SLOT(triggerUpdate()));
    QObject::connect(renderControl, SIGNAL(sceneChanged()), q, SLOT(triggerUpdate()));
```
I don't like this at all. Let's see if we can call `QQuickWindow::setPersistentSceneGraph()` as the code/comment suggests.

bstoeger, 2023-04-26:
Well.. It seems the problem is that `Qt::AA_ShareOpenGLContexts` is not set and I am not sure if this can always be set. Thus, we cannot prevent the `StatsView` object being regenerated when the statistics widget is reparented. 🤷 We always have to assume that the view is destroyed under our feet. That will be fun with the profile widget. Maybe will have to remove the reparenting thing and hide the widgets instead.

bstoeger, 2023-04-26:
Final comment for today:
Adding
```
        QGuiApplication::setAttribute(Qt::AA_ShareOpenGLContexts);
```
at the beginning of `main()` fixes the destruction problem. I have no idea if this is a general fix or only in my particular case.

mikeller, 2023-04-27:
Thanks for looking into this, @bstoeger.

> This is a duplicate of #3534, but I don't know how to merge the reports.

Apologies, I didn't know about #3534. I don't think there is a facility to merge isuses in GitHub, the usual approach is to close all but the issue with the latest / most activity with a comment like 'Closing as duplicate of #xxxx'.

> ```
>         QGuiApplication::setAttribute(Qt::AA_ShareOpenGLContexts);
> ```
> 
> at the beginning of `main()` fixes the destruction problem. I have no idea if this is a general fix or only in my particular case.

Tried this with Qt6 on x86 Mac, but unfortunately it is still crashing when opening Dive statistics.

bstoeger, 2023-04-27:
Yes, I was quite sure that these are different issues, as the crash here is during rendering.

mikeller, 2026-10-17:
Hi @wrogner,

This is a confirmed duplicate of #3534, bstoeger traced your crash and the one reported there to the same underlying cause (Qt6's QQuickWidget destroying/recreating the StatsView because Qt::AA_ShareOpenGLContexts isn't set, crashing during chart rendering). We're consolidating the investigation and any fix there rather than tracking it in two places.

Closing this as a duplicate, please follow #3534 for progress on an actual fix (it isn't fixed yet, just fully diagnosed), and reopen this one only if you find a case that's distinct from what's described there.

closed 2026-10-17 by mikeller