subsurface/subsurface #3871: Ventura 13.3.1 on Mac pro Apple M1 Max with iX3M dive computer

rgiordani71, 2023-04-26:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
iX3M easy pro not able to import log dives because the port where the device is connected is not recognised 
[image]

### Operating system:

Ventura 13.3.1 on Mac pro Apple M1 Max
iX3M easy pro

### Subsurface version:
5.0.10
### Steps to reproduce:
connect the dive computer on the usb port and try to import the dive logs

### Current behavior:
unable to connect

### Expected behavior:
download the dive logs

### Additional information:

### Mentions:

rgiordani71, 2023-04-26:
issue solved

closed 2023-04-26 by rgiordani71

mikeller, 2023-04-26:
@rgiordani71: In the interest of helping others who might have the same issue and find this, what was the solution?