subsurface/subsurface #3901: Bluetooth control/connectivity problem with Mac OSX Ventura and Shearwater Teric

Neill-Prohaska, 2023-05-14:
Does anyone have troubleshooting suggestions for a MacBook Pro (Intel, 2019) running OSX 13.2.1 (Ventura), trying to connect via Bluetooth to the Shearwater Teric?

I made sure to enable bluetooth access for Subsurface (5.0.10), but I always receive the same error when I try to scan for available bluetooth devices: it says "Scanning for remote devices..." for about 10 seconds, then gives the message: "Device discovery error: An unknown error has occurred...".

Also, on the right hand side for the bluetooth control display, it shows "Bluetooth power on" with a checkmark beside it, but it is greyed out (see attached image). The button for "Turn bluetooth on/off" is not greyed out, but when I click on it simply says "Trying to turn off the local Bluetooth device", but hangs up forever.

This suggests to me that maybe Subsurface is somehow NOT actually getting access to bluetooth control, despite me giving that access in the OSX security settings.

Any suggestions?

[image]

mikeller, 2026-10-17:
Hi @Neill-Prohaska,

Apologies this one never got a reply from us. The "Bluetooth power on" checkbox showing greyed out, and the toggle button hanging, does sound like it could genuinely be Subsurface not getting the Bluetooth permission grant macOS thinks it gave, a real, specific symptom worth chasing rather than a generic connection failure.

To make progress we'd need: confirmation this still happens on a current macOS (Ventura or later) build, and ideally a verbose Subsurface log covering the failed Bluetooth permission/discovery sequence (enable verbose logging in preferences, then attempt the scan).

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient, this looked like a real lead, so we'd like to follow up if you're still around.

tqc, 2026-11-12:
@mikeller I'm seeing the same behaviour of the turn on/off button on Tahoe 26.6.2 and Subsurface 6.0.5738.

The apparent hang can be resolved by turning bluetooth on/off using system settings - ie there appears to be some sort of read only access involved.

A scan does run but (with show all devices) finds only 3-4 devices highlighted in red rather than the dozens I would expect.

I was last able to connect to my Perdix 3 on 26 September 2026 and it was no longer working 3 October, so decent chance it coincides with the 26.6.2 security update installing automatically.

I wasn't able to find a debug logging option in the ui, so I tried running via `/Applications/Subsurface.app/Contents/MacOS/Subsurface -v` 

That made the prompt pop up as allow terminal rather than subsurface to access bluetooth. Turn on/off still doesn't work (and writes nothing to the log), but unlike starting the app normally, scan finds all the expected devices and I can download dives successfully.

There does seem to be inconsistency in how subsurface appears in security settings - the icon is missing and "Show in Finder" does nothing.

[image]

mikeller, 2026-11-13:
AI-drafted; reviewed by Michael.

Hi @tqc, thanks for the careful comparison. Being able to download from your Perdix 3 when Subsurface is launched through Terminal is a useful clue: the dive-computer connection works in that case, while macOS asks for Bluetooth access on Terminal’s behalf. We cannot yet tell why launching Subsurface normally gives a different scan result, but the missing icon and non-working “Show in Finder” entry make the app’s macOS permission record worth investigating.

I also need to correct my earlier suggestion that the hanging power button might indicate a missing Bluetooth permission. Qt cannot turn Bluetooth on or off programmatically on macOS. Subsurface nevertheless offers that button and leaves “Trying to turn off...” on screen; that is a misleading UI bug, which we’ll investigate. For now, please continue to use macOS System Settings to change Bluetooth power.

For the scan problem, one focused question: was Subsurface 6.0.5738 installed as a single app at /Applications/Subsurface.app, or do you have another copy or a development build? That will help us narrow down the odd permission entry before asking you to collect logs or change permissions. You can keep using your Terminal launch as a workaround for downloading dives in the meantime.

tqc, 2026-11-13:
It's a standard drag to applications install, replacing the previous version. I was previously using the ci build 6.0.5684 which I installed the same way when Perdix 3 support was added. I don't think that is the problem on its own though - that version was working correctly until last month.

Potentially worth trying some more thorough methods of removing the old version, but will hold off on that for now in the interest of debugging.

mikeller, 2026-11-13:
AI-drafted; reviewed by Michael.

A brief update: [PR #5126](https://github.com/subsurface/subsurface/pull/5126) is open as a **draft** for the misleading Bluetooth power control on macOS. It hides the unsupported toggle and points users to System Settings instead. This has not been merged or verified on a Mac, so it is not a released fix.

The PR does **not** address the different scan results when Subsurface is launched normally versus through Terminal. That remains a separate question. @tqc, when convenient, could you answer the installation question above? It will help us investigate that part without conflating it with the power button.

mikeller, 2026-11-13:
AI-drafted; reviewed by Michael.

@tqc, I see your installation answer arrived just before my follow-up comment. Sorry for asking the same question again. Replacing the earlier CI build in Applications is useful context, but it does not yet tell us why the normal launch and Terminal launch produce different scans.

Before removing or resetting anything, could you try one check? Quit Subsurface, switch its access off and back on under macOS System Settings → Privacy & Security → Bluetooth, then launch it normally from Applications and scan again. Please let us know whether the Perdix 3 appears. There is no need to remove the old settings or your dive log for this test.

tqc, 2026-11-13:
Permission turned off: ui shows "The local Bluetooth device was turned off"
Permission turned back on: Scan still broken.
Permission deleted: Prompt “Subsurface.app” would like to use Bluetooth. Scan remains broken.