subsurface/subsurface #3909: Can download from MacOS, can't from Arch Linux

eisGeist, 2023-05-23:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [X] Discussion request

### Issue long description:

I am able to download the dive logs to my old MacBook Pro (2012) but I can't do the same on my Arch Linux box.

When I open the download panel, there is a dropdown menu containing a long list of item: from /dev/ttyS0 to /dev/ttyS31 plus a /dev/ttyUSB0.

I tried each one of them, all of them return the error "Unable to open /dev/ttyS31 Cressi (Leonardo)"

Are there any means by which I can investigate and understand the source of the error?

I tried both the USB ports on my HP laptop, both returned the same error.

### Operating system:

I am running Arch Linux, regularly updated.

On an HP EliteBook, Core i5 vPro 7th Gen

On MacOS Catalina running on a MacBook Pro 2012, SubSurface downloads the diving logs without issues.

### Subsurface version:

Vers. 5.0.9

It doesn't happen on another Subsurface, AFAIK

I use the official build

### Steps to reproduce:

I connect the cradle, set the dive computer in PC Link mode, click on download from the window that opens after choosing the "Import from Dive Computer" menu item. 

### Current behavior:

The dive computer works, apparently, nice but for downloading the dive logs.

### Expected behavior:

Download the dive logs to my computer.

### Additional information:

### Mentions:

atdotde, 2023-05-23:
Sounds like a permissions problem. Do you have read and write permissions to the device? See the user manual how to fix those.

eisGeist, 2023-05-23:
Thank you, @atdotde for your suggestion!
I will look into it, and let you know.

eisGeist, 2023-05-26:
As far as I understand, I have read and write permissions on the interface:

`crw-rw---- 1 root     uucp 188,  0 May 26 21:44 /dev/ttyUSB0`

Each interface available in the dropdown menu has the same set of permissions.

dirkhh, 2023-05-26:
can you run 
`id`
as your login user?
I wonder if you are in the `uucp` group...

eisGeist, 2023-05-27:
This is the output of the `id` command:

`uid=1000(eisgeist) gid=1000(eisgeist) groups=1000(eisgeist),3(sys),998(wheel)`

mikeller, 2023-05-27:
Try `sudo usermod -a -G uucp eisgeist`, then log out and log in again, and see if works.

eisGeist, 2023-05-28:
Dear all, thank you!

Your hints and suggestion solved the issue!

closed 2023-05-28 by dirkhh