subsurface/subsurface #3915: App not installed message after update

Mateng2502, 2023-05-29:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

After running the latest update the app doesen't start (App Not Installed Message from OS)

### Operating system:

Google Pixel 6 Pro
Systemlanguage German 
Android 13 with Security Patch May 2023
Daily running updates by myself

### Subsurface version:

3.4.8 (5.0.10.177)
I'am subsurface-mobile beta tester...

### Steps to reproduce:

Update from Play Store
Try to start

### Current behavior:

It doesen't start

### Expected behavior:

It should start as expected until the last version

### Additional information:

### Mentions:

BastiSch, 2023-07-20:
I observe the same issue for probably some weeks now. I do not know what caused it.

### Operating system:
Nokia 8
Systemlanguage German
Android 9

### Subsurface version:
3.4.8 (5.0.10.177)
current version in the play store even after leaving beta program.

### Steps to reproduce:
Install from Play Store
Try to start

mikeller, 2026-10-17:
Hi @Mateng2502,

Thanks for the report. This has been fixed: Android 12+ (API 31+) requires any activity with an intent-filter to declare android:exported="true", and that was missing from our manifest. I've checked android-mobile/AndroidManifest.xml directly in current source and it's now present on the SubsurfaceMobileActivity main activity.

Closing this as fixed. Please reopen if you still see "App not installed" installing a current build.

closed 2026-10-17 by mikeller