subsurface/subsurface #3916: Subsurface-mobile start one time after update but not again

satoschu, 2023-05-30:
Describe the issue:

 x Bug

 Change request

 New feature request

 Discussion request

Issue long description:

After running the latest update the app does start one time, but not again also after deleting cache or close and reopen or restart mobile

Operating system:

Poco F3
Systemlanguage German
Android 13 with Security Patch Feb 2023

Subsurface version:

3.4.8 (5.0.10.177)
I'am subsurface-mobile beta tester...

Steps to reproduce:

Update from Play Store
Start
Try to start again

Current behavior:

It doesen't start

Expected behavior:

It should start as expected until the last version

Additional information:

Mentions:

mikeller, 2023-05-30:
Seens to be a duplicate report of #3915, as it's concerning the same version of the application.

satoschu, 2023-05-30:
Hi,

But his problem seems to be the App will not be installed. At my side the App started the first time afterwards it didn't start again....

Best regards

Tobias

atdotde, 2023-05-30:
It is a known issue that at the moment, the beta version is not functional. You can always revert to the latest production version by deregistering the beta program (you can easily rejoin later).

julienw, 2023-10-01:
I still have the problem with the beta version, but I can confirm that the production version works fine for me.

mikeller, 2026-10-17:
Hi @satoschu,

Thanks for the report. This has been fixed: Android 12+ (API 31+) requires any activity with an intent-filter to declare android:exported="true", and that was missing from our manifest. I've checked android-mobile/AndroidManifest.xml directly in current source and it's now present on the SubsurfaceMobileActivity main activity.

Closing this as fixed. Please reopen if the app still fails to relaunch for you after an update on a current build.

closed 2026-10-17 by mikeller