subsurface/subsurface #3932: Merge dives does not merge cylinders

dridb, 2023-06-26:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:

Merged two dives, imported from a Shearwater Teric with transmitter data. The log now assumes two different cylinders were used instead of also merging that data into one cylinder. Maybe a feature to also merge this data?
### Operating system:

Windows 11 Pro
Subsurface 5.0.10

Shearwater Teric
Mares Quad Air

### Subsurface version:

### Steps to reproduce:

Rightclick two selected dives. Merge

### Current behavior:

Merge dives, merged average depth, did not merge cylinders

### Expected behavior:

### Additional information:

### Mentions:

anthonycroberts, 2023-07-06:
I’m having a similar issue.  I’m not sure what the merge function picks from the gas options in the separate dives.

mikeller, 2023-07-09:
@dridb: Not sure I understand the situation you are describing. Are you manually mergeing two dives that were reported on the same dive computer (for example with a long surface swim in between)? Or are you (automatically) merging logs from two separate dive computers that were recorded on the same dive?

@anthonycroberts: The cylinder mergeing functionality in Subsurface is relatively basic - it essentially goes through the list of tanks reported in both logs, merges the cylinders that exist in both with exactly the same data into one, and adds all of the (used) tanks that only exist in one log.
I have opened #3937 to change this so that unused tanks are included in a merge if the user has enabled the 'Show unused cylinders' option in preferences.

anthonycroberts, 2023-07-18:
I think in my particular case it was creating some tank switch events.  I found an old tip from @torvalds on groups about manually editing the files so that I could remove tanks, so I was able to delete the events and just remove the excess tanks from the equipment list.  On one dive that recorded as six dives (from a lake dive with extensive surface times) using redundant air-integrated computers (Cobalt 2 & Perdix 2) on the same tank, it did some strange dive merging of the tanks.  Manually editing corrected.  On another set of dives, my brother’s Oceanic was glitching out, so I added his tank transmitter to my Perdix 2, so subsurface had both of my computers reading my tank plus the Perdix 2 reading his tank periodically.  Those just needed manual editing, as I don’t think there is a straightforward software logic to intuit the correct treatment there.  Thanks for the insight!