subsurface/subsurface #3935: Incorrect gas mix on Subsurface mobile when downloading from Shearwater

mturkia, 2023-07-02:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Downloading dives from Shearwater Petrel AI gets incorrect gas information. Dive computer is configured to EAN32 and download to mobile results in EAN33.

### Operating system:

iOS 16.5.1

### Subsurface version:

Beta 3.4.8(5.0.10.177)

### Steps to reproduce:

- Download EAN32 dives to Subsurface mobile.

### Current behavior:

Used cylinder (cylinder 1) is showing EAN33 instead of the used EAN32.

### Expected behavior:

Correct gas mix is shown

### Additional information:

### Mentions:

mturkia, 2023-07-11:
Took a look at my DC configuration, and first gas is set to off, second gas is actually in use. So it is this tiny detail that is being ignored on import.

mikeller, 2023-07-11:
There's also #3939 which fixes some brokenness when it comes to marking cylinders as used - could be that this fixes the issue you are seeing as well.

mwerle, 2023-07-18:
I had a long discussion on the mailing list with Michael. It seems there was an update a while back which changed how cylinders are imported and assigned usage in Subsurface (I haven't taken a look at the changes myself yet). Unfortunately this breaks Subsurface for all Shearwater users, as Shearwater dive computers don't send information as to which gas is "on" and which gas is "off" during the import.

Subsurface now imports ALL gas mixes, and in the case of Shearwater, in the order of highest O2 percentage first, and then assigns them as "used" in that order.

I'm planning to take a look to see if I can fix it as and when I get some time, as the current behaviour means I have to manually edit all of my dive logs in order to get rid of "ghost" cylinders. I guess the only other workaround is to delete all gas mixes off my Shearwaters. Neither is an attractive option.

I've also sent an email to Shearwater asking them to consider sending the gas mix "status" as part of their synchronisation data, but haven't heard back yet. I suspect other members in the community may have more clout.

mwerle, 2023-11-15:
To expand on Jef's comment, it seems that Shearwater computers now do send information as to whether or not a gas mix is active.  Jef has patched libdivecomputer to take this into account so once that patch (or an updated libdivecomputer) makes it into Subsurface, this issue should be resolved.

mikeller, 2026-10-17:
Hi @mturkia,

Following up on your own investigation here: this is fixed. PR #3939 "Equipment: Fix 'used' Gas Selection for CCR Dives" stopped treating the arbitrary first gas in the list as "used" by default for CCR dives, relying on actual gaschange-event-based usage detection instead. jefdriesen also landed a complementary libdivecomputer fix exposing Shearwater's active/inactive gas-mix flag, which mwerle confirmed resolves the broader "ghost cylinder" class of Shearwater gas-list problems.

Closing this as fixed. Please reopen if you still see the wrong gas mix imported from a Shearwater Petrel AI (or similar) on a current build.

closed 2026-10-17 by mikeller