subsurface/subsurface #3936: Subsurface mobile (Beta) crashes on exit

mturkia, 2023-07-02:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When exiting Subsurface mobile (Beta) on iPhone via swipe action, Subsurface crashes.

### Operating system:

iOS 16.5.1

### Subsurface version:

Beta 3.4.8(5.0.10.177)

### Steps to reproduce:

Install 

### Current behavior:

Subsurface crashes bringing up dialog allowing to send crash report to developers

### Expected behavior:

Clean exit

### Additional information:

### Mentions:

mikeller, 2026-10-17:
Hi @mturkia,

Thanks for reporting the crash on exit in the mobile beta. To look into it we'd need a crash log, on Android that's usually available via adb logcat around the time of the crash, or whatever crash report your device offers to share.

If we don't get a crash log within the next 4-6 weeks we'll close this as unconfirmed, but please comment or reopen whenever suits.

dirkhh, 2026-10-17:
Hehe - I love AI agents. The bug is about iOS, so adb isn't gonna help.
I'm 99% sure I remember what the issue was and this is long fixed, so closing this was the right decision. Just the comment was way off. :)