subsurface/subsurface #3944: Mobile App Bug

jaskev, 2023-07-23:
Hi having issues with the mobile app, keeps saying there is a bug.
have uninstalled and reinstalled and getting the same message anyone else having the same problem

<=H!-- Lines like this one are comments and will not be shown in the final output. -->

### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mikeller, 2023-07-25:
Hi @jaskev. Can you please fill in the information that you are asked to provide when opening an issue?

jaskev, 2023-07-25:
App won't open at all now. Can i confirm 4.0 is the latest version
Jason

jaskev, 2023-08-13:
Hi Michael
Have removed and Reinstalled android app from Google play.  When opening app i get the app screen image and then it crashes
Can you advise please,  seems odd as worked perfectly before
Jason

jfietz, 2023-10-17:
I have exactly the same issue.

To fill out the requested information:

### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Using a freshly installed version of Subsurface Android (in the beta channel), I open the app. It briefly displays the starting logo and then terminates.

It does work with an older version.

### Operating system:
OS: Android 14, UP1A.231105.002 - en_US
Device: Google Pixel 7 Pro

### Subsurface version:
Broken: Version: 3.4.8(5.0.10.177) - official version in beta channel

Working: Version Subsurface-mobile-4.9.9.0-arm64.apk from homepage works for me

### Steps to reproduce:
Install  beta-version from playstore, try starting.

### Current behavior:
Displays start screen and crashes

### Mentions:
@mikeller I provided the requested information.

mikeller, 2023-10-17:
@jfietz: Looks like your problem and a solution are described in #3916.

jfietz, 2023-10-18:
From logcat: ` 6707 10-17 23:46:03.840   466   466 E AndroidRuntime: java.lang.SecurityException: Permission Denial: starting Intent { act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 pkg=org.subsurfac      edivelog.mobile cmp=org.subsurfacedivelog.mobile/.SubsurfaceMobileActivity } from ProcessRecord{f235a57 466:com.android.vending/u0a161} (pid=466, uid=10161) not exported from uid 10477`

Looks like you'd need to set `android:exported` to true for the main activity? https://developer.android.com/privacy-and-security/risks/android-exported#:~:text=The%20android%3Aexported%20attribute%20sets,by%20its%20exact%20class%20name.

mikeller, 2026-10-17:
Hi @jaskev,

Thanks for the report. This has been fixed: Android 12+ (API 31+) requires any activity with an intent-filter to declare android:exported="true", and that was missing from our manifest. I've checked android-mobile/AndroidManifest.xml directly in current source and it's now present on the SubsurfaceMobileActivity main activity.

Closing this as fixed. Please reopen if the beta app still crashes shortly after launch for you on a current build.

closed 2026-10-17 by mikeller