subsurface/subsurface #3945: Unable to create cloud account / subsurface-divelog.org not responding

GitPuffy, 2023-07-23:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:

I am unable to register for the cloud service account

### Operating system:

macOS 12.6.5

iMac

Also behaves like this on iPhone iOS 16.5.1(c)

### Subsurface version:

5.0.10

Official release

### Steps to reproduce:

Attempt to register for a cloud account.  Never receive email to verify pin.  Used two different email accounts / email providers

### Current behavior:

Never receive the email to verify pin

### Expected behavior:

### Additional information:

I attempted to navigate directly to https://subsurface-divelog.org but it seems the site is not responding

### Mentions:

mikeller, 2023-07-23:
Yes, our website seems to be down / unreachable again. @dirkhh should be able to sort this out.

But independent of this the cloud storage for dive logs is working fine for me - have you done the obvious and checked your SPAM folders?

dirkhh, 2023-07-23:
power outage. And I was away from the house. Everything is back up again.
The cloud servers weren't affected since they are of course running in nice data centers... I need to move our website away from home as well... one of these days...

@GitPuffy - send me an email from the address you want to use for your cloud service. I can then verify if the account has been created and try to figure out why you aren't getting the PIN

GitPuffy, 2023-07-24:
Interesting.  Thank you for looking at your server / website.  I have received my verify PIN.  It was received at 5:00PM MDT.

For the interest of inquiring minds this is the raw header source of the email I received (with private details redacted):

> X-Spam-Checker-Version: SpamAssassin 3.4.2 (2018-09-13) on redacted
X-Spam-Level: 
X-Spam-Status: No, score=0.9 required=4.0 tests=BAYES_20,CUSTOM_DMARC_NONE,
	DKIM_SIGNED,DKIM_VALID,DKIM_VALID_AU,HTML_MESSAGE,URIBL_BLOCKED
	autolearn=no autolearn_force=no version=3.4.2
Received: from mailhub.gr8dns.org (mailhub.gr8dns.org [194.146.44.214])
	by redacted (Postfix) with ESMTP id 7437321B7F
	for <redacted>; Sun, 23 Jul 2023 17:00:05 -0600 (MDT)
Authentication-Results: name OpenDMARC; dmarc=none (p=none dis=none) header.from=subsurface-divelog.org
Authentication-Results: name OpenDMARC; spf=pass smtp.mailfrom=www-data@subsurface-divelog.org
Received: by mailhub.gr8dns.org (Postfix, from userid 1002)
	id 8A0FD2B3AD; Sun, 23 Jul 2023 16:00:04 -0700 (PDT)
DKIM-Signature: v=1; a=rsa-sha256; c=relaxed/relaxed;
	d=subsurface-divelog.org; s=2021; t=1690153204;
	bh=OC/u8Snu6lYXBCKuWIW0x9P+ae4dxdecqdC9eiO56TE=;
	h=From:To:Date:Subject:From;
	b=DPp1F1GgVZkkj053qzQf07oS2rtAtEtbnE4UF9XB/JxBve1uswc/2aZUq35dUGx6u
	 Gu1R8WCt6VyTsX5hXloVWHeOzt8eYc71a5xiVuFPTNLj4SIBhi0/b4fSRHLw7oJY4V
	 EKNku5TVjeiDbIwpSkK5yMUlZM2xyRg3oTnIxc3gpy9V7Ts9kmpPr0unbwOON8wRAK
	 57oE3vqKG8/lAe4TyRlKFwhRWLUS6WPNw6FvKBV/w54+lZ40ZE9lP8NGojW3drcin9
	 afwMwI8pwem+YjplGB+5ClbNB47Vb9oxolAEaiWTfcrq7FphLuUPk7SzXUSYhhUSKA
	 UagS8DRu2qTQA==
Received: from mail.subsurface-divelog.org (mailhost.subsurface-divelog.org [192.168.111.10])
	by mailhub.gr8dns.org (Postfix) with ESMTP id 2D05F2B3AC
	for <redacted>; Sun, 23 Jul 2023 16:00:03 -0700 (PDT)
Received: by mail.subsurface-divelog.org (Postfix, from userid 112)
	id 9C24920689; Sun, 23 Jul 2023 16:00:02 -0700 (PDT)
Received: from ssrf-cloud-01.lan (unknown [172.22.12.1])
	by mail.subsurface-divelog.org (Postfix) with ESMTP id 9239820184
	for <redacted>; Sun, 23 Jul 2023 16:00:01 -0700 (PDT)
Received: by ssrf-cloud-01.lan (Postfix, from userid 33)
	id 2D9D01F63C8; Sun, 23 Jul 2023 19:21:19 +0000 (UTC)
From: do_not_reply__@subsurface-divelog.org
To: redacted
Date: 23 Jul 2023 19:21:18 +0000
Subject: Subsurface Cloud Account PIN
MIME-Version: 1.0
Content-Type: multipart/alternative; boundary=asdfghjkl
Message-Id: <20230723192119.2D9D01F63C8@ssrf-cloud-01.lan>
>

mikeller, 2026-10-17:
Hi @GitPuffy,

Both parts of this turned out to be transient, not code bugs: the subsurface-divelog.org unreachability was a power outage at the hosting location (the cloud servers, hosted separately, were unaffected), and the missing verification PIN email was a spam-filtering false positive on your end, as the headers you forwarded showed it passed DKIM/SPF but still landed in spam. You confirmed receiving the PIN once you checked there.

Closing this as resolved. Please reopen if you hit the same "site not responding" symptom again and it isn't a transient outage.

closed 2026-10-17 by mikeller