subsurface/subsurface #3958: Unable to import database from shearwater peregrine

Mitusin, 2023-09-10:
When I try to import a database file created with Shearwater Cloud Desktop 2.10, no dives are imported.

### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
If I create a database with Shearwater Cloud Desktop 2.10 or the mobile app, the import (Import log files) does not do anything (no errors, no dialogs). I attach an example of database which is not imported.

### Operating system:
Windows 10.

### Subsurface version:

5.010

### Steps to reproduce:

Import the *.db file by means of the menu Import->Import log file

### Current behavior:

Nothing happens

### Expected behavior:

Dives should be imported.

### Additional information:

A dive log file is attached.
[shearwater_db.zip](https://github.com/subsurface/subsurface/files/12570392/shearwater_db.zip)

### Mentions:

mikeller, 2023-09-12:
I was able to replicate this behaviour with a .db file generated from my own logs with Shearwater Cloud 2.10.0 - the import 'succeeds' without an error message, but no dives are imported.
Based on this it looks like Shearwater has changed the format of their .db files in a way that results in our parser failing to extract the dives.
Does anybody have a link to Shearwater's specification of their latest .db file format?

yomgui1, 2023-10-02:
Got the same issue with a Perdix2 on same Subsuface version.
On GNU/Linux, when I run SS on a console, no  messages when I do the import.

This is the fullschema Sqlite3 output of my .db

```
CREATE TABLE IF NOT EXISTS "dive_details"(
  "DiveId" varchar primary key not null ,
  "FileName" varchar ,
  "LastModified" datetime ,
  "MainTimestampReliable" varchar ,
  "DiveDate" datetime ,
  "Depth" varchar ,
  "SerialNumber" varchar ,
  "DiveLengthTime" varchar ,
  "Location" varchar ,
  "Site" varchar ,
  "Buddy" varchar ,
  "DateAndTime" varchar ,
  "DiveNumber" varchar ,
  "Environment" varchar ,
  "EnvironmentNotes" varchar ,
  "Visibility" varchar ,
  "Weather" varchar ,
  "Conditions" varchar ,
  "Platform" varchar ,
  "AirTemperature" varchar ,
  "GPS_StartLocation" varchar ,
  "GPS_EndLocation" varchar ,
  "GPS_CustomWaypoints" varchar ,
  "Tank1PressureStart" varchar ,
  "Tank1PressureEnd" varchar ,
  "Tank2PressureStart" varchar ,
  "Tank2PressureEnd" varchar ,
  "Tank3PressureStart" varchar ,
  "Tank3PressureEnd" varchar ,
  "Tank4PressureStart" varchar ,
  "Tank4PressureEnd" varchar ,
  "AverageSAC" varchar ,
  "TankSize" varchar ,
  "GasNotes" varchar ,
  "Weight" varchar ,
  "GearNotes" varchar ,
  "Dress" varchar ,
  "Apparatus" varchar ,
  "ThermalComfort" varchar ,
  "Workload" varchar ,
  "Problems" varchar ,
  "Malfunctions" varchar ,
  "Symptoms" varchar ,
  "ExposureToAltitude" varchar ,
  "Notes" varchar ,
  "Other1" varchar ,
  "Other2" varchar ,
  "Other3" varchar ,
  "IssueNotes" varchar
);
CREATE TABLE IF NOT EXISTS "dive_log_records"(
  "id" integer primary key autoincrement not null ,
  "diveLogId" varchar ,
  "currentTime" integer ,
  "currentDepth" float ,
  "firstStopDepth" integer ,
  "ttsMins" integer ,
  "averagePPO2" float ,
  "fractionO2" float ,
  "fractionHe" float ,
  "firstStopTime" integer ,
  "currentNdl" integer ,
  "systemByte" integer ,
[... 396 lines cut ...]
CREATE TABLE IF NOT EXISTS "CustomDiveComputer"(
  "SerialNumber" bigint primary key not null ,
  "Version" bigint ,
  "UpdateTime" datetime ,
  "JsonData" varchar
);
CREATE UNIQUE INDEX "CustomDiveComputer_SerialNumber" on "CustomDiveComputer"(
  "SerialNumber"
);
CREATE TABLE IF NOT EXISTS "log_data"(
  "log_id" varchar primary key not null ,
  "table_version" bigint ,
  "created_unixtime" bigint ,
  "modified_unixtime" bigint ,
  "file_name" varchar ,
  "format" varchar ,
  "format_version" integer ,
  "data_bytes_1" blob ,
  "data_bytes_2" blob ,
  "data_bytes_3" blob ,
  "data_bytes_4" blob
);
CREATE INDEX index_for_log_data_log_id ON log_data(log_id);
/* No STAT tables available */
```

yomgui1, 2023-10-02:
Ok, I've done some code review (I'm a new user of subsurface... but I'm a developer) and found in [this file](https://github.com/subsurface/subsurface/blob/c5feae126b7f965ba33efddaa69d7635b042336e/core/file.c#L135) that a test is done by looking at a SQL table named "system". But this table seems to no longer exist  (in my .db version).
if this test fails, I looks like nothing special is done that explains the observed behaviour.

The best table that indicate the searched parameter is 'SWC_TableVersion'.

I'm going to try a custom build to verify this.

yomgui1, 2023-10-02:
I've succeed to make my own build and confirm that the sql test fails as suspected.
I'm able to pass the test but it's not enough and the application crashes later during the DB data handling:

```Database query failed 'xxxxxx/dive_data.db'.
xxxxxx/dive_data.db:1: parser error : Start tag expected, '<' not found
SQLite format 3
^
Erreur de segmentation (core dumped)
```

Investigation ongoing...

LouisSung, 2023-12-31:
Hi, I am new user to Subsurface who bumped into same issue
Just to share some findings when code review these days

1. according to [`subsurface/core/file.c#L138`](https://github.com/subsurface/subsurface/blob/v5.0.10/core/file.c#L138), the `shearwater_cloud_test[]` returns `1` (making it passed the test)
2. However, the SQL query that used by [`parse_shearwater_cloud_buffer`](https://github.com/subsurface/subsurface/blob/master/core/import-shearwater.c#L519) joins the tables `dive_details` and `dive_logs`, where **the table `daive_logs` is empty and no longer exists any log**
    > `select l.number,strftime('%s', DiveDate),location||' / '||site,buddy,notes,imperialUnits,maxDepth,DiveLengthTime,startSurfacePressure,computerSerial,computerModel,d.diveId,l.sampleRateMs / 1000 FROM dive_details AS d JOIN dive_logs AS l ON d.diveId=l.diveId`
3. The only place seems to store the log would be field `data_bytes_` in table `log_data`, which is BLOB (Binary Large Object) type (I guess it may be [PNF (Petrel Native Format)](https://subsurface.hohndel.narkive.com/sYbDKXoX/shearwater-teric)? but the parser [shearwater_predator.c](https://github.com/subsurface/libdc/blob/v5.0.10/src/shearwater_predator.c) seems to be really complex and need more time to dig in)

Screenshots:
1. `shearwater_cloud_test` passed 
    [image]
2. no data in `dive_logs` and therefore no row returned
    [image]
    [image]
3. table `log_data` that might store the actual data
    [image]

---
Env:
* Shearwater Cloud: 2.10.2
    > both `*.db` provided by @Mitusin and those exported with my macOS Desktop app and iOS mobile app have the same issue shown above
* Dive Computer: Petrel 3 with Firmware v97, Log version v14, DB version v11
     >details are revealed in the log using Shearwater Cloud
* Subsurface: 5.0.10