subsurface/subsurface #3959: Suit

jtslush, 2023-09-11:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

"Suit" option not available on PC. I can add the wetsuit type only using smartphone app.
### Operating system:
windows 11
surface pro

### Subsurface version:
 5.0.10

official

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

jtslush, 2023-09-11:
Can't choose wetsuit type

mikeller, 2023-09-11:
Have a look in the 'Equipment' tab:
[image]

jtslush, 2023-09-12:
I see that, but I can't and a suit in my logbook on PC.

mikeller, 2023-09-12:
Not sure what you mean - are you not able to edit the 'Suit' field in your setup?

This seems to be working just fine for me:

[image]

jtslush, 2023-09-12:
Thanks... I'll dig into it and see if I'm doing something wrong. Thanks for
your help.
Justin

jtslush, 2023-09-12:
Here's what mine looks like. No option to enter anything in the suit
section, like there is in weights and cylinders.
[image: image.png]

jtslush, 2023-09-12:
> Not sure what you mean - are you not able to edit the 'Suit' field in your setup?
> 
> This seems to be working just fine for me:
> 
> [image]

Correct, I cannot edit anything in the suit section.

mikeller, 2023-09-17:
Ok, this looks like problem happening only on the 'Surface Pro' version of Windows, as it seems to be working for other users. Unfortunately I do not have access to a device running this version, so I cannot reproduce to investigate, sorry.

jtslush, 2023-09-17:
Thanks... I will try it on a different computer.

mikeller, 2026-10-17:
Hi @jtslush,

Thanks for the report about the suit option not being available. Could you confirm whether this only happens with touch input on your Surface Pro (as opposed to mouse/keyboard), and give us the exact steps to reproduce it, which screen you're on and what you're tapping/clicking when the option doesn't appear?

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.

jtslush, 2026-11-04:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!
> <subsurface/subsurface/issues/3959/6238487217 ***@***.***

Hi Michael,

Thanks for getting back to me. I only use the Surface Pro with a keyboard
and mouse. See the screenshot below. You can see in the  "Suit" section,
there is no drop down to select my saved wetsuits as it does on the the
android mobile version.  I have to type it in on the Windows version;
however, you can see that "weights" and "cylinders" allow for drop-down
options.

[image: image.png]

Thanks,

Justin

mikeller, 2026-11-04:
AI-drafted; reviewed by Michael.

Hi @jtslush, thanks for the screenshot and for confirming you're using a keyboard and mouse.

It looks like this is a difference between the desktop and mobile versions rather than something missing on your Surface Pro. On desktop, the Suit field is a text box with auto-complete, not a drop-down list like the one on Android. Cylinders and weights have drop-down lists because they have their own tables. There is no separate list of saved suits.

To pick a previous suit on the desktop:

1. Select a dive and open the Equipment tab.
2. Click in the Suit field and start typing the first letters of a suit you've used before, e.g. "We".
3. A suggestion list should pop up below the field. Click your suit, or press Enter, to accept it.
4. Make sure the dive is saved afterwards.

The suggestions come from the Suit values already stored on dives in your current logbook. If you've only entered suits on your phone and those dives haven't been synced to the logbook open on your PC, nothing will be suggested. The first time you type a new suit on the desktop, you have to enter it in full. It is offered as a suggestion from then on.

The manual describes this briefly in the Equipment tab section: "Here the type of dive suit used can be entered. Auto selection of the suit description is available." (https://subsurface-divelog.org/documentation/, "Subsurface User Manual").

I agree it isn't obvious that there's no drop-down arrow, and your feedback helps with that. If this works for you, a short note here would be great. If you don't get any suggestions after typing the first letters of a suit that is already in your log, please let us know your exact Subsurface version (Help → About) and whether those dives show the suit in the Equipment tab.