subsurface/subsurface #3960: Dive mode being changed by selecting trips in filtered view

kruegerha, 2023-09-19:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I'm using an extensive logbook including open and closed circuit dives, all filed under numerous "trips".
To separate OC and CCR, I use a filter, which acts on a specific tag.
When using the filtered view, selecting individual dives, everything seems ok. However, if a "trip" is selected, all dives filed under this trip are surprisingly changed to mode "open circuit" and also the "Warning: edited # dives" appears. Immediately selecting "undo" prevents further damage. Failing to immediately undo this, has the potential to mess up the logbook of unwary users.

### Operating system:
linux

### Subsurface version:
origin/master  97a76a6615f3a4451b3f3d71fd4861ceb0a66f9e
no other versions tested yet

### Steps to reproduce:
I've produced a stripped subset of my logbook, which still reproduces the problem.
open the logbook, browse dives. note the the trip "See" has OC and CCR dives.
Then select filtered view, load filter-set "revo-total", select the trip "See" and note that all dives in this trip have been changed to mode OC.
[testlog.xml.gz](https://github.com/subsurface/subsurface/files/12659299/testlog.xml.gz)

### Current behavior:

### Expected behavior:
never change dive mode unexpectedly

### Additional information:

### Mentions:

kruegerha, 2024-01-16:
In my usage of subsurface this bug is an annoyance. It is still present up to today (commit 4bab38b).
Any ideas where to look for it?  what is the relevant code/source file, event triggered by selecting a "trip"?

dirkhh, 2024-01-16:
That does seem like a serious issue. I will try to find time to look at this.
Also, @bstoeger - any idea what this could be?

bstoeger, 2024-01-16:
Can reproduce, will look at this after dinner. Should be easy to debug.

dirkhh, 2024-01-16:
> Can reproduce, will look at this after dinner. Should be easy to debug.

Awesome. Thanks!

dirkhh, 2024-01-16:
So that's the ancient and beloved multi dive edit. 
But it should be "if I have multiple dives selected and then edit something, Subsurface is smart about how it edits all the selected dives"

This sounds like the user selected a trip and suddenly dives are edited. That seems very different, no?

bstoeger, 2024-01-16:
Here's a quick fix:
```
diff --git a/desktop-widgets/tab-widgets/TabDiveInformation.cpp b/desktop-widgets/tab-widgets/TabDiveInformation.cpp
index 58cf7ecd5..e2c64820d 100644
--- a/desktop-widgets/tab-widgets/TabDiveInformation.cpp
+++ b/desktop-widgets/tab-widgets/TabDiveInformation.cpp
@@ -13,6 +13,8 @@
 #include "core/statistics.h"
 #include "core/divelist.h"
 
+#include <QSignalBlocker>
+
 #define COMBO_CHANGED 0
 #define TEXT_EDITED 1
 
@@ -375,8 +377,10 @@ void TabDiveInformation::on_chill_valueChanged(int value)
 void TabDiveInformation::updateMode()
 {
        divecomputer *currentDC = parent.getCurrentDC();
-       if (currentDC)
+       if (currentDC) {
+               QSignalBlocker blocker(ui->diveType); // Don't send signals when updating the combobox
                ui->diveType->setCurrentIndex(currentDC->divemode);
+       }
 }
 
 void TabDiveInformation::diveModeChanged(int index)
```
Not sure how happy I am about that, but it does the job. A fundamental question is: what should the dive mode display if multiple dives are selected? Especially if they have multiple dive modes.

dirkhh, 2024-01-16:
I'm guessing 'blank' ?

dirkhh, 2024-01-16:
This will be included in the CICD release 5061 - depending on which Linux flavor you use, how you can get that will be somewhat different.

closed 2024-01-16 by dirkhh

bstoeger, 2024-01-18:
Feel free to contact me directly with such issues if I miss the report. Currently I am so busy that I don't look at the issue tracker on a regular basis.

PR #4047 (2024-01-16): desktop: block signals when setting combobox-indexes
On the InformationTab a signal is emitted when programatically setting the index of the dive mode combobox which in turn then edits the dive.

Usually not a problem, because the editing code realizes that the value is not changed. It is however a problem when multiple dives are selected.

Therefore, block the signals when setting the index. Same for the other comboboxes on the same page.

Fixes #3960.

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Do [... cut]
commit 35ff6eea (2024-01-16): desktop: block signals when setting combobox-indexes

On the InformationTab a signal is emitted when programatically
setting the index of the dive mode combobox which in turn then
edits the dive.

Usually not a problem, because the editing code realizes that
the value is not changed. It is however a problem when multiple
dives are selected.

Therefore, block the signals when setting the index. Same for
the other comboboxes on the same page.

Fixes #3960.