subsurface/subsurface #3969: Feature: Mobile app. view dive plots from multiple dive computers

ps68060, 2023-10-09:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ X] New feature request
- [ ] Discussion request

### Issue long description:
Mobile app only shows the graph of one computer.  It would be good to be able to switch the grpah between the different computers like the desktop version.

### Operating system:
Android 10 - English
Moto G7

### Subsurface version:
3.4.7 (5.0.9.35)

### Steps to reproduce:
Import data for 1 dive from 2 or more dive computers.
Open the dive to see the dive plot/graph.

### Current behavior:
The plot/graph only shows data from one dive-comp. I think it is the last one imported.

### Expected behavior:
I would like to be be able to swipe to show the graph from each comp. for the same dive.

### Additional information:

### Mentions:

mikeller, 2026-10-18:
Hi @ps68060,

The underlying data model already supports multiple dive-computer records per dive (desktop's existing switcher proves that), so this would mainly be a mobile UI addition rather than new data-model work. Are you still interested in this? It's a reasonable, low-moderate-complexity feature-parity gap, but we haven't had any other reports of it, so confirming continued interest would help us prioritise it.

ps68060, 2026-10-18:
Hello
The mobile app recently added two buttons for previous and next DC so this feature is now available. 

Thanks.

closed 2026-10-18 by ps68060