subsurface/subsurface #3971: Fresh install / Can't get the files on cloud

emigeoant, 2023-10-10:
### Describe the issue:

- [x ] Bug

### Issue long description:
Fresh install on Windows 10.  Cannot connect to the cloud :  Message "wrong password". I can connect with an other PC under Windows with the same version  and on an android mobile (  3.4.7).  Checked my password on both PC and Android.  As I suspected a firewall issue I disable the firewall with no result.

### Operating system:
Windows 10
### Subsurface version:
5.0.10 just downloaded

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2023-10-10:
there are a couple of explanations I can see:
- you mistyped your email or password. Yes, I know you said you checked. You would not believe how many times we have gone down that road, and then the user responds "err, oops, I meant..."
- you have network issues. Can you connect with your browser to `https://cloud.subsurface-divelog.org` ?
- you are on a corporate network or with some ISP that blocks certain ports for outbound connections - is that other PC under Windows that works on the same network?

emigeoant, 2023-10-12:
Thank you for your answser.  I  manage to connect at the end  without changing anything in the configuration.  Here is the explanation : The menu to upload the current existing  data does not change color until the 1st upload is done. So this is misleading to the user as it seem that the operation is not authorized.   I typed on it by mistake and everything went well. Hope this helps.