subsurface/subsurface #3977: Problem: Cannot download from Suunto Stinger

ps68060, 2023-10-20:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Tried to download from a Suunto Stinger using a third-party usb interface adapter.
The adapter is recognised but no dives are listed or downloaded.

### Operating system:
Android 13
Moto  Edge 30 Fusion

### Subsurface version:
3.4.7 (5.0.9.35)
Official release

### Steps to reproduce:
Plug in usb adapter it is recognised.
Put Stinger into Transfer-PC mode.
Open Subsurface and the Suunto Stinger can be selected in the dropdown menus.
 
### Current behavior:
Initiate the download but no dives are listed.
They cannot be downloaded.
Eventually the Stinger times-out.

### Expected behavior:
Iit should list the dives as  normal so they can be downloaded.

### Additional information:

### Mentions:

ps68060, 2023-10-20:
Additional info:
Using an OTG cable between the phone and the Stinger interface.

In SubSurface, the Connection field shows "SmartInterface [2:2]"

dirkhh, 2023-10-20:
A lot of things could be causing this. The USB serial chip in the cable could be unsupported by the user space drivers we use on Android, there could be connector issues (those seem very common with early Suuntos), the battery could be too low (the required voltage for successful data download is higher than that for it to turn on / appear to function during a dive). Etc.

ps68060, 2023-10-24:
The cable works with a laptop and Subsurface and data can be downloaded.

With the mobile version, the serial number of the Stinger is correctly reported, so some communication is working. This suggests that the serial chip in the adapter cable is handled, at least partly.

atdotde, 2023-10-25:
I have a SmartInterface cable as well. They use a run of the mill FTDI chip. If that isn't supported what is? Does the LED on the cable blink in red or green? Can you send the Subsurface log file?

ps68060, 2023-10-27:
My friend reports that the LEDs flash:
    RED/GREEN/RED/GREEN
i.e. two of each colour then nothing.

Information about the cable may be found here:
https://en.smartinterface.de/artikel.php?id=39&kat=1

For the Stinger+Android this indicates that the cable is only compatible with s/w called "DiveMate".

User downloaded this app and tried an import.
This worked (with much LED flashing). He could view the dives in this app.

What this seems to show is that the Stinger, USB OTG cable and Android are ok.

ps68060, 2023-10-29:
[subsurface.txt](https://github.com/subsurface/subsurface/files/13197718/subsurface.txt)

mikeller, 2026-10-17:
Hi @ps68060,

Thanks for the log you provided for this, apologies that it hasn't been analysed yet. Before we dig into it, could you confirm whether you're still seeing this issue with your Suunto Stinger on a current build? If so, we'll aim to work through the log and get back to you with findings within the next few weeks; if it's resolved itself or you've found a workaround, let us know that too so we can close this out accurately.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed/stale, but please comment or reopen whenever suits and we'll pick it back up.