subsurface/subsurface #3978: Current sources fail to compile

pahhur, 2023-10-21:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The sources that were downloaded just some minutes ago (by the yay git component) are not compileable

.../subsurface-git/src/subsurface/core/configuredivecomputerthreads.cpp:2221:46: error: too many arguments to function ‘dc_status_t hw_ostc3_device_fwupdate(dc_device_t*, const char*)’
 2221 |                 rc = **hw_ostc3_device_fwupdate**(m_data->device, qPrintable(m_fileName), m_forceUpdate);
      |                      ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ....cache/yay/subsurface-git/src/subsurface/core/configuredivecomputerthreads.cpp:4:
/usr/include/libdivecomputer/hw_ostc3.h:58:1: note: declared here
   58 | **hw_ostc3_device_fwupdate** (dc_device_t *abstract, const char *filename);
      | ^~~~~~~~~~~~~~~~~~~~~~~~

### Operating system:

Arch linux, up to date
6.5.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 13 Sep 2023 08:37:40 +0000 x86_64 GNU/Linux
gcc version 13.2.1 20230801 (GCC) 

### Subsurface version:

It should be the latest 

### Steps to reproduce:

See the cmake, the package builder did in
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=subsurface-git

### Current behavior:

see error message above

### Expected behavior:

I expect the sources that are downloaded here to be compileable

### Additional information:

### Mentions:

pahhur, 2023-10-22:
Ok, the problem was caused by a non-updated subsurface-libdc-git package.
Not really sure, why the package system hasn't updated that package when I requested so.

The only point that is remaining now is that at startup, subsurface claims

**The server returned the following information:
["You are running an unknown product version 5.0.10.234, latest released version is 5.0.10.223"]**

but I guess that's not so bad at all.

atdotde, 2023-10-23:
So this is resolved.

closed 2023-10-23 by atdotde