subsurface/subsurface #3979: Support for Android 14

D-Vanditmars, 2023-10-29:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

Support for Android 14

### Operating system:

Android 14

Pixel 5

Not sure if Android 13 was supported

### Subsurface version:

Latest one as of Oct 20, 2013 from Google play store, (app does not work with this version of OS)

Do not know

Official release

N/A

### Steps to reproduce:

### Current behavior:

Android 14 Google Play store does not app to be installed

### Expected behavior:

The would install

### Additional information:

### Mentions:

matasbbb, 2024-12-05:
Request to bump target version to API 35

dirkhh, 2024-12-05:
Please check https://subsurface-divelog.org/current-release and install the APK provided there. 

The Google Play store has become an impossible target for an all volunteer open source project, so we are providing our app for sideloading instead.

matasbbb, 2024-12-05:
Not sure if it is due to specific accounts added, APK cannot be sideloaded due target SDK being 30. For Android 15, at least on my device it complains that target SDK should be updated.

dirkhh, 2024-12-05:
Hmm. I'm running the apk on a Pixel Fold with Android 15. So... not sure what to tell you.
We can certainly look into bumping the API target version.
As with everything else in the Android ecosystem, all of this is designed to create maximum pain and frustration for the developers who aren't using the Google provided tools and Google approved libraries...

D-Vanditmars, 2024-12-05:
OK, thanks for the explainations, I was using the old Google Play app and not side loading the APK...

mikeller, 2026-10-17:
Hi @D-Vanditmars,

Two things from this thread: the original Play Store install problem was because the Play Store listing was outdated, as you confirmed yourself: "OK, thanks for the explanations, I was using the old Google Play app and not side loading the APK." Sideloading a current APK from https://subsurface-divelog.org/current-release/ is the recommended path. The separate target-SDK/Android-15 concern raised by another commenter has since been addressed by our Android API 36 migration.

Closing this as resolved. Please reopen if a current sideloaded build still won't install or run on your Android version.

closed 2026-10-17 by mikeller

mikeller, 2026-10-17:
Hi @D-Vanditmars,

Quick correction to my comment above: I said the Play Store listing was outdated, that's no longer accurate. A current build of Subsurface-mobile is available again through the Play Store now, so sideloading isn't the only way to get an up-to-date Android install any more. It's still a fine option if you prefer it, just not the only path.

Not reopening this, just didn't want the earlier comment to leave anyone thinking the Play Store was a dead end.