subsurface/subsurface #3980: Subsurface freezes on startup

pjsanfil, 2023-10-31:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
On Windows 10 Subsurface 5.0.10 freezes on startup with the following command line output:

`QObject::connect(QQuickWindow, QDeclarativeGeoMap_QML_6): invalid nullptr parameter `

Subsurface displays an empty window titled: Subsurface (Not Responding).

### Operating system:
Windows 10 Home
language: English

### Subsurface version:
5.0.10
Official release

### Steps to reproduce:
1. Start Subsurface

### Current behavior:
Subsurface displays an empty window titled: Subsurface (Not Responding).

### Expected behavior:
Subsurface starts and displays the dive log.

### Additional information:

### Mentions:

mikeller, 2023-11-01:
Have a look at https://github.com/subsurface/subsurface/issues/3363#issuecomment-1000600469 - the error you are experiencing can be caused by a corrupted Subsurface configuration, and in this case switching to a new configuration will solve the issue.

pjsanfil, 2023-11-02:
Duplicate of #3363

closed 2023-11-02 by pjsanfil