subsurface/subsurface #3988: Dive Numbers are not shown if dive is part of a trip

marcvs, 2023-11-08:
### Describe the issue:
- [x] Bug
- [x] Change request

### Issue long description:
Dive numbers are not shown in trips

### Operating system:
- OS: Debian Linux, `subsurface-beta_202310210411-1`
- Laptop

### Subsurface version:
-  `subsurface-beta_202310210411-1`
- provided from you apt repo

### Steps to reproduce:
- add dive to trip
- observe dive number not shown in overview

### Expected behavior:
- show the dive number (as the android app correctly does)

The-Lid, 2024-01-20:
I am seeing the same problem.

OS: Windows 11
Subsurface version 6.0.5054-CICD-release

Please advise when this has been rectified.

dirkhh, 2024-01-20:
How about a screen shot?
My guess is that your first column is too narrow.

The-Lid, 2024-01-20:
OK, I feel a "bit" silly now.  It was the column width, sort of, but all the dive numbers under trips are indented so they're not visible with the default column width.

[image]

dirkhh, 2024-01-20:
Arguably that UI could be cleaner - but this all is implemented in the toolkit that we use; I'm not sure how much control we'd even have to change that.

closed 2024-01-20 by dirkhh

PeterZaal, 2025-05-11:
Had the same issue after updating Subsurface on Windows, came her to find the solution to make the column wider.
I'm sure this behavior has changed somehow, the indent of the dive number on trips wasn't there before.