subsurface/subsurface #3989: CREST4/Genesis_Centauri/DeepSix_Excursion unable to import logs

pyhoff, 2023-11-09:
After many tries we may be successful after 20 to 30 tries, it's  random. Once it works you have on shot to download the logs.  Happens on iPhone, Mac and Windows, App Images on Ubuntu. I have tried everything except Android.

On iPhone I had a trick where I connect DiveStory close it and then Subsurface was able to use this connection, must have been an iOS exploit or bug that is now patched/ So now left with Windows PC for logging. 

Latest error logs:
``
INFO: dev_info: No new dives downloaded from dive computer
INFO: dev_info: Connecting to BLE device LE:60:77:71:22:CF:FF
Unable to get value for characteristic "{f000ffe2-ab12-45ec-84c8-46483f4626e9}" of the service "{f000ffe0-ab12-45ec-84c8-46483f4626e9}" : "No attribute found within the given attribute handle range."
Unable to get value for characteristic "{f000ffe3-ab12-45ec-84c8-46483f4626e9}" of the service "{f000ffe0-ab12-45ec-84c8-46483f4626e9}" : "No attribute found within the given attribute handle range."
Unable to set value for descriptor "{00002902-0000-1000-8000-00805f9b34fb}" for characteristic "{f000ffe1-ab12-45ec-84c8-46483f4626e9}" of the service "{f000ffe0-ab12-45ec-84c8-46483f4626e9}" : "Access is denied."
INFO: dev_info: Connecting ...
INFO: dc_device_open error value of 0
INFO: dev_info: Starting import ...
Unable to set value for characteristic "{f000ffe1-ab12-45ec-84c8-46483f4626e9}" of the service "{f000ffe0-ab12-45ec-84c8-46483f4626e9}" : "Access is denied."
[2.523121] ERROR: Failed to receive the packet. [in src/deepsix_excursion.c:128 (deepsix_excursion_recv)]
[2.523138] ERROR: Failed to load the settings. [in src/deepsix_excursion.c:266 (deepsix_excursion_device_foreach)]
INFO: dev_info: No new dives downloaded from dive computer
INFO: dev_info: Connecting to BLE device LE:60:77:71:22:CF:FF
Unable to get value for characteristic "{f000ffe2-ab12-45ec-84c8-46483f4626e9}" of the service "{f000ffe0-ab12-45ec-84c8-46483f4626e9}" : "No attribute found within the given attribute handle range."
Unable to get value for characteristic "{f000ffe3-ab12-45ec-84c8-46483f4626e9}" of the service "{f000ffe0-ab12-45ec-84c8-46483f4626e9}" : "No attribute found within the given attribute handle range."
Unable to set value for descriptor "{00002902-0000-1000-8000-00805f9b34fb}" for characteristic "{f000ffe1-ab12-45ec-84c8-46483f4626e9}" of the service "{f000ffe0-ab12-45ec-84c8-46483f4626e9}" : "Access is denied."
INFO: dev_info: Connecting ...
INFO: dc_device_open error value of 0
INFO: dev_info: Starting import ...
Unable to set value for characteristic "{f000ffe1-ab12-45ec-84c8-46483f4626e9}" of the service "{f000ffe0-ab12-45ec-84c8-46483f4626e9}" : "Access is denied."
[2.518469] ERROR: Failed to receive the packet. [in src/deepsix_excursion.c:128 (deepsix_excursion_recv)]
[2.518480] ERROR: Failed to load the settings. [in src/deepsix_excursion.c:266 (deepsix_excursion_device_foreach)]
INFO: dev_info: No new dives downloaded from dive computer
``
Tested 5.0.7 - 5.0.10 it's random. (Note Sons Dive computer) My Shearwater works s expected on all version. has no issue on any device. See attached logs
[subsurface_err.log](https://github.com/subsurface/subsurface/files/13303595/subsurface_err.log)
[subsurface_out.log](https://github.com/subsurface/subsurface/files/13303596/subsurface_out.log)

jefdriesen, 2023-11-09:
This shows the same symptoms as reported in PR #3956: for some reasons enabling BLE notifications fails, which means we can't receive data packets.

Can you try to download with the dctool command-line client to check if enabling notifications works there? That will help figuring out whether the problem is in the Windows BLE stack or in the Qt BLE library.

https://libdivecomputer.org/builds/experimental/windows/dctool-ble.exe

Execute with these options:
```
dctool-ble.exe -vv -l excursion.log -f excursion download -o dive.%n.bin -f raw -t ble
```
And attach the excursion.log file to this bug report.

pyhoff, 2023-11-09:
Here you go, hope this works
[excursion.log](https://github.com/subsurface/subsurface/files/13307475/excursion.log)

pyhoff, 2023-11-09:
I was looking at compiling a daily build on my Linux box (PopOS, Ubuntu with flatpak) , but I can't seem to get Qt5 to work. I noticed in the macOS instruction you do mention we can try Qt6. 
The mac I have is work, don't like to mix Church and state if you know what I mean

jefdriesen, 2023-11-15:
The download failed with:
```
[0.691721] ERROR: The process cannot access the file because it is being used by another process (32) [in src/ble-win32.c:401 (win32_ble_open)]
[0.712971] ERROR: Failed to open the BLE service. [in src/ble.c:738 (dc_ble_open)]
```
There is already another application connected to the dive computer. You'll have to find which application (could be some generic app that just tries to connect to every bluetooth device) and close that first.

paminhoff, 2023-11-30:
Sorry I took so long to reply. I actually was able to download one of the daily macos dmg builds from Nov-4th and I was able to sync without issue. The version from Nov-06 crashed.

vahid76758, 2024-02-06:
Thank you for contacting GoTyme Bank!

We're glad to acknowledge receipt of your email and strive to address your concern within 3 hours. We are working on it now and we'll get back to you at the earliest possible time.

While you wait, you can check our self-help FAQ found on our website or for urgent concerns feel free to connect with our Customer Success Team (Customer Service) by calling #GO8888. We are open 24 hours a day, 7 days a week and our team will be more than happy to assist you. Have a great day!

Kind Regards,
Customer Success Team

Hotline: #go8888
Email: help@gotyme.com.ph 
In-App Chat: GoTyme Bank Mobile App