subsurface/subsurface #3991: Dive statistics are not showing data for latest date in selection

kittykat, 2023-11-14:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The dives from the most recent dates aren't showing on the dive stats graph. e.g. if the selection is from 3rd Nov to 11th Nov, the graph shows only dives from 3rd Nov to 10th Nov and the rightmost x axis point at the end of the graph is 11th Nov.

[image]

### Operating system:

Arch Linux (up to date)

### Subsurface version:

Subsurface-5.0.10-x86_64.AppImage

### Steps to reproduce:

1. Select some dives
2. "View" -> "Dive Statistics"
3. "Base variable" -> "Binning" -> "None"
4. "Restriction" -> "Restrict to selection"

### Current behavior:

Last date is not shown on the graph

### Expected behavior:

All selected dives are shown on the graph

### Additional information:

### Mentions:

jwobser, 2023-12-10:
Duplicate of #3544

Reproduced with AppImage version 5.0.10 using dives/abitofeverything.ssrf and selecting the heart rate test dives.

Looks like this was fixed in 5bd5ff2c2d0c4cef8a6895f07508e6706f66ce45.
### Before
[image]

### After
[image]

bstoeger, 2024-03-31:
Sorry, only saw this now.

For some reason I can't reproduce this. Could you post a minimal log that shows the problem? Maybe it has to do with the particular date range...?

While experimenting, I found two other bugs though. Sometimes the date scale is weird and shows only the year. And once the tooltip showed the wrong dives.

mikeller, 2026-10-17:
Hi @kittykat,

Thanks for reporting that the statistics view doesn't show data for the latest date in your selection. To reproduce this we'd need the exact steps, which date range/filter you had selected, and ideally a sample logbook (or dive count/date range) that shows the gap, since we haven't been able to reproduce it without more specifics.

If we don't hear back within the next 4-6 weeks we'll close this as unconfirmed, but please reply or reopen whenever convenient.