subsurface/subsurface #4008: Login issue with subsurface mobile iOS

Julaye, 2023-12-30:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Everything in the title ... I try to connect to the cloud. It's working on MacOS version and it's also working on the web viewer (https://cloud.subsurface-divelog.org/user/julie@aceblue.fr/dives.html)

But on iOS, the same couple of email and password is not working since few hours.

Could be related to some recent change on Cloud servers ? As the Subsurface mobile iOS version has more than one year ...

### Operating system:
iOS 17.1.2
iPhone 12

### Subsurface version:
Subsurface-Mobile iOS 3.4.7

### Steps to reproduce:
Login to cloud ...

### Current behavior:
Error message "couple email/passwd invalid"

### Expected behavior:
Cloud logged

### Additional information:
Same email/passwd working on MacOS and web viewer
Login was working few hours ago and suddenly stop working 30 December 2023 around 19:00 (not sure) Paris Time

### Mentions:
[subsurface.log](https://github.com/subsurface/subsurface/files/13799363/subsurface.log)

dirkhh, 2023-12-30:
Well, the log file makes it pretty clear:
```
Received error response from cloud storage backend: "Error transferring https://ssrf-cloud-eu.subsurface-divelog.org//storage - server replied: Service Unavailable"
Completed connection with cloud storage backend, response "<!DOCTYPE HTML PUBLIC \"-//IETF//DTD HTML 2.0//EN\">\n<html><head>\n<title>503 Service Unavailable</title>\n</head><body>\n<h1>Service Unavailable</h1>\n<p>The server is temporarily unable to service your\nrequest due to maintenance downtime or capacity\nproblems. Please try again later.</p>\n<hr>\n<address>Apache/2.4.52 (Ubuntu) Server at ssrf-cloud-eu.subsurface-divelog.org Port 443</address>\n</body></html>\n"
```
Should be fixed

closed 2023-12-30 by dirkhh

Julaye, 2023-12-31:
It's working NOW. Thanks.