subsurface/subsurface #4013: Unreasonable error message about Internet connection

knabed, 2024-01-03:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When I do a manual search for a new version I get the following error:

```
Subsurface konnte nicht nach neuen Versionen suchen.

Fehlermeldung erhalten:
Error transferring http://updatecheck.subsurface-divelog.org/updatecheck.html?os=osx&version=5.0.10.293&uuid= - server replied: Bad Gateway

Bitte überprüfe deine Internetverbindung.

```
This is a wrong error message, as I can access my cloud dive log, I get the map within subsurface, can call website within the about dialog and in general my internet connection works fine. When I paste the above URL "updatecheck.subsurface-divelog.org..." I get a valid reply.

### Operating system:

macOS 14.2.1

MacBook Pro M1

### Subsurface version:

Subsurface 5.0.10-293-g60f1deb11681

Since the rolling updates

Official download

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2024-01-03:
Thanks for the report.
I can reproduce the issue.

We have stopped making new releases and switched to a rolling release model: https://subsurface.github.io/en/release/2023/12/release-changes/
I tried to change the update function to also reflect that, but clearly messed something up after my last test.
I'll fix this.

dirkhh, 2024-01-08:
This should be fixed now. Please let me know if it works for you.

knabed, 2024-01-08:
Yes, works. Thanks!

closed 2024-01-08 by dirkhh