subsurface/subsurface #4014: Subsurface crashes when double clicking on gas or dive mode in planner

ejrobinson, 2024-01-04:
### Describe the issue:

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When planning a dive, double clicking or trying to use the dropdown for gas or dive mode in the dive planner points window causes subsurface to hard crash after a short delay
### Operating system:

Win 10.0.19045 Build 19045
### Subsurface version:

5.0.10-290-g8992ca629d71
Also happens on the last few 'nighly' builds
### Steps to reproduce:

Go to plan a dive, the double click either the gas or dive moce in the dive planner points window (so possibly something to do with drop down lists)
### Current behavior:

Crash
### Expected behavior:

Display drop down menu
### Additional information:

### Mentions:

ejrobinson, 2024-01-05:
After a bit more investigation, it seems like sometimes it's stable while the box is selected, and its when clicking off the box again the crash occurs.

dirkhh, 2024-01-16:
Windows crash information is just spectacularly useless. I always find it surprising how people get any work done on Windows, TBH.
Thanks for submitting it, though - clearly not your fault that Windows just doesn't even include a stack trace (literally the very first thing EVERY developer asks for after EVERY crash).

I haven't used Windows in earnest in more than two decades - but I finally broke down and ordered a tiny Windows 11 PC so I can maybe reproduce some of these bugs myself and see if "on device" I can get better debug information.

bstoeger, 2024-01-16:
To be fair: The Linux backtrace is just as useless:

```
#0  0x00007ffff5577f4b in QObject::connect(QObject const*, char const*, QObject const*, char const*, Qt::ConnectionType) ()
   from /lib/x86_64-linux-gnu/libQt6Core.so.6
#1  0x00007ffff74a1d17 in QComboBox::setModel(QAbstractItemModel*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6

Fatal signal: Segmentation fault
----- Backtrace -----
0x5593e307d8c6 ???
0x5593e31a9fdd ???
0x5593e31aa1b3 ???
0x7f70ade4290f ???
        ./signal/../sysdeps/unix/sysv/linux/x86_64/libc_sigaction.c:0
0x5593e304c5d8 ???
0x5593e30fa136 ???
0x5593e31e5e8f ???
0x5593e30a6f3e ???
0x5593e34c031c ???
0x5593e33c3411 ???
0x5593e33c930d ???
0x5593e33caf0b ???
0x5593e33cd48b ???
0x5593e30b7db4 ???
0x5593e342fa67 ???
0x5593e31aaa47 ???
0x5593e31ac243 ???
0x5593e31ab432 ???
0x7f70af1c877c ???
0x5593e31ab595 ???
0x5593e31ab763 ???
0x5593e3472b7c ???
0x5593e363e7e5 ???
0x5593e363f36d ???
0x5593e3292a79 ???
0x5593e3295944 ???
0x5593e2fc2deb ???
0x7f70ade280cf __libc_start_call_main
        ../sysdeps/nptl/libc_start_call_main.h:58
0x7f70ade28188 __libc_start_main_impl
        ../csu/libc-start.c:360
0x5593e2fd10b4 ???
0xffffffffffffffff ???
---------------------
```

bstoeger, 2024-01-16:
OK, after recompilation, I got a good one:
```
#0  0x00007ffff5577f4b in QObject::connect(QObject const*, char const*, QObject const*, char const*, Qt::ConnectionType) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#1  0x00007ffff74a1d17 in QComboBox::setModel(QAbstractItemModel*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#2  0x00005555557fe40c in ComboBoxDelegate::createEditor (this=0x555556560070, parent=<optimized out>, index=...) at /home/bs/src-qt6/subsurface/desktop-widgets/modeldelegates.cpp:109
#3  0x00007ffff7622029 in ?? () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#4  0x00007ffff7619214 in QAbstractItemView::edit(QModelIndex const&, QAbstractItemView::EditTrigger, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#5  0x00007ffff76160c7 in QAbstractItemView::mouseDoubleClickEvent(QMouseEvent*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#6  0x00007ffff73d1933 in QWidget::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#7  0x00007ffff745d396 in QFrame::event(QEvent*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#8  0x00007ffff553459a in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#9  0x00007ffff738b912 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#10 0x00007ffff738416e in QApplication::notify(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#11 0x00007ffff55369a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#12 0x00007ffff7380d78 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
   from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#13 0x00007ffff73df5b5 in ?? () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#14 0x00007ffff73e08f5 in ?? () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#15 0x00007ffff738b922 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Widgets.so.6
#16 0x00007ffff55369a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#17 0x00007ffff6394cb0 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /lib/x86_64-linux-gnu/libQt6Gui.so.6
#18 0x00007ffff63ddfcc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt6Gui.so.6
#19 0x00007ffff11b4f4e in ?? () from /lib/x86_64-linux-gnu/libQt6XcbQpa.so.6
#20 0x00007ffff4113b2c in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#21 0x00007ffff416f46f in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007ffff4111d20 in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007ffff5724420 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#24 0x00007ffff5540cea in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#25 0x00007ffff553988a in QCoreApplication::exec() () from /lib/x86_64-linux-gnu/libQt6Core.so.6
#26 0x000055555572db44 in main (argc=<optimized out>, argv=<optimized out>) at /home/bs/src-qt6/subsurface/subsurface-desktop-main.cpp:110
```

bstoeger, 2024-01-16:
The problem seem to be that `DivePlannerWidget` is initialized before `PlannerWidgets`, but the former needs the models of the latter. Weird.

dirkhh, 2024-01-16:
This should be fixed in the [next CICD build](https://github.com/subsurface/nightly-builds/releases/tag/v6.0.5060-CICD-release). The release page is ready but not yet populated - please check back in about 15 minutes. There should be a file 
`subsurface-6.0.5060-CICD-release-installer.exe` at the bottom of that page in the Assets list. Please give that one a try.

closed 2024-01-16 by dirkhh

jspg, 2024-01-17:
Tried the `subsurface-6.0.5060-CICD-release-installer.exe` build, it works! Thanks for the quick resolution!

ejrobinson, 2024-01-17:
Fab, fixed here as well. 

Thank you for all the work you put into subsurface. I've been using it as a logbook for years, but recently also find myself almost ready to ditch multideco for subsurface as a planner given how much more visual and user friendly it is, just one or two shortfalls in CCR planning to go. 

On that note, is it more cost effective for us to support the project through Ko-fi, or GitHub fees wise?

dirkhh, 2024-01-17:
Glad this is fixed.

At this point all the donations go to me (the primary maintainer and the person who runs the infrastructure, web, cloud, app signing, etc) - so the developers fixing this bug (Berthold) and working primarily on the planner (Robert) don't directly benefit from these donations. It is, however, surprisingly expensive and a lot of work to give something away for free these days.

As to your specific question, Microsoft, the company behind GitHub, so far is covering the credit card fees and therefore all sponsorship through GitHub reaches me 100%. Since Ko-Fi doesn't have that Trillion Dollar company behind it, I decided to sign up with a plan with them that gives 5% of all incoming sponsorships to help cover their costs.

PR #4048 (2024-01-16): Fix crash in planner

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Fix a very severe bug: double click on combo-boxes crashes and a minor bug (initialized variable that could lead to funky results).
### Related issues:

Fixes #4014

### Mentions:

#ejrobinson
commit bdade9e4 (2024-01-16): planner: initialize subobject of PlannerWidgets in correct order

The DivePlannerWidget was initialized before the planner models.
However, it attaches these (non existing) models to the comboboxes.
That can't work. Initialize in correct order.

Fixes #4014