subsurface/subsurface #4018: Please make signed versions available

mikeebowen, 2024-01-07:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ x] New feature request
- [ x] Discussion request

### Issue long description:
I am a developer, I understand that it is safe to install an unsigned version of an application if you trust the source, but most divers aren't tech people and the warnings from macOS are scary and it's confusing to find the steps to allow it to be installed. It's not as bad on Windows, but there is still a warning. The download also comes from GitHub, instead of your own website, which to non-technical people might look suspicious. 

I am a multi OS, multi brand dive computer user, a big supporter of open source and avid scuba diver, so I want this project's user base to grow as much as possible and I think that these extra warnings and hoops to install will push away many non-technical divers. I would humbly request that you make signed downloadable versions (obviously not daily, but at some interval, maybe quarterly or half yearly) available to download or be used with the update button from the app's UI.

I marked this as discussion and new feature, because I would like to get other's thoughts on this and because I don't know how much change/work it would involve. I do hope this is something that can happen. 

Thank you for making this fantastic app that has enhanced my diving experiences. 

### Operating system:

All

### Subsurface version:

NA

### Steps to reproduce:

NA

### Current behavior:

NA

### Expected behavior:

NA

### Additional information:

### Mentions:

dirkhh, 2024-01-07:
Well, quite a few large companies don't bother with signing apps. It actually appears to be more and more companies that just tell their users "yeah, just click on OK", especially on Windows.

And the effort to get certificates, maintain them, ensure that they are handled safely, etc, is significant. Actual corporate certificates are prohibitively expensive and complex to get. And "open source developer certificates" are designed to be extremely cumbersome to use.

Specifically, in most cases it's not possible to create signed binaries via CICD which means a developer (for the past ten years that means: @dirkhh ) has to manually build the artifacts and sign them.

And you know the funniest thing? A frequent request we get from Windows users was to make Subsurface available as a zip file because the installer was so inconvenient for their use case.

I get it, you want signed binaries. I get why vendors push for them. At the same time, pretty much all malware distribution that you get today is actually signed. Signing apps is 99.99% about creating revenue and 0.005% about actually improving the security of anything.

For iOS we still sign the updates. As has been well established by European courts - that's ALL about creating revenue for Apple, but sadly people who aren't lucky enough to live in the EU still can't sideload on iOS, so we HAVE to jump through all those hoops.
If I ever get overwhelmed by boredom I may set up signing for macOS again, it's only mildly painful and I need to keep paying for those certificates anyway in order to create iOS binaries. I'm certainly not going to sign every CICD build, but I might sign one every once in a while. Again. If I get that bored.

For Windows it's not gonna happen. After my last certificate provider screwed me over twice and I paid quite a bit of money for useless random numbers (that's all a certificate is), I will not go down that path again.

For Android I'm pretty sure I'll even stop trying to push the app into the Google Play store because they are making it absolutely impossible for a hobby project to keep up with their constantly changing rules.

atdotde, 2024-01-07:
So far, the "official" releases have been signed, just the intermediate, automatically generated testing version were not signed. Since currently there whole versioning is undergoing major changes I don't know what is planned but I would assume that there will still be signed release version (which are produced at somewhat stable states).

atdotde, 2024-01-07:
Oops, @dirkhh beat me.

mikeebowen, 2024-01-07:
@dirkhh Thank you for the quick and thoughtful response. The hoops and costs you describe sound like a tedious, expensive, and mostly pointless endeavor, I wish we had a better more honest, less profit-based system that supported passion projects given away for free like this to thrive and grow. I'll keep trying to talk the divers I meet to adopt subsurface, thanks for all you do for the divers out there!

closed 2024-01-07 by mikeebowen

dirkhh, 2024-01-07:
Thanks for your support and for understanding my side of all this.
Subsurface has exceeded all of our wildest expectations. We have more than 20k users.

Crazy. And scary.

mikeller, 2024-01-07:
Just as a side note, I am working on getting the CICD releases for android signed with a fixed, but self-signed certificate - this will improve things as it will make it possible to sideload new versions over the top of old versions without having to uninstall first, and remove the need to re-synchronise the data after every update.
But I suspect that there won't be similar quick wins for the automated builds for other platforms, as they all require manufacturer issued certificates to make a difference for the user, and adding such certificates to the cloud based CICD infrastructure seems to pretty much defeat the security aspect of these certificates.